xCuri0/ReBarUEFI #92: .EFI vs .FFS

r8bit99, 2023-11-05:
I don't understand specifics of EFI / DXE drivers, but seems they are closly linked. Just thinking.....can the ReBarDxe be loaded with an EFI shell, kind of like this:

```
echo step1
connect -r
set -v efishellmode 1.1.2
map -u
echo step 2
load fs0:\EFI\Drivers\Nvm.efi
echo step2
map -u
echo step3
fs[YOUR-IDENTIFIED-NUMBER-HERE]:\EFI\boot\bootx64.efi

```

Found this online, where loading of NVME drivers was using this method. Thought this could be a possibility for OEM BIOSes where modifications was difficult?

xCuri0, 2023-11-05:
@r8bit99 it won't work because ReBarDxe works by hooking `PreprocessController` which gets executed for every PCI device when the UEFI firmware initializes PCI devices.

This has been discussed before #6

closed 2023-11-05 by xCuri0

zoulztealer, 2024-05-27:
> @r8bit99 it won't work because ReBarDxe works by hooking `PreprocessController` which gets executed for every PCI device when the UEFI firmware initializes PCI devices.
> 
> This has been discussed before #6
> 
> There are still ways to get Resizable BAR without modifying BIOS, Using Linux with 4G decoding on or DSDT edit is the easiest.

hello curio, im currently investigating a similar issue with the dxe driver module from nvstrabsrebar.

i was trying to find a way to side- / chainload the preconfigured NvStrapsReBar.ffs with a grub 2 / linux based live usb prior to windows 10 to have the gpu driver inside win 10 access to the driver module and enable full or partial rebar  (anything higher than the default resized bar of 214/256mb).

ive created a question about the possibility on stackoverflow:

https://stackoverflow.com/questions/78533307/possibility-to-side-chainload-ffs-dxe-uefi-bios-rebar-driver-module-with

to me it seems theoretically possible but i have no clue about any of this stuff.

you said there would be an issue to trigger pci enumeration.

would it be possible to load the dxe driver as runtime persistently and then have the machine rebooted with the driver loaded and injected into pci enumeration?

8.8 DXE Runtime Driver A DXE runtime driver executes in both boot services and runtime services environments. This means the services that these modules produce are available before and after ExitBootServices() is called, including the time that an operating system is running. If SetVirtualAddressMap() is called, then modules of this type are relocated according to virtual address map provided by the operating system.

xCuri0, 2024-05-27:
i don't think it's possible for uefi to do pci re-enumeration.

it is very much technically possible to resize BARs after PCI initialization. linux does it, the first commit of this project did it, xelafics first version of nvstrapsrebar also did it. however it's kinda complicated to write an efi application which can do all this without any hardcoding

zoulztealer, 2024-05-27:
> i don't think it's possible for uefi to do pci re-enumeration.
> 
> it is very much technically possible to resize BARs after PCI initialization though. linux does it, the first commit of this project did it, xelafics first version of nvstrapsrebar also did it. however it's kinda complicated to write an efi application which can do all this without any hardcoding

thanks for quick answer. thats very interesting. according to my theory resizing the bar could be everything that needs to be done because all the rest of the mechanisms seem to be in place working already by default natively (rebar always present, always on with d3d9+ capable gpu, bar is active and has been resized, but size is statically limited to 214mb vulkan, 256mb hw).

do you have any idea why capframex is reporting full rebar in d3d for gtx1080 and 2080ti even when rebar is "physically disabled" (which it never is imo and is just a marketing scam)?

ive tried to investigate this with the capframex support but i received no answer.

could it be that d3d somehow can do a full software rebar via the a4g setting?

xCuri0, 2024-05-28:
> even when rebar is "physically disabled" (which it never is imo and is just a marketing scam)?
2080ti and all turing 16/20 series cards can do rebar using nvstrapsrebar
1080ti also could but nvidias driver does not support it

> do you have any idea why capframex is reporting full rebar in d3d for gtx1080 and 2080ti (as well as probably any other d3d gpu) even when rebar is "physically disabled" (which it never is imo and is just a marketing scam)?

probably just an error

> why are there different sizes reported for hw and vulkan? doesnt this mean its possible to achieve different rebar sizes with different graphic apis? like vulkan could probably support full memory as well like d3d. what would be the difference between d3d and vulkan in regards to this and why is this being done by the industry?

on stuff like amd polaris (my gpu) the driver disables rebar by default because it's experimental and needs a registry edit to enable it.

> another idea would be to merely trick windows 10 into believing rebar is "on" by only setting the environment / uefi variables that the machine receives in the case of actually fully configured rebar:

no such thing

zoulztealer, 2024-05-28:
> 2080ti and all turing 16/20 series cards can do rebar using nvstrapsrebar 1080ti also could but nvidias driver does not support it
>
>
> probably just an error
>
>
> on stuff like amd polaris (my gpu) the [...]

thx. im beginning to realize how much of a dead end this is because theres even issues left over with the officially fully supported and matured version of 3/4k series rebar that can lead to crashes and even reduced fps in some cases.

i wonder why its possible though to enable it with a reg tweak on amd but not with nvidia. thats why one of my main critic points was that all these mobos, gpus, etc are not fully compatible and not fully functional according to pcie specs. this is just not designed unitarily.

according to pcie specs its also provided in the implementation note that system software be provided to the user that can control and identifiy rebar capable devices which the industry never supplied. it was also meant for the function to report back to the user several optimal bar sizes to chose from. 

since the loss of performance was well aware to the industry at that time is why i argued that all these products come with a legal flaw to trick consumers into paying again an extra fee for something they already own.

but who am i to say this but a nasty, greedy gamer.

my next step will be to harass zotac and nvidia about this before in finally give up.

;)

zoulztealer, 2024-05-28:
capframex screen from my other wholly unmodified pc:

r1700
rx570 8gb
asus prime b450 k2 w/ factory bios

[image]

a4g is disabled on this machine and capframex reports no full rebar in d3d. bar sizes are also different for vulkan than with nvidia. i still suspect the a4g setting to enable d3d to access all of the vram. ill try it later with a bios update. i didnt enable a4g because it caused black screens with that gpu on x79 which is apparently a bug where you need to to connect a dvi connection to get a picture again.

any suggestions on the lightest / least invasive tweaks i could do to enable rebar and make some checks on the way with that rig? reg tweak a4g should be enough probably?

i have another system with r3600, rx590, asrock ab350 pro4.

and hey, theres a box with my old gtx1080, but its too big to fit one of the other cases!

and heres a gtx970, too, which a friend refuses to take even for free to upgrade his gtx960.

this must be my destiny.

zoulztealer, 2024-05-29:
Rakesh P N: Hi, my name is Rakesh P N. How may I help you?
 . .: hello
 Rakesh P N: hello!
 . .: i have a question regarding rebar availability for rtx 2000 and gtx 1600 series
 Rakesh P N: Rebar?
 Rakesh P N: May I know what it is?
 . .: https://www.nvidia.com/en-us/geforce/news/geforce-rtx-30-series-resizable-bar-support/
 Rakesh P N: Oh, resizable bar!
 . .: yes exactly
 Rakesh P N: GTX series doesn't have resizable bar
 Rakesh P N: RTX 2000 is it a quadro card?
 . .: generally rtx 2000 series
 . .: more exactly rtx2080ti in this case
 Rakesh P N: Then its supported
 . .: so on this screenshot rebar is officially supported by nvidia 
 . .: https://i.redd.it/favc5wjhhxw91.jpg
 . .: bar and rebar were introduced with the pcie 2.0 and updated with pcie 2.1 specifications
 Rakesh P N: Its not supported on your PC?
 . .: yes the issue its not supported but it should be
 Rakesh P N: May I know the name of card manufacturer of your RTX 2080ti?
 . .: yes its zotac amp rtx2080ti
 File attachment upload has started.
 The file gpuz4.gif (24.26KB) was received.
 File attachment upload has started.
 The file gpuz4rebar.gif (14.16KB) was received.
 Rakesh P N: The motherboard you're using is?
 . .: gigabyte x570 ud rev. 1.0
 . .: the thing is rebar is always on but the user is prohibited from changing the bar size
 Rakesh P N: I suggest you contact Zotac and check with them for the Resizable bar update.
https://www.zotac.com/news/get-resizable-bar
 . .: basically every d3d9+ capable gpu can do partial or full rebar
 . .: in capframex full rebar is reported for gtx1080 and rtx2080ti even when rebar is disabled
 File attachment upload has started.
 The file capframex.png (248.91KB) was received.
 Rakesh P N: Sorry, not sure about Capramex
 Rakesh P N: You need to check with ZOtac once
 . .: ok i will try thanks
 . .: can you confirm that nvidia officially supports full rebar for rtx2080ti?
 Rakesh P N: Yes it doesn
 Rakesh P N: But
 Rakesh P N: Its not the chipset
 Rakesh P N: Its the card
 Rakesh P N: Zotac needs to support
 . .: ok very interesting thank you

> a few hours later ...

This is not available. Only updates done were for the 30 series cards and none for 20 series unfortunately.

Cheers,

ZOTAC Technical Support

Department: Technical Support
Type: Issue
Status: Closed

END OF STORY