xCuri0/ReBarUEFI #96: i can't flash modded uefi file

leonpano2006, 2023-11-08:
**System**
 - Motherboard: HPE ML350 Gen9
 - BIOS Version:P92_3.30
 - GPU:RTX A5000 + ilo4 2d (disabled after post)
 - [ ] CSM is turned off --> i have no reason to install legecy system at all
 - [ ] 4G decoding is enabled --> even i find no option of 4G decoding, but gpuz confirm is enabled
 - [ ] UEFIPatch is applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [ ] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**

after i modify with ReBarDxe thing and build
then i tried to flash and get this error
[image]

[P92_3.30_09_21_2023.zip](https://github.com/xCuri0/ReBarUEFI/files/13299338/P92_3.30_09_21_2023.zip)
unmodded uefi and modded uefi inculded

xCuri0, 2023-11-09:
It probably has a signature check you will need to use another method like hardware flasher

leonpano2006, 2023-11-09:
> It probably has a signature check you will need to use another method like hardware flasher

And then ilo may complain about UEFI on startup and halt device

closed 2023-11-21 by xCuri0