xCuri0/ReBarUEFI #97: Can't enable rebar(Asus X99-E WS)

2ztech, 2023-11-16:
Specs:
Intel Xeon E5-2670v3
Asus X99-E WS
32GB DDR4 2133Mhz ECC Ram
Intel Arc A380
Lexar 128GB 2.5" SSD

I will explain what I'm doing till the last point.

1.I downloaded 4001 bios file.

2.Add rebardxe file using UEFITool.

3.Run UEFIPatch with command .\UEFIPatch.exe .\biosfilename.cap -o biosfilename.cap

4.I tried check the pad file changes, actually I'm confused here as the original file only have 1 pad-file while the inserted rebardxe file and patched file have 2 pad-file under different filename. (Maybe I'm doing wrong here.

5. Thinking that there's nothing wrong in step 4, i flash the bios using afudos.

6. Successfully flash the modded bios, i know this because i use arc a380 so before this mod it can't even display output but after bios update it can produce output.

7.Installed windows 10 for testing, however on windows auto update there's conflict with the a380 driver which make the system freeze after startup. The cursor can move but can't receive input.

8.Installed new driver in safe mode and run ReBarState.

9. Launch windows normally without any freeze occuring. Open the gpu-z but Rebar is disabled.

Here i give the relevant picture.

[image]

[image]

[image]

2ztech, 2023-11-16:
I just tested benchmark with Unigine Heaven, got 66fps ultra quality and extreme tessellation. 80+ fps medium setting CS2.

xCuri0, 2023-11-16:
@2ztech Send screenshot of ReBarState open

2ztech, 2023-11-16:
> @2ztech Send screenshot of ReBarState open
> 
Like this?

[image]

xCuri0, 2023-11-16:
@2ztech are you sure you flashed a BIOS with ReBarDxe and that it's inserted correctly ?

every issue like this has turned out to be a mistake like that

2ztech, 2023-11-18:
> @2ztech are you sure you flashed a BIOS with ReBarDxe and that it's inserted correctly ?
> 
> every issue like this has turned out to be a mistake like that

I'm sure I flashed the rebardxe cuz if not I can't boot up using the A380. Anyway i think it's okay not having able to enable rebar cuz I just plan to use it in my home server for transcoding . Thanks for the help.

xCuri0, 2023-11-18:
@2ztech rebardxe doesn't make any difference until you use rebarstate, it's disabled until you enable it using rebarstate.

you should be able to boot a380 without rebar btw, just bad performance

xCuri0, 2023-11-21:
update ?

xCuri0, 2023-12-02:
Closing due to inactivity