xCuri0/ReBarUEFI #98: MMTool method creates unusable bios, UEFITool OC issue.

FPSUsername, 2023-11-20:
**System**
 - Motherboard: ASUS ROG Maximus VII Ranger
 - BIOS Version: 3503
 - GPU: ASUS RT6600
 - [x] CSM is turned off
 - [x] 4G decoding is enabled
 - [x] UEFIPatch is applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [x] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**

Unfortunately, this is one of those motherboards that will break overclocking and post issues when the bios gets modified by UEFITool. I first found out when I used UBU Tool to update microcode and other firmwares in the past and since just used the 3503 provided by ASUS ROG (original).

I found out about ReBar to add support for my newly acquired RX6600 and went with the normal method of using UEFITool and UEFIPatch. The bios flashes successfully and ReBar works, except overclocking now doesn't work (which is quite essential on DDR3 ram in my case).

I used MMTool 5.0.0.7 and 5.2.0.24 and they both modify the bios in the exact same way (but different to UEFITool of course). After successfully applying the patches with UEFIPatch, the system rejects the bios as an update via the EZ Update tool, nor does it accept via the flashback method (the indicator stays turned on instead of blinking).

Did I miss any step when using MMTool (I followed the steps in the wiki)? Or am I supposed to use MMTool v4.x instead?

xCuri0, 2023-11-21:
It was fixed here [https://winraid.level1techs.com/t/asus-z97-maximus-vii-hero-readjustable-bar/90930/38](https://winraid.level1techs.com/t/asus-z97-maximus-vii-hero-readjustable-bar/90930/38)

@Koekieezz do you remember how you fixed this ? Want to add it to the guide

FPSUsername, 2023-11-21:
Yes! The keyboard bug was indeed present as well on my end as well.

xCuri0, 2023-11-22:
Have you tried other versions of MMTool ?

FPSUsername, 2023-11-22:
It seems that MMTool 4.50.023 generates a different output than 5.0.0.7 (which does the same as 5.2.0.24).
I'll try to flash that later today.

xCuri0, 2023-11-22:
@FPSUsername don't replace any modules

follow the method in UEFIPatch wiki page so only MMTool is used on the final BIOS, preventing any corruption.

xCuri0, 2023-11-22:
@FPSUsername NvramSmi patch is done by UEFIPatch, you don't need to do anything but follow the written pad file workaround in the guide.

FPSUsername, 2023-11-22:
[image]
Unfortunately the bios is rejected when patching with MMTool 4.50.023 and using UEFIPatch afterwards.

I attached the original, modified and patched bios files.

[VII Ranger 3503 ReBar MMTool 4.50.023.zip](https://github.com/xCuri0/ReBarUEFI/files/13441970/VII.Ranger.3503.ReBar.MMTool.4.50.023.zip)

xCuri0, 2023-11-22:
@FPSUsername You should **NOT** run UEFIPatch directly on the BIOS you're going to flash because it will cause issues like what you have with OC.

https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch#pad-file-issue-workaround this is what you should do so the BIOS you flash only get's touched by MMTool, and make sure you insert ReBarDxe with MMTool too

Use version 4.5.023 it's what others have been successful with

FPSUsername, 2023-11-22:
@xCuri0 It doesn't seem to matter. I created the patched bios files with MMTool 4.50 and UEFITool v28 method and both generate the exact same file before patching with UEFIPatch. The hashes are the same, pad file is not altered.

So it seems like the VII Ranger is not affected by the bug, however, the OC issue still stands. 

Continuing from the pad file workaround, I extracted the PciBus and PciHostBridge from the patched bios and replaced the PciBus and PciHostBridge from the bios with ReBar injection via MMTool and just to try, also the stock bios.

Reopening the bios files in UEFITool to verify, the PciBus full size and body size is different on the files with the patched ffs injected via MMTool. I'll try to flash both and see what happens.

To clarify, ReBar does work, but altering the bios in any way breaks overclocking and adds the no keyboard bug found in the VII Hero motherboard, before Kookieezz found a solution.

xCuri0, 2023-11-23:
@FPSUsername It's a know issue that UEFITool causes stuff like this.

If you can figure out how to flash the MMTool BIOS it will most likely work

xCuri0, 2023-11-23:
@FPSUsername Your issue **is not NvramSmi**, that module only handles NVRAM like ReBarState and gets fixed by UEFIPatch.

If you do MMTool 4.5 like I said you're BIOS will probably work with OC and everything, not just Z97 I saw this exact issue on a Z170 board too fixed by using MMTool exclusively.

FPSUsername, 2023-11-25:
I've used [this](https://winraid.level1techs.com/t/offer-asus-rog-maximus-vii-ranger-bios-3503-mod/32388/6) bios as a base (as it's updated and it works with overclocking), then did the following:
1. Add ReBar to the PCIe bus with MMTool v4.50.023
2. Ran UEFIPatch with the patch.txt from this repo
3. Opened the .patched bios in UEFITool A67 and extracted "as is" the PCIeBus and PCIeHostBridge
4. Opened the ReBar bios (from step 1) in MMTool v4.50.23 and replaced the PCIeBus and PCIeHostBridge with the extracted ffs files.
5. Save as M7R.CAP and flash
6. Overclocking works, rebar doesn't as it shows not enabled in BIOS.

[image]
[image]

xCuri0, 2023-11-25:
@FPSUsername You have to do the same thing you did with PciBus to NvramSmi.

It will work then

xCuri0, 2023-11-25:
Also later once you have it all working can you test using UEFITool for the first step only (adding ReBarDxe) and the same MMTool steps for the rest ?

I want to know if I have to update first step of guide to say use MMTool on ASUS.

FPSUsername, 2023-11-25:
[image]
I had to do the same to NvramSmi to get ReBar working. Overclock & ReBar.

I did get the no keyboard bug. I'll investigate it later and also using UEFITool. The issue I believe is UEFIPatch, because UEFITool makes identical files compared to MMTool 4.50.23.

So a preliminary conclusion: for ASUS motherboards, you'll have to replace the modules with MMTool after extracting them from a patched version.

xCuri0, 2023-11-25:
> I'll investigate it later and also using UEFITool. The issue I believe is UEFIPatch, because UEFITool makes identical files compared to MMTool 4.50.23. 

If that's the case then there's no need to do what I said.

About the keyboard bug did you apply this patch
[image]

It modifies PciBus btw, so you will need to apply both patches before extracting the PciBus module for using in MMTool

FPSUsername, 2023-11-25:
@xCuri0 Thank you for guiding me through, it worked!

To make things easier in the future, I believe the trick is as following:
1. Inject/update/do anything with UEFITool/MMTool/UBU and save the bios
2. Apply patches with UEFIPatch
3. Extract what's patched and the NvramSmi using UEFITool from the patched bios
4. Open the BIOS from step 1 with MMTool 4.50.023 and replace the modules with the extracted ffs modules

So this bios can also be added to the list:

| User  | Motherboard | GPU | BAR size tested | UEFIPatch req (Y/N) | Notes |
| ------------- | ------------- | ------------- | ------------- | ------------- | ------------- |
| @FPSUsername  | ASUS VII Ranger | RX 6600 8GB | 8GB | Y | Need to extract PCIeBus, PCIeHostBridge and NvramSmi from the UEFIPatch patched bios and replace the modules in an unpatched bios. [BdwUSB3.txt](https://github.com/xCuri0/ReBarUEFI/blob/master/UEFIPatch/BdwUSB3.txt) patch must also be included. |

As a treat, here is the modified ReBar bios with the following changes (Thanks to [voron00](https://winraid.level1techs.com/t/offer-asus-rog-maximus-vii-ranger-bios-3503-mod/32388/6) for updating the BIOS in 2019 with pretty much the final updates).

Modules:
- OROM IRST RAID for SATA v13.0.0.2075 -> v14.8.2.2397
- EFI IRST RAID for SATA v13.0.0.2075 -> v14.8.2.2397
- EFI GOP Driver HSW-BDW v5.5.1024 -> v5.5.1034
- OROM Intel Boot Agent GE v1.5.47 -> v1.5.62
- EFI Intel PRO/1000 UNDI v6.0.03 -> v6.6.04
- **ReBarDxe -> 0.3**

Microcode:
- Haswell: 27 - cpu306C3_plat32_ver00000027_2019-02-26_PRD_4F5914B9.bin
- Broadwell: 20 - cpu40671_plat22_ver00000020_2019-03-07_PRD_5246E87E.bin

[MAXIMUS-VII-RANGER-ASUS-3503-REBAR.zip](https://github.com/xCuri0/ReBarUEFI/files/13465243/MAXIMUS-VII-RANGER-ASUS-3503-REBAR.zip)

closed 2023-11-26 by xCuri0

iulisan27, 2024-06-05:
@FPSUsername Hey, how are you doing? I was trying to flash the 3503 rebar bios you uploaded here but i get the error saying "Selected File is not a proper BIOS". Do you know what could it be?

FPSUsername, 2024-06-05:
You need to flash it via the emergency flasher.
Note the following:
1. The file needs to be renamed using the rename tool. For this specific motherboard it's M7R.CAP.
2. The file should be placed on a FAT32 formatted USB stick.
3. The motherboard doesn't accept all USB sticks. I used an old 2GB stick that works, my other 16GB stick didn't work. (You'll see the light flicker constantly when it's flashing. If it doesn't, it's not working).

iulisan27, 2024-06-05:
The strange thing is that i could flash another bios with the same usb stick on fat32. I'll try again but with the flashback button as you said. Thank you!

iulisan27, 2024-06-05:
@FPSUsername I think i managed to install it with the flashback button but i don't see the rebar option in the bios. Do you know where is that located? 
[image]

FPSUsername, 2024-06-05:
If you didn't flash the bios by voron00 and use the latest provided by Asus, then try the following.

With one of the bad tools from Asus, you can read the current bios on your system (I assume you already have it as you tried to flash it the normal way first).

Make a backup/readout of your current installed bios. Then open it in UBU tool and see if the modules have older versions than what I posted in [this](https://github.com/xCuri0/ReBarUEFI/issues/98#issuecomment-1826419382) comment. Note that they're the same as the bios provided by voron00. All I did was take his bios and add rebar.

I don't exactly remember how the light is supposed to react, but I think it blinks 3 times, then starts flashing rapidly till the BIOS is flashed. If it continues doing the same slow flashing pattern, then it's waiting for a device with the BIOS .

iulisan27, 2024-06-05:
@FPSUsername I installed the one you uploaded on November 25 2023. https://github.com/xCuri0/ReBarUEFI/files/13465265/MAXIMUS-VII-RANGER-ASUS-3503-REBAR.zip is this the right one?

FPSUsername, 2024-06-05:
There's no new option in the bios. Run the rebar tool in your operating system and set rebar to 8GB. Reboot and check.

iulisan27, 2024-06-05:
Oh i think i missed that step, i'm sorry. I'm gonna check it and inform you later.

iulisan27, 2024-06-05:
@FPSUsername what value should i set for it to be 8192mb?

FPSUsername, 2024-06-05:
2^13 = 8192.
It's simple math

iulisan27, 2024-06-05:
Yeah you are right. Now i feel dumb

iulisan27, 2024-06-05:
I got it working thanks to you! Now i got to squeeze the last bit of juice of this old system
[image]

FPSUsername, 2024-06-05:
You need a compatible GPU to make use of ReBar, for now it's enabled on the motherboard, but it will do nothing (aa BAR1 is still at 256MB).

Once you get a compatible GPU, set the rebar size to the same size as the GPU (mine is 8GB so I chose 8GB).

From ny understanding (and correct me if I'm wrong), ReBar allows the CPU to grab chunks of data from the GPU,  by default limited to 256MB. Increasing it to 8GB makes it possible to grab all data at once, reducing the need of grabbing multiple chunks, freeing up CPU time.

So I think you'll see massive improvements if a system is bottlenecked by the CPU (as is with mine, the RX6600 was however the best bang for the buck, it didn't make sense to settle with less for only a 20 euro difference).

xCuri0, 2024-06-05:
RX Vega supports Resizable BAR, just make sure to enable it in registry. BAR0 is what's used for the VRAM on AMD

iulisan27, 2024-06-05:
I'm currently using a Rx Vega 64 with 8gb vram so i set that bar size. Its as xCuri0 said so it should work fine with my gpu. I'm not sure how much performance I may able to gain, but I wanted to try this since i've heard old cpus take more advantage from this (mine is an i7 4790k).

Nero10578, 2024-06-19:
> @xCuri0 Thank you for guiding me through, it worked!
> 
> To make things easier in the future, I believe the trick is as following:
> 
> 1. Inject/update/do anything with UEFITool/MMTool/UBU and save the bios
> 2. Apply patches with UEFIPatch
> 3. Extract what's patched and the NvramSmi using UEFITool from the patched bios
> 4. Open the BIOS from step 1 with MMTool 4.50.023 and replace the modules with the extracted ffs modules
> 
> So this bios can also be added to the list:
> 
> User	Motherboard	GPU	BAR size tested	UEFIPatch req (Y/N)	Notes
> @FPSUsername	ASUS VII Ranger	RX 6600 8GB	8GB	Y	Need to extract PCIeBus, PCIeHostBridge and NvramSmi from the UEFIPatch patched bios and replace the modules in an unpatched bios. [BdwUSB3.txt](https://github.com/xCuri0/ReBarUEFI/blob/master/UEFIPatch/BdwUSB3.txt) patch must also be included.
> As a treat, here is the modified ReBar bios with the following changes (Thanks to [voron00](https://winraid.level1techs.com/t/offer-asus-rog-maximus-vii-ranger-bios-3503-mod/32388/6) for updating the BIOS in 2019 with pretty much the final updates).
> 
> Modules:
> 
> * OROM IRST RAID for SATA v13.0.0.2075 -> v14.8.2.2397
> * EFI IRST RAID for SATA v13.0.0.2075 -> v14.8.2.2397
> * EFI GOP Driver HSW-BDW v5.5.1024 -> v5.5.1034
> * OROM Intel Boot Agent GE v1.5.47 -> v1.5.62
> * EFI Intel PRO/1000 UNDI v6.0.03 -> v6.6.04
> * **ReBarDxe -> v0.3**
> 
> Microcode:
> 
> * Haswell: 27 - cpu306C3_plat32_ver00000027_2019-02-26_PRD_4F5914B9.bin
> * Broadwell: 20 - cpu40671_plat22_ver00000020_2019-03-07_PRD_5246E87E.bin
> 
> [MAXIMUS-VII-RANGER-ASUS-3503-REBAR.zip](https://github.com/xCuri0/ReBarUEFI/files/13465265/MAXIMUS-VII-RANGER-ASUS-3503-REBAR.zip)
> 
> SHA256 from M7R.CAP `4CCCB56ABB490E7A0745E1D2C8F90EB9B4BA582169524754314F32F06ECBD9AF`

Hey can I ask for clarification for the steps? I am trying to add Rebar to an Asus Rampage V Edition 10 which according to the supported motherboard list requires this method your showed.

Specifically I want to ask what do you mean in step 3 by extract what's patched? Then for step 4 you put the patched modules back in to the bios that was already modified in step 1? 

Any help is appreciated. Thanks!

FPSUsername, 2024-06-20:
You need to extract the patched modules using MMTool. Then open the modified bios (before it was run with UEFIPatch for rebar) inMMTool and there replace the modules with the extracted patched modules.
Don't forget the USB patch if the keyboard doesn't work.

Jobboys, 2025-11-17:
> When using MMTool 4.50.0.23 I cannot save the bios after replacing the PciHostBridge and PciBus .ff modules, when i try "Save image as" it says "Error in Saving" and the file that it creates is empty.
> 
> I have a z170 board if that makes any difference.

If i were to just use UEFITool 0.28.0 instead of MMTools to replace the PciHostBridge and PciBus because one of the earlier replies state that they output the same files. would that work?

xCuri0, 2025-11-24:
All these issues with Haswell/Broadwell being limited to 8GB BAR or 16GB RAM are fixed now, you need to remake the BIOS with the new UEFIPatches.

Intel software expects 16GB BAR and likely doesn't work properly in games without it.

Koekieezz, 2025-11-24:
> All these issues with Haswell/Broadwell being limited to 8GB BAR or 16GB RAM are fixed now, you need to remake the BIOS with the new UEFIPatches.

Wait, so now H97 and Z97 could do full rebar to 39bit unlike before? I always tought even with the latest uefipatches, whenever you do 32gb (4x8gb) of ddr3 ram on H97 and Z97, it wont be able to rebar past 8GiB bar size, which was caused by "old ivy bridge era bios left over" as you stated before, is the tutorial/issues now updated with this? Last time i checked it says H97/Z97 does have the same issue as ivy/sandy chipsets for rebar size.

Koekieezz, 2025-11-24:
> Is it possible to get the modified BIOS with the new UEFIPatches?

What board? I might able to do it later.

Koekieezz, 2025-11-25:
> @Koekieezz 
> Wiki guide was updated and there are new patches.
> 
> They are in a separate file though because for some reason they also apply to LGA1151 and LGA1155 which might break stuff on them.

Did you mean these:
[image]

[image]

Koekieezz, 2025-11-26:
> WOW !!! Amazing...
> Thank you very much for the modded bios @Koekieezz .
> It works absolutly fine.
> 
> Options 14 and 32 in ReBarState works without any issues.
> USB ports working at boot without any issues too.
> XMP @2400 MHZ and cpu overclock to 4.8 GHZ working too.
> I report it to the post you tell me too.
> Screenshots following here now.
> 
> [image]
> [image]
> [image]
> [image]
> [image]
> [image]
> [image]

Thank you so much for actually testing and confirming that Cpu OC and Ram OC with your previous settings worked, it was an issue back then and i cant figure out how to avoid it, so the past z97/h97 i've done on asus brand was left half made (working, but there is a quirk).

I have long been someone who avoid asus board due to them being sensitive with padding issues, this is probably one of the mod i've done with all complete modules updated and microcodes too. Basically this bios is the same as @FPSUsername bios in terms of modules updated, but this time i downgrade to Raid driver 13.5 since i personally never use raid on 9 series chipset since i worried about its compatibility, updated the gop vbios to version 1044 from 1029, and the Intel ME to version 9.1.40 (if you dont mind, confirm this by going to bios -> main menu, it should state the current intel me firmware version).

choigamevui621, 2025-11-28:
> > H97 - PRO have 4G decode.
> > <img alt="Image" width="1593" height="607" src="https://private-user-images.githubusercontent.com/246315987/519263458-07a63c12-ea7b-441b-b24c-ffa98f1a59eb.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjQyNjExNDIsIm5iZiI6MTc2NDI2MDg0MiwicGF0aCI6Ii8yNDYzMTU5ODcvNTE5MjYzNDU4LTA3YTYzYzEyLWVhN2ItNDQxYi1iMjRjLWZmYTk4ZjFhNTllYi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUxMTI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MTEyN1QxNjI3MjJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05MGQ0Yjg3NmEzMzc0MTQ4OGQ2Y2YyNjk1NjZiZTgyMWY3Y2Q4MzFhZTE3MmQxNjhjZmY0YThhZDBlNTNlMmMwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.RuVUL_7huOgHzjGzeMotNeOE05CGUB0OinBFxvIwyRk">
> > So can you make me a copy on bios h97-pro ? I will try to update the bios you gave. ( add help me patch USB 3 inside )
> > Intel 9 Series Chipset (H97/Z97) will additionally need to apply [BdwUSB3.txt](https://github.com/xCuri0/ReBarUEFI/blob/master/UEFIPatch/BdwUSB3.txt) for functioning USB 3 ports in BIOS with 4G Decoding enabled. Use .\UEFIPatch YOURBIOS.bin.patched .\BdwUSB3.txt -o YOURBIOS.bin.patched after downloading and placing it in the same folder as UEFIPatch.
> 
> Here, try this [@choigamevui621](https://github.com/choigamevui621): [H97Pro_Rebar.zip](https://github.com/user-attachments/files/23807261/H97Pro_Rebar.zip)

thank you. let me ask if this patch has added usb3 yet ?. and it seems i can't update bios because its original file format is .cap

Koekieezz, 2025-11-28:
> thank you. let me ask if this patch has added usb3 yet ?. and it seems i can't update bios because its original file format is .cap

Yes, every patch is included. Please refer to [this](https://winraid.level1techs.com/t/guide-how-to-flash-a-modded-ami-uefi-bios/30627) on how to flash the .rom.

And use this ME updater to run the ME Firmware Update: [ME Updater ASUS.zip](https://github.com/user-attachments/files/23819367/ME.Updater.ASUS.zip)

Koekieezz, 2025-11-29:
> i tried your bios. But EFI Error ( at create Timer ). and front usb3.0 is not usable.
> 
> CSM disable 4G decode enable
> 
> [image]
> 
> [image]

Did you flash it correctly? Try to go to Device Manager -> View -> Resource By Type, and see on large memory.

Koekieezz, 2025-11-29:
> i tried your bios. But EFI Error ( at create Timer ). and front usb3.0 is not usable.
> 
> CSM disable 4G decode enable

@choigamevui621 Im sorry that i didn't mention it earlier, but i use the latest pre-release (v0.4) version, run [this](https://github.com/terminatorul/NvStrapsReBar/releases/download/v0.4/NvStrapsReBar.exe) as admin instead (if you didn't use it yet), and see if it works. Also check on Device Manager -> View -> Resource By Type -> Large Memory and see if it successfully alligned.

If you dont mind, please make a gpuz screenshot like this:
[image]

And if you did, can i use your screenshot (i will credit you for it) to make a post in win-raid, after everything successfull (including USB 3 working)?

choigamevui621, 2025-11-29:
> > Tôi đã thử BIOS của bạn. Nhưng lỗi EFI (khi tạo Timer). và cổng USB3.0 phía trước không sử dụng được.
> > CSM vô hiệu hóa giải mã 4G bật
> 
> [@choigamevui621](https://github.com/choigamevui621)Xin lỗi vì đã không đề cập đến vấn đề này sớm hơn, nhưng tôi đang sử dụng phiên bản tiền phát hành mới nhất (v0.4), chạy phiên [bản này](https://github.com/terminatorul/NvStrapsReBar/releases/download/v0.4/NvStrapsReBar.exe) với quyền admin (nếu bạn chưa sử dụng) và xem nó có hoạt động không. Ngoài ra, hãy kiểm tra trong Device Manager -> View -> Resource By Type -> Large Memory và xem nó đã được căn chỉnh thành công chưa.
> 
> Nếu bạn không phiền, vui lòng chụp ảnh màn hình gpuz như thế này: <img alt="Hình ảnh" width="888" height="785" src="https://private-user-images.githubusercontent.com/112951868/520376810-9410af55-e149-40a1-8c88-38468ae9983a.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjQ0MjQzNDEsIm5iZiI6MTc2NDQyNDA0MSwicGF0aCI6Ii8xMTI5NTE4NjgvNTIwMzc2ODEwLTk0MTBhZjU1LWUxNDktNDBhMS04Yzg4LTM4NDY4YWU5OTgzYS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUxMTI5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MTEyOVQxMzQ3MjFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wYjA2ZDdhN2NmZGY5OThhNjMyMzlkOTI4MzRjMWI4NzIzZTdlYWJlZTExNzFlYTllNDQ2NzQyNWNkNmU2YWUwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.3XDYc1Cm0EWZJnctRHKICOVqiB6lt2bg4Q1nYCOxslc">
> 
> Và nếu bạn đã làm vậy, tôi có thể sử dụng ảnh chụp màn hình của bạn (tôi sẽ ghi công bạn) để đăng bài trên win-raid sau khi mọi thứ đã thành công (bao gồm cả USB 3 hoạt động) không?

[image]
[image]

[image]

choigamevui621, 2025-11-29:
> [@xCuri0](https://github.com/xCuri0)Tôi biết đây là git sai cho nvstraps, nhưng nhìn vào dung lượng bộ nhớ lớn của anh ấy, liệu điều này có nghĩa là Giải mã Above 4G không được căn chỉnh thành công không? Nó không hiển thị 7F. dsdt của anh ấy cũng tương tự như QWordMemory, ghi rõ tham chiếu ngoài.
> 
> [@choigamevui621](https://github.com/choigamevui621)Bạn đã flash bios thành công chưa? Vì thanh này có vẻ nhỏ, vì tôi cũng làm tương tự trên Maximus VII Ranger (chipset Z97).

I have upgraded the BIOS you sent successfully according to the video below, but the REBAR in GPU-Z says it is not enabled in BIOS.

https://youtu.be/DwJS8-GTgRU?si=HcAISzGMcucqflT_

Koekieezz, 2025-11-30:
> [@Koekieezz](https://github.com/Koekieezz) The large memory looks like it only goes upto 64GB, did you modify the DSDT ? Most Haswell/Broadwell don't need it because the BIOS automatically passes the above 4g region (which should be UEFIPatched) to the DSDT.

@xCuri0 I didnt, since i tought it is not needed based on my result, also i hestitant to do the DSDT because of these lines:
[image]
[image]
Its Patches and DSDT is similiar to Maximus VII Ranger i did earlier, so i didn't do it.

@choigamevui621 Try to downgrade to any bios lower than the current (try flashing stock bios 2803), then try to re update with this bios i provide below.
Changes:
[image]

Bios: [H97Pro_RebarDSDT.zip](https://github.com/user-attachments/files/23837815/H97Pro_RebarDSDT.zip)

choigamevui621, 2025-11-30:
> > [@Koekieezz](https://github.com/Koekieezz) Bộ nhớ lớn trông như chỉ lên đến 64GB, bạn đã chỉnh sửa DSDT chưa? Hầu hết Haswell/Broadwell không cần nó vì BIOS tự động chuyển vùng 4g ở trên (cần phải được vá UEFIP) sang DSDT.
> 
> [@xCuri0](https://github.com/xCuri0)Tôi không làm vậy vì tôi nghĩ là không cần thiết dựa trên kết quả uefipatch của tôi: <img alt="Hình ảnh" width="960" height="183" src="https://private-user-images.githubusercontent.com/112951868/520453213-5a00a523-d86f-44e3-a74d-b3a13fdf35f8.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjQ1MDI3OTQsIm5iZiI6MTc2NDUwMjQ5NCwicGF0aCI6Ii8xMTI5NTE4NjgvNTIwNDUzMjEzLTVhMDBhNTIzLWQ4NmYtNDRlMy1hNzRkLWIzYTEzZmRmMzVmOC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUxMTMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MTEzMFQxMTM0NTRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05YjA5MTliMjQ4ZDZhZmQ2YjUyYmFmMTMzODU0N2M5ZmExMzVmYzE5YzRkNDA2ZTVmYTBkYjAzNWZlNDUyNzcyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.L9K4ih5TJHZhae2n5I0bJHZz1zEIsqwwQbRpIVynWgQ">
> 
> ngoài ra tôi còn do dự khi thực hiện DSDT vì những dòng này: <img alt="Hình ảnh" width="731" height="522" src="https://private-user-images.githubusercontent.com/112951868/520453201-92f83215-5d91-41b5-bc83-d7bea1e74ba6.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjQ1MDI3OTQsIm5iZiI6MTc2NDUwMjQ5NCwicGF0aCI6Ii8xMTI5NTE4NjgvNTIwNDUzMjAxLTkyZjgzMjE1LTVkOTEtNDFiNS1iYzgzLWQ3YmVhMWU3NGJhNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUxMTMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MTEzMFQxMTM0NTRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jYmE5MzI5NTkxMjVhMzE3MjIwNDY5NGIzNGFiNDI1YTVjYjJkMGU0NDg3ZDFhYWEyNDNiNzE3MmQ1MzUxZjEyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.AD89NqlTzP3lkPbYqjE4-rNF1FL58SV38Uw-GUevvL8">
> 
> Patches và DSDT của nó tương tự như Maximus VII Ranger mà tôi đã làm trước đó, vì vậy tôi đã không làm điều đó.
> 
> [@choigamevui621](https://github.com/choigamevui621)Hãy thử hạ cấp xuống bất kỳ bios nào thấp hơn bios hiện tại (thử flash bios gốc 2803 bằng ez flash như bình thường), sau đó thử cập nhật lại bằng bios mà tôi cung cấp bên dưới, theo phương pháp bạn đã thực hiện để flash bios đã mod của tôi. Thay đổi: <img alt="Hình ảnh" width="710" height="623" src="https://private-user-images.githubusercontent.com/112951868/520457820-8e24bc07-2fcf-40f1-8970-7b289c1885a6.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjQ1MDI3OTQsIm5iZiI6MTc2NDUwMjQ5NCwicGF0aCI6Ii8xMTI5NTE4NjgvNTIwNDU3ODIwLThlMjRiYzA3LTJmY2YtNDBmMS04OTcwLTdiMjg5YzE4ODVhNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUxMTMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MTEzMFQxMTM0NTRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02YWMzMzBlNmY2ODBlMGY1ZWIzMWEwNGU5OGVjMWU4NjY0Yjc4MGJhYjc5YzY4ZjQyNDFhZTAzNzk0YzY4MWE1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.ywYNIUJVmiK4LLr8JdMqTWOhyYEPnJGiVxLZmO2QHgU">
> 
> Bios: [H97Pro_RebarDSDT.zip](https://github.com/user-attachments/files/23837815/H97Pro_RebarDSDT.zip)
> 
> Giờ tôi tò mò không biết bạn có thực sự hiểu cách cập nhật bios đã mod không, bằng cách dùng AFUWIN/AFUDOS phiên bản 3.05 có hỗ trợ /GAN, rồi dùng .CAP gốc trước, đợi đến khi hoàn tất, rồi làm tương tự với .ROM tôi đưa, nhưng thêm /GAN vào cuối thì flash xong. Thật sự mà nói, điều này làm tôi tò mò không biết bạn đã flash bios đã mod mà tôi đưa trước đó chưa.

After updating bios, the main board shows a ram error light. Tried removing the ram but failed. ^^

[image]
[image]

Koekieezz, 2025-11-30:
Have you tried Clear Cmos after flashing? Because the bios is the same as before, only dsdt gets patched as per image shown above...

Would you elaborate on how did you update the bios step by step? It's strange because the one i did with Z97-A, it was succesfull but i was using afuwinx64 v3.05.04 with /GAN.

xCuri0, 2025-11-30:
@choigamevui621 Use BIOS flashback to restore stock BIOS.

Koekieezz, 2025-11-30:
> It doesn't need DSDT mod, you can see by `External reference` that it gets the 4G decode range from BIOS.
> 
> The problem is likely that the `HswAbove4G` patch isn't getting applied.
@xCuri0 Well, HswAbove4G only applies to NBPei only to H97 Pro Stock Bios 2906, maybe the bios itself borked from stock?
[image]

xCuri0, 2025-11-30:
@Koekieezz it's only getting applied partially, the `AddMemorySpace` patch isn't getting applied. Will need to look into this.

xCuri0, 2025-11-30:
I updated `HswAbove4G.txt` to fix the patch not getting applied try and see if it works now.

xCuri0, 2025-11-30:
@Koekieezz There were never any issues with patches causing no POST, the only issue was that it wasn't applying because of small difference in assembly code.

choigamevui621, 2025-12-01:
> [@choigamevui621](https://github.com/choigamevui621)Đây là bios của bạn, bây giờ bạn có thể sử dụng an toàn rồi: [H97Pro_RebarFull.zip](https://github.com/user-attachments/files/23848248/H97Pro_RebarFull.zip)
> 
> Nếu bạn muốn sử dụng CH341A để flash bios này, vui lòng sao lưu bios hiện tại của bạn bằng CH341A và sử dụng FD44 Editor để chuyển Địa chỉ MAC, UUID, v.v. Nếu bạn không biết cách chuyển, bạn có thể gửi bios đã sao lưu bằng CH341A bằng cách nhắn tin riêng cho tôi.

I thought updating bios was normal, why do I need FD44 ?

xCuri0, 2025-12-04:
@choigamevui621 That file has pad file issue so it's not unusual it doesn't work.

Original (proper pad file) on right
[image]

You can try this file I made using pad file workaround with MMTool 4.50.0.23, I checked the pad file and it matches the original.

All UEFIPatches including USB 3.0 fix included along with ReBarDxe and NvStrapsReBar 0.3. Make sure to use the 0.3 exe for NvStrapsReBar when configuring it.

[h97-pro-asus-2906-rebar-nvstrapsv3.zip](https://github.com/user-attachments/files/23919369/h97-pro-asus-2906-rebar-nvstrapsv3.zip)

Flash using BIOS menu, not hardware flasher. You might need to rename the file after extracting if it doesn't get detected.

choigamevui621, 2025-12-04:
> > [h97-pro-asus-2906-rebar-nvstrapsv3.zip](https://github.com/user-attachments/files/23919369/h97-pro-asus-2906-rebar-nvstrapsv3.zip)
> > Flash bằng menu BIOS, không phải flash phần cứng. Bạn có thể cần đổi tên tệp sau khi giải nén nếu hệ thống không phát hiện được.
> 
> Sử dụng menu bios/flasher bên trong thường không có tác dụng trên modified.cap.
> 
> Thao tác này sẽ hoạt động (hiện không còn vấn đề gì về pad nữa): [H97Pro_RebarFull.zip](https://github.com/user-attachments/files/23919485/H97Pro_RebarFull.zip)

[image]
2 file same name ?

@xCuri0 I have to use NvStrapsReBar v0.3 and not v0.4 right ?

Koekieezz, 2025-12-04:
> 2 file same name ?

Yes, apparently i sent the wrong bios yesterday. This one should be the correct one, im sorry for my mistake 🙏

My mod bios version use NvStrapsReBar v0.4 btw.

Koekieezz, 2025-12-05:
@wwe2kx I think TPM 2.0 is not supported by asus, only select motherboard got a bios update for tpm 2.0 support.

But you could try to ask asus support for a new bios, try request them for tpm 2.0 support along with pkfail mitigation. I did the same to gigabyte eSupport but more on PKFail mitigation for Z77 HD3, while they rejected me 2 times, 3rd time's a charm (i also asked chatgpt to rephrase my words), and i did get a beta bios for it 😂.

If you did get the bios (with tpm 2.0 support, and maybe also include PKFail mitigation for secure boot keys), send it here, i might re do your bios.

Koekieezz, 2025-12-10:
@choigamevui621 How is it going with the last bios i gave to you?
@wwe2kx did asus replied you yet?
@kalimdore-gif here is your bios for Maximum IX Code: [M9C_Complete.zip](https://github.com/user-attachments/files/24068227/M9C_Complete.zip)

All paddings are preserved, structure remains the same. Downgrade bios using ez flash just 1 version below your current version, and continue to update with this bios using Asus Bios Flash Back (just partition 2GB in your flash drive and format it as fat32 so that it'll work. Please Test if iGPU works under both CSM and pure UEFI, and post your results here (if everything successful, can i use your photo to re post the bios in win-raid?).

- Normal CoffeeLake cpus (not engineering and qualifying samples, not xeons) are possible with this bios, just follow these pinmod:
[image]

[15 more replies]