xCuri0/ReBarUEFI #99: Z77X-UD5H 4G decode does not activate

fabriguzden, 2023-11-21:
[image]
[image]

I just checked to see if 4G Decode was enabled and realized it is not, although GPU-Z shows it is active and the step by step instructions are correct, my specs are
i7 3770
Bios F16K Z77X-UD5H
16GB of ram at 1300mhz
Rtx 3060 evga 12gb LHR

When I try to upload 2GB in BAR my PC does not start, it gives me code 95 which means "Check PCI device requested resources"

Any suggestions?

fabriguzden, 2023-11-21:
> [image] [image]
> 
> I just checked to see if 4G Decode was enabled and realized it is not, although GPU-Z shows it is active and the step by step instructions are correct, my specs are i7 3770 Bios F16K Winraid Mod Z77X-UD5H 16GB of ram at 1300mhz Rtx 3060 evga 12gb LHR
> 
> When I try to upload 2GB in BAR my PC does not start, it gives me code 95 which means "Check PCI device requested resources" I also get this "If you get an error about GUID mismatch"
> 
> Any suggestions?

fabriguzden, 2023-11-21:
I realized that I was making mistakes with the commands when enabling 4G decode but now I have a problem and that is that when loading Windows it hangs on the loading wheel and the board sends me an A0 code that is meant to have started correctly but still hangs during boot

xCuri0, 2023-11-22:
@fabriguzden 

> I realized that I was making mistakes with the commands when enabling 4G decode but now I have a problem and that is that when loading Windows it hangs on the loading wheel 

did you do DSDT patch and UEFIPatch ?

fabriguzden, 2023-11-22:
> @fabriguzden
> 
> > I realized that I was making mistakes with the commands when enabling 4G decode but now I have a problem and that is that when loading Windows it hangs on the loading wheel
> 
> did you do DSDT patch and UEFIPatch ?

[image]

When I had the problem of the boot in Windows that freezes until that moment I have not applied the DSDT patch, I only applied the UEFI Patch, now a few moments ago the idea of applying the DSDT patch occurred to me because of what it says in your Wiki about DSDT, however I have problems decompiling the DSDT.aml to dsl, iasl shows me 6 uncorrected warnings, it seems decompilation depends on 6 external controls or a declaration in a refs.txt file, but, I leave it as is, I ignore it and continue, after having changed the code to the new one according to the instructions on your wiki and when I am going to compile

> `DSDTMod.dsl   1963:         Device (PCI0)
Warning  3073 -        Multiple types ^  (Device object requires either a _HID or _ADR, but not both)

DSDTMod.dsl   2371:                     CreateDWordField (BUF0, \_SB.PCI0._Y0F._LEN, MSLN)  // _LEN: Length
Warning  3128 -                                    ResourceTag larger than Field ^  (Size mismatch, Tag: 64 bits, Field: 32 bits)

DSDTMod.dsl   3550:                         Acquire (MUT0, 0x0FFF)
Warning  3130 -                                                ^ Result is not used, possible operator timeout will be missed

DSDTMod.dsl   3860:                     Method (DCR4, 2, NotSerialized)
Remark   2146 -     Method Argument is never used ^  (Arg1)

DSDTMod.dsl   4031:                 Method (SIOW, 1, NotSerialized)
Remark   2146 - Method Argument is never used ^  (Arg0)

DSDTMod.dsl   4047:                 Device (PS2K)
Warning  3141 -            Missing dependency ^  (Device has a _PRS, missing a _SRS, required)

DSDTMod.dsl   4113:                 Device (PS2M)
Warning  3141 -            Missing dependency ^  (Device has a _PRS, missing a _SRS, required)

DSDTMod.dsl   7120:                         Name (UPCA, Package (0x04)
Remark   2173 -                                     ^ Creation of named objects within a method is highly inefficient, use globals or method local variables instead (\_SB.PCI0.EHC1.HUBN.PR01._UPC)

DSDTMod.dsl   7132:                         Name (PLDP, Package (0x01)
Remark   2173 -                                     ^ Creation of named objects within a method is highly inefficient, use globals or method local variables instead (\_SB.PCI0.EHC1.HUBN.PR01._PLD)

DSDTMod.dsl   7148:                             Name (UPCP, Package (0x04)
Remark   2173 -                                         ^ Creation of named objects within a method is highly inefficient, use globals or method local variables instead (\_SB.PCI0.EHC1.HUBN.PR01.PR11._UPC)

DSDTMod.dsl   7160:                             Name (PLDP, Package (0x01)
Remark   2173 -                                         ^ Creation of named objects within a method is highly inefficient, use globals or method local variables instead (\_SB.PCI0.EHC1.HUBN.PR01.PR11._PLD)

DSDTMod.dsl   7177:                             Name (UPCP, Package (0x04)
Remark   2173 -                                         ^ Creation of named objects within a method is highly inefficient, use globals or method local variables instead (\_SB.PCI0.EHC1.HUBN.PR01.PR12._UPC)

DSDTMod.dsl   7189:                             Name (PLDP, Package (0x01)
Remark   2173 -                                         ^ Creation of named objects within a method is highly inefficient, use globals or method local variables instead (\_SB.PCI0.EHC1.HUBN.PR01.PR12._PLD)

DSDTMod.dsl   7206:                             Name (UPCP, Package (0x04)
Remark   2173 -                                         ^ Creation of named objects within a method is highly inefficient, use globals or method local variables instead (\_SB.PCI0.EHC1.HUBN.PR01.PR13._UPC)

DSDTMod.dsl   7218:                             Name (PLDP, Package (0x01)
Remark   2173 -                                         ^ Creation of named objects within a method is highly inefficient, use globals or method local variables instead (\_SB.PCI0.EHC1.HUBN.PR01.PR13._PLD)

DSDTMod.dsl   7235:                             Name (UPCP, Package (0x04)
Remark   2173 -                                         ^ Creation of named objects within a method is highly inefficient, use globals or method local variables instead (\_SB.PCI0.EHC1.HUBN.PR01.PR14._UPC)

DSDTMod.dsl   7247:                             Name (PLDP, Package (0x01)
Remark   2173 -                                         ^ Creation of named objects within a method is highly inefficient, use globals or method local variables instead (\_SB.PCI0.EHC1.HUBN.PR01.PR14._PLD)

DSDTMod.dsl   7261:                             If ((Arg0 == ToUUID ("a5fc708f-8775-4ba6-bd0c-ba90a1ec72f8") /* Unknown UUID */))
Remark   2184 -                                                                      Unknown UUID string ^

DSDTMod.dsl   7263:                                 Name (_T_0, Zero)  // _T_x: Emitted by ASL Compiler, x=0-9, A-Z
Remark   2011 -               Use of compiler reserved name ^  (_T_0)

DSDTMod.dsl   7308:                             Name (UPCP, Package (0x04)
Remark   2173 -                                         ^ Creation of named objects within a method is highly inefficient, use globals or method local variables instead (\_SB.PCI0.EHC1.HUBN.PR01.PR15._UPC)

[... cut ...]
DSDTMod.dsl  12427:                 Name (_T_1, Zero)  // _T_x: Emitted by ASL Compiler, x=0-9, A-Z
Remark   2011 -                             ^ Use of compiler reserved name (_T_1)

DSDTMod.dsl  12572:         Name (_HID, "ABCDEFGH")  // _HID: Hardware ID
Error    6035 -                                 ^ _HID suffix must be all hex digits (GH)

ASL Input:     DSDTMod.dsl -  426835 bytes   5395 keywords      0 source lines

Compilation failed. 2 Errors, 16 Warnings, 101 Remarks
No AML files were generated due to compiler error(s)`

2 errors that do not allow me to compile
[image]
[image]

xCuri0, 2023-11-22:
@fabriguzden it says at the bottom of the page how to fix these errors.

once you have modified dsdt and uefipatch resizable bar will work without issues. multiple other users (including me) have it working perfectly on gigabyte lga1155 motherboards

closed 2023-11-22 by xCuri0

fabriguzden, 2023-11-22:
I came up with the idea of using AI to correct the code, it has resolved those two errors, let's see if the patch works for me now

fabriguzden, 2023-11-22:
> @fabriguzden it says at the bottom of the dsdt wiki page how to fix these errors.
> 
> once you have modified dsdt and uefipatch resizable bar will work without issues. multiple other users (including me) have it working perfectly on gigabyte lga1155 motherboards

ooou I hadn't realized xd, I honestly don't know how to program code, that's why it occurred to me to seek help

fabriguzden, 2023-11-22:
> @fabriguzden it says at the bottom of the dsdt wiki page how to fix these errors.
> 
> once you have modified dsdt and uefipatch resizable bar will work without issues. multiple other users (including me) have it working perfectly on gigabyte lga1155 motherboards

Sorry, but I already applied the two patches, but the problem continues, the Windows loading bar and the Gigabyte logo freeze, it starts loading and then freezes after a while.

xCuri0, 2023-11-22:
@fabriguzden can you upload your modified bios ?

fabriguzden, 2023-11-22:
> @fabriguzden¿Puedes subir tu BIOS modificada?

https://github.com/xCuri0/ReBarUEFI/issues/11#issuecomment-1822265060

fabriguzden, 2023-11-23:
@xCuri0 
[Z77XUD5H ReBar Mod + NVME Support (4G Decode not working).zip](https://github.com/xCuri0/ReBarUEFI/files/13445354/Z77XUD5H.ReBar.Mod.%2B.NVME.Support.4G.Decode.not.working.zip)

xCuri0, 2023-11-23:
@fabriguzden this bios isn't dsdt patched, that's why it's not working
[image]

xCuri0, 2023-11-23:
this is what a proper patched dsdt looks like 
[image]

fabriguzden, 2023-11-23:
@xCuri0 That's strange, if I have applied the dsdt patch, well, way, I'll try it again.
edit: I just enabled the hidden menu with hidden options like Above 4G Decode and I'm wondering if with the hidden option already shown and activated in the BIOS menu it will no longer be necessary to use Grub

xCuri0, 2023-11-23:
Yes if the menu shows you don't need grub

fabriguzden, 2023-11-23:
@xCuri0 Thank you very much, ReBar with Above 4G Decode to 16GB ReBar is working correctly, I think I had made the mistake when importing the patch to UEFI Tools, it must be that I added the one that was not Patched and I had not realized

Flyw4vy, 2023-11-23:
> @xCuri0 Thank you very much, ReBar with Above 4G Decode to 16GB ReBar is working correctly, I think I had made the mistake when importing the patch to UEFI Tools, it must be that I added the one that was not Patched and I had not realized

Upload your latest BIOS please if you have it working now please m8.