xCuri0/ReBarUEFI #100: DSDT Patching - Skylake

UnidentifiedTag, 2023-11-24:
**System**
 - Motherboard: Z170A GAMING M7 (MS-7976)
 - BIOS Version: American Megatrends Inc. 1.L0, 28/06/2018
 - GPU: GTX 1660 SUPER
 - [✅ ] CSM is turned off
 - [✅ ] 4G decoding is enabled
 - [✅ ] UEFIPatch is applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [✅] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**

Describe the issue here.

at step (Extracting DSDT (Skylake and up))
i use command iasl DSDT.aml with out error or problem, and can read the file and see it in human language the problem comes when trying to recompile it back, i get 10 errors with a whole list of warnings and comments which prevents a file being made, if i add the extra entries that i am being told need to be added i end up with 13 errors, so the problem gets worse, im not sure where im going wrong or if the guide is just to old now ?
[E7976IMSmodifiedWithResizeBar2.zip](https://github.com/xCuri0/ReBarUEFI/files/13461022/E7976IMSmodifiedWithResizeBar2.zip)

xCuri0, 2023-11-24:
@UnidentifiedTag it says only ASRock Z170 / LGA1151 need DSDT patching why are you doing it on MSI ?

closed 2023-11-24 by xCuri0

UnidentifiedTag, 2023-11-24:
> @UnidentifiedTag it says only ASRock Z170 / LGA1151 need DSDT patching why are you doing it on MSI ?

Because mines not showing large memory under the  device manager properties of my gtx 1660 super (resources) so I believe I need this patch as well. I'm also not seeing it under Nvidia control panel system information for gtx 1660 super, under heinfo64 it has a green tick next to resize bar supported but a cross on rebar state saying disabled.... The rebarstate.exe gives me 2 options 

0: disabled
Above 0: maximum bar size set to 2^x MB
32: unlimited bar size

After typing 32 I get writing value to rebarstate successful wrote rebarstate uefi variable

But still after reboot everything stays the same?

On the original bios I did step 1 and 2, and skipped 3 but I think 3 is needed as it doesn't seem to be working

UnidentifiedTag, 2023-11-24:
i've tried to by pass the errors but it still fails even with commands to ignore all the errors

`iasl -p dsdt.aml -tc -vw 2095 -vi DSDTog.dsl
DSDTog.dsl(3970) : warning  3073 - Multiple types (Device object requires either a _HID or _ADR, but not both)
DSDTog.dsl(4603) : error    6074 - Name already exists in scope (\_SB.PCI0._CRS.M2LN)

    Original name creation/declaration below: (\_SB.PCI0._CRS.M2LN)
    DSDTog.dsl(4597) :
DSDTog.dsl(4604) : error    6074 - Name already exists in scope (\_SB.PCI0._CRS.M2MN)

    Original name creation/declaration below: (\_SB.PCI0._CRS.M2MN)
    DSDTog.dsl(4598) :
DSDTog.dsl(4605) : error    6074 - Name already exists in scope (\_SB.PCI0._CRS.M2MX)

    Original name creation/declaration below: (\_SB.PCI0._CRS.M2MX)
    DSDTog.dsl(4599) :
DSDTog.dsl(4898) : remark   2146 - Method Argument is never used (Arg0)
DSDTog.dsl(4916) : remark   2146 - Method Argument is never used (Arg0)
DSDTog.dsl(5024) : remark   2146 - Method Argument is never used (Arg0)
DSDTog.dsl(5199) : warning  3130 - Result is not used, possible operator timeout will be missed
DSDTog.dsl(5445) : warning  3129 - ResourceTag smaller than Field (Size mismatch, Tag: 1 bit, Field: 8 bits)
DSDTog.dsl(5622) : remark   2089 - Object is not referenced (Name [IRQS] is within a method [DSR3])
DSDTog.dsl(5687) : remark   2146 - Method Argument is never used (Arg0)
DSDTog.dsl(5702) : warning  3141 - Missing dependency (Device has a _PRS, missing a _SRS, required)
DSDTog.dsl(5781) : warning  3141 - Missing dependency (Device has a _PRS, missing a _SRS, required)
DSDTog.dsl(5949) : remark   2173 - Creation of named objects within a method is highly inefficient, use globals or method local variables instead (\_SB.PCI0.RP09._DSM)
DSDTog.dsl(6194) : remark   2173 - Creation of named objects within a method is highly inefficient, use globals or method local variables instead (\_SB.PCI0.RP10._DSM)
DSDTog.dsl(6439) : remark   2173 - Creation of named objects within a method is highly inefficient, use globals or method local variables instead`

xCuri0, 2023-11-25:
It's giving errors saying already exists because there is no need for DSDT
patch on your MSI motherboard which I already said before.

UnidentifiedTag, 2023-11-25:
when i run UEFIPatch.exe i get the following after adding the resize module : 
patch: replaced 11 bytes at offset 763Dh B800000000100000004C3B -> B8FFFFFFFFFFFFFF004C3B
Image patched
when i use the attached image [https://github.com/xCuri0/ReBarUEFI/files/13461022/E7976IMSmodifiedWithResizeBar2.zip](https://github.com/xCuri0/ReBarUEFI/files/13461022/E7976IMSmodifiedWithResizeBar2.zip) which has steps 1 and 2 done to it, i see the following : 
[image]

i guess youre right, but when i ask chatgpt it says its supported

Yes, the GTX 1660 Super supports Resizable BAR. [This feature can improve the performance of the graphics card from 10% to 20% in many games](https://www.nvidia.com/en-us/geforce/forums/geforce-graphics-cards/5/455474/resizable-bar-on-gtx-1660-super/)[1](https://www.nvidia.com/en-us/geforce/forums/geforce-graphics-cards/5/455474/resizable-bar-on-gtx-1660-super/). You can check whether Resizable BAR is active through the Nvidia Control Panel. Click “System Information” in the bottom-left corner and you should find it listed among your GPU details. [If marked “Yes”, it will take effect in supported games, and if it’s marked “No”, it’s currently disabled](https://www.nvidia.com/en-us/geforce/forums/geforce-graphics-cards/5/455474/resizable-bar-on-gtx-1660-super/)[1](https://www.nvidia.com/en-us/geforce/forums/geforce-graphics-cards/5/455474/resizable-bar-on-gtx-1660-super/).

xCuri0, 2023-11-25:
@UnidentifiedTag GTX 1660 and other Turing GPUs **DO NOT support Resizable BAR**, ChatGPT is wrong. HWInfo only shows Support because it supports downsizing it to 64Mb from 256Mb

If you are interested in some experimental ways that may or may not work see https://github.com/xCuri0/ReBarUEFI/discussions/89