xCuri0/ReBarUEFI #103: Problem that bios is in exe format

mojarrah, 2023-12-10:
**System**
Alienware a51 R4 pc
 - Motherboard: ms-7a87
[image]

 - BIOS Version: 2.0.7
 - GPU: 3090

Hey. Hope you are doing great. I wanted to follow your steps to activate rebar. but I was surprised that dell bios is .exe and not .bin

How can we fix it?

This is the bios link

https://dl.dell.com/FOLDER08945518M/1/Alienware_Area_51_R4_2.0.7.exe

xCuri0, 2023-12-10:
On most prebuilts usually the only way to flash is through use of a hardware flasher like CH341A

@Electron-PSO might know more about this

closed 2023-12-10 by xCuri0

mojarrah, 2023-12-10:
Could be talk on discord? whats your username

Electron-PSO, 2023-12-10:
Hello @mojarrah 

You can extract the .rom file from the EXE file by using the /writeromfile switch at the command line

Alienware_Area_51_R4_2.0.7.exe /writeromfile

Once you have the file, you should be able to use the AMI AFUWin64 v3.05.04 with GAN support  to flash it.

As xCuri0 said, make backups first though

AFUWINx64.EXE /O      to save current ROM image to file

mojarrah, 2023-12-10:
@Electron-PSO Got it, but could we discuss it on Discord now? whats your username? mine uchihamj

mojarrah, 2023-12-10:
@Electron-PSO here or there, Use what's best for you. The problem is that any version of the bios I can't even open the exe file normally when clicking it in windows 11. Any version of the bios. It gives an error unable to load driver!!! Can you try please to extract it on your side and upload it. That would be l insanely helpful!!!

Electron-PSO, 2023-12-11:
Hello @mojarrah,

Here is the rom file extracted from the 2.0.7 exe you posted

[alienware_area_51_r4_2.0.7.rom.zip](https://github.com/xCuri0/ReBarUEFI/files/13632962/alienware_area_51_r4_2.0.7.rom.zip)

Did you use the command line when you tried to extract it?  Like this?

[image]

mojarrah, 2023-12-11:
Hello @Electron-PSO 

Can confirm the ROM is working. Thanks

Moreover, I have written the same command but didn't work on the machine. It is Windows 11. Maybe it needs to be on Windows 10 to be extracted

Plus, can you please now in the form of steps, tell me exactly what I should do now(anything other than the rebar steps since they are on the site) and if you name any program send its link please

Electron-PSO, 2023-12-11:
Hello @mojarrah ,

After you have the ReBAR modified file, you'll need something like AMI AFUWin64 v3.05.04 with GAN support to flash it.

See this guide for instructions.  I'm not sure about Win11, but I have managed to get Dell Alienware machines to flash in windows before.

https://winraid.level1techs.com/t/guide-how-to-flash-a-modded-ami-uefi-bios/30627

mojarrah, 2023-12-11:
@Electron-PSO Downloaded the correct AMI version now

Can you mod the rom file with the uefirebar and upload it, please? since it is different from what other people have shown

Electron-PSO, 2023-12-11:
Hello @mojarrah ,

It shouldn't be that different.

Search for the DXE section and insert the ReBar ffs at the end of it?

mojarrah, 2023-12-11:
@Electron-PSO Look what it is giving me

[image]

mojarrah, 2023-12-11:
@Electron-PSO But it is giving this error when trying to flash

Electron-PSO, 2023-12-11:
P.S    I'm at work, so it is difficult for me to respond quickly.

mojarrah, 2023-12-11:
@Electron-PSO downloaded windows 10. Now. The problem was from windows 11

But when flashing, it gives an error

Rom file size doesn't match the bios file and that true

Rom file 16mb

.exe around 8mb

[image]

Electron-PSO, 2023-12-11:
Can you read the BIOS chip with the /O  option?

Electron-PSO, 2023-12-11:
AFUWINx64.EXE /O to save current ROM image to file

mojarrah, 2023-12-11:
@Electron-PSO [image]

It is showing this when writing the command

Electron-PSO, 2023-12-11:
You need to specify the file name you want to make

AFUWINx64.EXE backup.rom /O

mojarrah, 2023-12-11:
Did and it said it's done. I restarted then used rebar program to to put 32 value. It said it did success. Restarted again but it still shows as disabled in gpuz 

[image]

mojarrah, 2023-12-11:
@Electron-PSO look

[image]

xCuri0, 2023-12-11:
Make sure you have updated vbios on 3090

NVIDIA put out an update for it that's required for resizable bar to work

mojarrah, 2023-12-11:
@Electron-PSO @xCuri0 

vbios and everything updated
[image]

mojarrah, 2023-12-11:
Literary the only option saying no in gpu z about rebar is: enabled in bios: no

mojarrah, 2023-12-11:
@xCuri0 what do you think?

mojarrah, 2023-12-11:
@Electron-PSO I used /O command again to check if the bios really flashed. And put it in uefi tool. And it's still the same. It didn't flash the modded bios

Electron-PSO, 2023-12-11:
Did you run  ReBarState.exe  to set the BAR size?

mojarrah, 2023-12-11:
@Electron-PSO Yes

mojarrah, 2023-12-11:
@Electron-PSO also,

I used /O command again to check if the bios really flashed. And then I did put it in uefi tool. And the result were the bios didn't change. It didn't apply the modded bios. it's still the same. It didn't flash the modded bios

mojarrah, 2023-12-12:
@Electron-PSO Is there a way to force it 100% to flash?

xCuri0, 2023-12-12:
Maybe you need to set service jumper or something for it to flash

But best and safest method for flashing prebuilts is using a hardware
flasher like Ch341a

mojarrah, 2023-12-12:
@xCuri0 Damn, I don't have a hardware flasher currently. Is their another way/software I can try

Electron-PSO, 2023-12-12:
Would ReBarState be happy if the flash hadn't worked?

mojarrah, 2023-12-12:
Didn't get what you mean @Electron-PSO

Electron-PSO, 2023-12-12:
Did you use the /GAN command to flash the BIOS file?

It might not be needed, but worth a try.

AFUWINx64.exe modified.rom /GAN

mojarrah, 2023-12-12:
Yes

Electron-PSO, 2023-12-12:
If ReBarState couldn't set the variable it should error?

Try setting it to 1GB bar size

Then reboot

Then check with CPU-Z

mojarrah, 2023-12-12:
Tried but didn't work also!!

Electron-PSO, 2023-12-12:
OK

Ignore that last post

I see it does create and is happy with the variable on systems without the BIOS mod

Electron-PSO, 2023-12-12:
So ReBarState shows it is set to some GB?

mojarrah, 2023-12-12:
I set it to 10, which is 1gb(as far as I now) and it says successful. But still says disabled after reboot

[image]

xCuri0, 2023-12-12:
> Would ReBarState be happy if the flash hadn't worked?

Yes all it does is write a UEFI variable

mojarrah, 2023-12-12:
So what to do?

Electron-PSO, 2023-12-12:
I'm still finding it hard to believe the flash program can erase, program and verify without it actually doing it.

Not sure what to do now.

You could try a different flashing program or you could just do it the physical way.

If I had the board on my desk I would do this:

- Unsolder the chip  
- Put it in a programmer
- Read and save the current file
- Erase the chip
- Program it with the new file
- Resolder it

But that depends on your ability to do the job.

mojarrah, 2023-12-12:
@Electron-PSO 
Do you recommend another flashing program? What is it? 

Also, is their somehow a bios settings that is activated that didn't allow it to be flashed

Btw, when it flashed it didn't reboot by itself like I would expect it to happen, it just closed the cmd pop up. I rebooted manually

Electron-PSO, 2023-12-12:
For some Dell machines, I have used a Windows GUI version of the AMI flash tool

Electron-PSO, 2023-12-12:
I'll have a look for it

Electron-PSO, 2023-12-12:
Here is the official link to the AMI flash utilities

https://www.ami.com/bios-uefi-utilities/

There are Win GUI versions in the zip folders

x64

mojarrah, 2023-12-12:
There are multiple links for download 

Drop the link for the exact one I should use

mojarrah, 2023-12-12:
@Electron-PSO got it now and opened successfully The correct version

Moreover, what settings I should check to make it work 100%

[image]

mojarrah, 2023-12-12:
@xCuri0 I flashed it successfully now using fpt win32 version as winpad site suggests. The problem now is if I activate the rebar and reboot screen no longer work. The system boot but their is no screen. I have to reset the bios using the battery to get it working. What is the solution?

Electron-PSO, 2023-12-12:
Apologies for the delay,

I was travveling home and had some sleep.

It should be the Aptio V version

Electron-PSO, 2023-12-12:
Try a small BAR size first

1GB

Electron-PSO, 2023-12-12:
Then check in GPU-Z

mojarrah, 2023-12-12:
What number to put in rebar to equal 1gb? @xCuri0 @Electron-PSO

Electron-PSO, 2023-12-12:
It is 2^x        so for 1GB  set it to 10

mojarrah, 2023-12-12:
@Electron-PSO it worked!!!!

Now I can set it to the maximum of what my gpu can handle 

It's a 3090 24gb vram

Electron-PSO, 2023-12-12:
You'll need 15  as the bar size to give you 32GB.  That will be enough to fit the 24GB VRAM

mojarrah, 2023-12-12:
What's the best value?

Electron-PSO, 2023-12-12:
Which Intel FPT did you use?

Have a link to it also?

I'm helping someone with a system and they have flashing issues too

mojarrah, 2023-12-12:
Will send the link after the pc boot from pc bios reset

Moreover, it isn't working with anything above 1 or 2gb

@xCuri0 

Tried multiple values, all of them led to failure!! What should I do?

[97 more replies]