xCuri0/ReBarUEFI #106: I got issues with Resizable BAR enabled in BIOS is No

phoang0693, 2023-12-16:
**System**
 - Motherboard: 212B (HP Z440)
 - BIOS Version: 02.61
 - GPU: RX 6600M
 - [Y] CSM is turned off. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [ I turn MMIO, 64-bit in BIOS ] 4G decoding is enabled. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [N] UEFIPatch is applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [Y] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**
Resizable BAR enabled in BIOS: No

Describe the issue here.
I got issues with Resizable BAR enabled in BIOS is No
I run ReBarState and put 32.

[image]

xCuri0, 2023-12-17:
Have u flashed modified BIOS ?

phoang0693, 2023-12-17:
> Have u flashed modified BIOS ?

No, I got trouble to mod in UEFItool and MMtool.

phoang0693, 2023-12-17:
I can't find GUID in UEFItool and cant find Pcibus in MMtool

xCuri0, 2023-12-17:
@phoang0693 ask other HP prebuilt users for help and if it still doesnt work then i will re open this

closed 2023-12-17 by xCuri0