xCuri0/ReBarUEFI #107: ASUS Z270G + ARC750 - is not working

yargun, 2023-12-18:
**System**
 - Motherboard: ASUS Z270G
 - BIOS Version: 1501
 - GPU: Arc 750
 - [X] CSM is turned off. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [X] 4G decoding is enabled. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [X] UEFIPatch is applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [X] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**
I did everything and still cannot get it to work
1. I updated the bios according to instructions using UEFITool
2. I ran the UEFIPatch
3. Flashed the modified bios after extracting the body with /GAN 
4. Changed mbr -> gpt , cms off, 4GB ON (Device manager shows large memory for pci bus after the change)
5. All ok to that point
6. With rebarstate set to 8, 12, 14, 32 - I get the below: 
7. GPU is NOT working, only iGPU is giving display (showing error 12 - resource collision)
8. Tried to connect 2 monitors on GPU and iGPU - same behavior
9.  rebarstate  set to 0 - GPU is showing display again - no error 12 - no rebar
10. Any suggestions??? 
-Rafi.

yargun, 2023-12-18:
Also - no pad issue was found and DSDT patch not required

xCuri0, 2023-12-19:
rebarState not working at even 8 is weird, because that is same as 256mb default.

Try re doing everything with MMTool, common issue on ASUS motherboards for stuff to break when using UEFITool/UEFIPatch. See https://github.com/xCuri0/ReBarUEFI/issues/98#issuecomment-1826386025 if you need help doing it

yargun, 2023-12-19:
I saw that someone with exact same board wrote that it worked for me according to instructions, so its indeed weird
I will give it a try using the MMTool

xCuri0, 2024-01-13:
Update ?

takem94, 2024-01-13:
@yargun I have a very similar situation, but as exception works also values 7 and 8. For some reason it seems to me that the problem is in the GPU, I’m almost sure that it would work with another video card, have you tried it by chance?

closed 2024-02-09 by xCuri0