xCuri0/ReBarUEFI #108: Can't change bar size

Toyo-tez, 2023-12-21:
**System**
 - Motherboard: B560M DS3H V2 rev 1.0
 - BIOS Version: F4 mod (No patches can be applied to input file)
 - GPU: RX 580 2048SP with rx 570 bios 4GB
 - [ x] CSM is turned off. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [x ] 4G decoding is enabled. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [ ] UEFIPatch is applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [ x] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**

I have managed to mod the bios with rebar and now the rebar is working at 4GB, but according to the wiki (https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes)#how-do-i-enable-resizable-bar-on-unsupported-amd-gpus-) i should downgrade my bar to 2GB because my gpu is unsupported, but when i change the bar size with ReBarState and i reset, the bar size is still 4GB.
[image]

xCuri0, 2023-12-21:
@Toyo-tez Disable Resizable BAR in your BIOS menu, thats the only thing that can cause an issue like this.

closed 2023-12-21 by xCuri0

Toyo-tez, 2023-12-21:
Thank you so much!