xCuri0/ReBarUEFI #110: 4GB BAR not working, not sure if i modded my BIOS correctly.

jeremie-78, 2024-01-03:
I have a 4GB RX 580, an i7 3770 on an Asrock P67 Extreme4 (BIOS L3.21A) and 16GB of RAM. Windows 11 23H2.

After adding the ffs module and using UEFIPatch, I noticed that an extra padding file appeared.
[image]
I wasn't sure if it that was because of the bug, as it is outside of the module the ffs is added in, but i used the MMTool workaround anyway, which did make the padding file disappear.
Then when patching the DSDT, the compiler raised an "Invalid object type for reserved name (FDE: found Package, Buffer required)"error, which isn't in the list of common errors. The error pointed to the line "Name (_FDE, Package (0x05)  // FDE: Floppy Disk Enumerate", which I changed to "Name (_FDE, Buffer (0x05)". Not sure if that is the correct way to solve this error. I also used the MMTool workaround for this step.
I then flashed the modified BIOS, ran ReBarState and set the BAR size to 4096. After rebooting, i noticed Windows got quite stuttery, the AMD control pannel woudn't open anymore, and no 3d application would launch, simply crashing to desktop, or displaying a vague error message along the lines of "d3d device unavailable". GPU-Z would still show a 4GB BAR size. Then when I set a BAR size of 2048, everything started working again. This is what it looks like in GPU-Z :
[image]
And in device manager :
[image]

Is this behaviour normal ? Or did I make a mistake ?

xCuri0, 2024-01-05:
In the wiki it says this is AMD driver bug

closed 2024-01-05 by xCuri0

jeremie-78, 2024-01-05:
OK, thank you. What about the compiler error and the padding file ? I'm still unsure what I did is correct.