xCuri0/ReBarUEFI #114: Only 1GB ReBarSize is available Gigabyte Z170-HD3-DDR3-CF

takem94, 2024-01-12:
**System**
 - Motherboard: Gigabyte Z170-HD3-DDR3-CF
 - BIOS Version: F21f (modded for Coffelake)
 - CPU: QQLT (ES i7 8700h)
 - GPU: inno3d RTX 4060
 - [X] CSM is turned off. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [X] 4G decoding is enabled. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [X] UEFIPatch is applied. 
 - [-] DSDT is not applied. 
 - [X] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**
I have a very specific case, since I use a “mutant” processor, this is QQLT (ES i7 8700h). Therefore, I am already doing the modification for the modified BIOS in coffetime. I'm afraid this is the main problem, but maybe someone understands better.
I've seen people successfully use rebar on the same motherboard or similar ones, but I can't reach anything larger than 1 gigabyte for ReBarSize.
With a value of 2 gigabytes, it seems that initialization beyond the BIOS does not occur at all. With 4 gigabytes or more, something seems to be happening, the OS system even boots, but no image is displayed on the GPU. There is no image during BIOS loading, as well as in Windows itself. When I'm using iGPU, the image is output and 8 GB and even 16 GB seems to work, but the iGPU itself does not support rebar, so this is not certain. 
I did the BIOS firmware using the fptw64 utility, the authenticity will not pass through the Q-flash, I think the problem may be that the wrong BIOS is being flashed, since there are two of them on this board? Another option is to try flashing it via efiflash, but when I do backup via Q-flash or through fptw64, I getting a BIOS with FFS ReBar modification and the patches are not reapplied.
Also, for some reason, the GPU shows 3 BAR sizes and the size changes only for BAR1. 
There is an impression that 4G decoding is not enabled, although both in the BIOS and in GPU Z it appears enabled; perhaps the BIOS itself is not working correctly.
[image]

xCuri0, 2024-01-13:
@takem94 this is all expected behavior. You need to switch your monitor from iGPU to the 4060 when using 16GB BAR and you will get output. The proper workaround for this is using a dummy HDMI/DP (you can buy this) on the iGPU

Only BAR1 is used for VRAM and can be resized on NVIDIA, for AMD it's BAR0. The other BARs serve other purposes and cannot be resized.

takem94, 2024-01-13:
@xCuri0 I tried different methods, I have 2 monitors, and getting the rtx 4060 to load with a larger ReBarState than 1 GB does not work, there is always error 12 in the device manager. I tried switching 1 monitor from iGPU to GPU, it also didn’t give an image, I tried turning off iGPU in the device manager, but it didn’t make it work. I saw it work for a person from Win-Raid under the nickname Adiel235, but it seems that in my case there is something else, I’m now almost sure that it’s the VBIOS or the X8 bus on the 4060.

takem94, 2024-01-13:
@ xCuri0 how big is the risk of ruin the BIOS by replacing the DsdtAsl module from another BIOS from another motherboard? It’s just that the version of the HD3 motherboard with DDR4 memory has a slightly newer BIOS, maybe it makes sense to remove this module from it? 
It seems to me that this condition is probably not fulfilled as expected.
[image]

xCuri0, 2024-01-14:
I don't see how this can fix this issue but you can try it

takem94, 2024-01-16:
> I don't see how this can fix this issue but you can try it
> […](#)

You was right, problem was not in DSDT module, and also both of them are identical. 
I found solution and it was in turned out to be 2 things, the incorrect display of the CSM in the UEFI-BIOS, but the more important is the initialization of the PCIE bus.
So, when I tried flash the BIOS using the efiflash utility in MS-DOS, I turned on CSM to see the flash-drive in boot option. While I was doing this, I noticed that the display of the BIOS logo during boot changed, it stopped stretching and was in the center of the screen. I realized that I had entered in real UEFI mode, but after reflashing the stretched logo returned again and I still don’t know how to remove this effect. But after that I started turning on CSM with all the options and then gradually turning everything off.
But the most important option turned out to be “IGFX” in the "Initial Display Output" item-list, before that I always had the PCIE1 bus selected. Usually I just turned on only the iGPU, but it turns out that until I forcibly set it as a priority during initialization, the GPU always loaded first. At the same time, it is not necessary to insert anything into the iGPU; I can easily boot into Windows without access to the BIOS. Sleep mode also works.

[image]

So, I consider separately in the topic of problems and solutions to indicate the full procedure for actions if Rebar of more than 1 gigabyte is not enabled on 1151 motherboards from Gigabyte.
1. Reflash bios using the efiflash 0.87 mod utility, since it is the only one can flash two BIOS-chips at once.
2. Load into BIOS and tirst turn "enable" CSM and turn "disable" 4g above decoding. ReBarState must be set to 0 before this!
3. Save these changes and boot into the BIOS again.
4. Now you need to disable CSM again in the BIOS and enable 4g above encoding.
5. Again save these changes and boot into the BIOS.
6. Now it is very important to find the "Initial Display Output" option in the peripherals section and choose "IGFX" in it. And also find the option with built-in GPU and enable it too.
7. Save everything again and reboot into the system.
8. Now it is preferable to switch the monitor wire to the iGPU.
9. Then you can finnaly set ReBarState to 13, 14 or 32. 
10. After rebooting, you can change the monitor wire in the GPU back.
11. Check the video output after loading in Windows, if there is no image output, switch back wire to iGPU and then try another value in ReBarState one by one with reloads.

As a bonus, here are the results of my tests of rebar in Dying Light 2. No big difference with 1GB vs 8GB, but noticeble without rebar. 
[image]

A small problem is that you need to leave the iGPU turned on, since in theory it consumes RAM and additionally creates interruptions and increases delays in the system and games.

closed 2024-01-16 by takem94

JGSphaela, 2024-09-24:
>     1. Reflash bios by using the "efiflash 0.87 mod" utility, not 0.65 version, since it is the only one can flash two BIOS-chips at once.

Hi @takem94 , I have the same issue that my A770 will report code 12 when ReBar >= 2GB: https://github.com/xCuri0/ReBarUEFI/issues/4#issuecomment-2371411410
I came across your issue and decide to try it since mine is also a gigabyte board. Is this step necessary? I always want to keep the backup bios untouched just in case if I break something.

takem94, 2024-09-25:
Hello @JGSphaela!
I'm sorry to hear that my instructions didn't help you.

Regarding the first point, I think you could try flashing only one BIOS ROM, but there's a chance that during memory initialization, as often happened to me, the backup BIOS version from the second chip might load. If you don't have a programmer, you could try flashing only the active flash memory.
The most challenging part is finding out in which case Rebar works. The thing is, I had a situation where I could boot through the iGPU with a discrete graphics card, and I didn't get an error until I tried to connect a monitor to the GPU. 

And how many monitors do you have? I have two, and one was connected to the graphics card, while the other was connected to the motherboard's iGPU. In general, I struggled with this problem for a very long time. Possibly more than a week, I knew it could work and therefore didn't stop trying. 

I wish you luck in finding the cause in your case. And if it's not just for curiosity's sake, I certainly hope you'll be able to upgrade soon your setup. I know it's difficult in our time, but it's worth it even on not very powerful graphics cards like mine.

JGSphaela, 2024-09-25:
@takem94 Thanks for your warm reply! I change my Intel A770 GPU to an AMD RX580 and it works. I updated right here: https://github.com/xCuri0/ReBarUEFI/issues/4#issuecomment-2371411410
It seems that Intel GPU is the issue here, I don't know if it's even fixable. Though many others have A770 working, but I never saw one with Gigabyte Z170.

I already trying to get this working for almost a week now... I will continue trying it for now!