xCuri0/ReBarUEFI #115: Gigabyte GA-Z170M-D3H with QQLS CPU modded bios can't boot after adding ReBarDxe.ffs

Igo131, 2024-01-15:
**System**
 - Motherboard: Gigabyte ga-z170m-d3h
 - BIOS Version: F22f (modded for QQLS cpu)
 - GPU: QQLS
 - GPU: rx6700
 - [x] CSM is turned off. Checked in bios.
 - [x] 4G decoding is enabled. Checked in bios.
 - [ ] UEFIPatch is not applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [x] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

Description

Hi. I have a modded bios file from QQLS supplier to work with this cpu. Added ReBarDxe.ffs in UEFITool_0.28.0. After flashing with q-flash or ch341 board can't load (bootloop). No image from internal/external gpu (somehow i managed to boot once but Rebar was disabled as well). Tried MMTool but can't find PciBus string there.

[image]

[Z170M-D3H F22f-ReBAR.zip](https://github.com/xCuri0/ReBarUEFI/files/13943254/Z170M-D3H.F22f-ReBAR.zip)
[Z170M-D3H F22f.zip](https://github.com/xCuri0/ReBarUEFI/files/13943253/Z170M-D3H.F22f.zip)

takem94, 2024-01-16:
I don’t quite understand, after flashing the BIOS with the ReBarState module, were you able to enter the OS or BIOS? I looked at your BIOS via coffeetime 0.99, there are a lot of unnecessary patches and microcodes and there is no patch for PCIE.

[image]

It may be worth rebuilding the BIOS, the ME version is probably too new, although it is turned off, it takes up a lot of space.

If you still manage to get into the system with the ReBar module and its value at 0, 8, 9 or even 10. It may be useful to look at my topic https://github.com/xCuri0/ReBarUEFI/issues/114 .

Igo131, 2024-01-16:
[takem94](https://github.com/takem94)
Thank you for the reply.

The issue was in the modded bios from qqls seller. Modded clean bios:

[image]
[image]

Resizeble Bar works now.

takem94, 2024-01-17:
@Igo131 you can do this patch: 
`# PciBus | Don't downgrade 64-bit BARs to 32-bit (Coffee Lake)
3C1DE39F-D207-408A-AACC-731CFB7F1DD7 10 P:A80274334038AF95:A802EB334038AF95`
But they all seem to be unnecessary. 

In theory, it should be possible to get into the BIOS via the iGPU, at least for me, I connect by hdmi cable.
With High Definition Audio Bus, as I understand it is the same problem for everyone https://github.com/xCuri0/ReBarUEFI/issues/4 or most, I use a separate sound card in any case, so it’s not critical for me.

Igo131, 2024-01-17:
[takem94](https://github.com/takem94) Seems like the patch is not needed or am I patching it wrong:

[image]

I saw you benchmarks with 1gb rebar. Maybe 1gb will be enouth for games performance boost if there is no fix for High Definition Audio Bus problem. What you think?

xCuri0, 2024-01-17:
@takem94 these are usual issue with LGA1151 and resizable bar, there's already an issue about this. It's weird considering older Sandy/Ivy Bridge and Haswell/Broadwell have zero issues like this.

closed 2024-01-17 by xCuri0

Igo131, 2024-01-17:
> @xCuri0 as i see there is no fix for now?
Anyway if someone will do some patches i can test it on my system (have ch341).

takem94, 2024-01-17:
@Igo131 I think it's very individual and depends on many factors. Firstly, I have a card from Nvidia and with a small VRAM size, which is almost always full of game data, so there is very little cache left for other memory operations and apparently 1 GB is the limit. Also important is the game itself and how it uses Rebar. You need to do tests in main games for yourself and decide worth it or not.

xCuri0, 2024-01-18:
@takem94 Nvidia driver only uses 1GB BAR in most games anyways unless you use NVIDIA Inspector to manually increase it. Sometimes it results in fps loss though

AMD always uses the full BAR size but even then in my experience the difference between 2GB and 8GB was negligible for me even tho 256MB vs 2GB was an almost 15% increase

takem94, 2024-01-18:
@xCuri0 yeah, thank you. I did not know about the limitation at the driver level and used Rebar only through the inspector, because with a value of 1 gigabyte, the driver panel indicates that Rebar is disabled. 
Also, on Intel cards without Rebar support, many games are on borderline of playability. I think for them your work cannot be overestimated. 
It's still hard to believe that this works on my old platform and even older ones. It’s a pity, of course, that new games are increasingly greedy for processor power, even CS2 on a full server in DM with an analogue i7 8700 I can get an average of 120 frames and 1 percent on 70. I played CS:GO and COD all my life and always had about 200-300 FPS with more bad CPUs. Now I hope to at least buy a Ryzen 7500f to get the same result.