xCuri0/ReBarUEFI #120: My ASUS P8Z68-M pro R.I.P

Daishuku, 2024-02-04:
**System**
 - Motherboard: ASUS P8Z68-M Pro
 - BIOS Version: 4003
 - GPU: 3060 Ti GDDR6X
 - [ Y ] CSM is turned off. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [ Y ] 4G decoding is enabled. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [ Y ] UEFIPatch is applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [ Y ] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

My PC using 3770K, I have follow the guide carefully, BIOS looks okay after the patching. But after flashing the BiOS, Only Fans is okay, everything unable to recover. The BIOS sounds is serious damaged. However the Chinese New Year coming, I'm hard to grab same motherborad to recover in week. And my Job data is lockdown at the moment... My PC never failure like this time, I gonna cry.

Daishuku, 2024-02-05:
Possible part cuase my black out: 

-DIST module modifying. Have some syntax error when compiling. Padding seems okay.

-BiOS patching. Removed some Patch.txt part this was not related Ivy bridge

xCuri0, 2024-02-05:
@Daishuku you can use ASUS USB flashback to restore BIOS

It's most likely pad file issue causing this which u didn't notice

Daishuku, 2024-02-05:
This borad wasn't modern enough, so didn't have any BIOS recover function. The only way is detach the chips and replace a health one.

Padding issues I have notice from the guide and double checked before I flash my borad.

Also, found a same model board to hot swap, I may try upload my broken BIOS rom to let you analysis.

Daishuku, 2024-02-05:
Here is the BIOS I prepared that kill my Board. 
[Broken_Bios.zip](https://github.com/xCuri0/ReBarUEFI/files/14170091/Broken_Bios.zip)

I guess I maybe NOT killed my board, but hard to confirm due to my old board ownership exchanged.
When I hot swap my board and black out at the start, found this was DRAM Timing issues due by BIOS "first run" ...
However, I can't take a Risk right now.

xCuri0, 2024-02-05:
You need to re do the guide using only MMTool on the final BIOS you're flashing.

I'm closing because this is a known with ASUS motherboards. I know the workaround for it is complicated to do but it's the only option till UEFITool gets a update to fix it

closed 2024-02-05 by xCuri0

Daishuku, 2024-02-06:
Get idea what's I have done stupidly, There have a lot of step get wrong.

Maybe prepare next try when back up board arrive after Chinese New Year