xCuri0/ReBarUEFI #121: ASUS H87I-PLUS Can't flash modified BIOS; Security verficiation failed.

qauff, 2024-02-05:
**System**
 - Motherboard: ASUS H87I-PLUS
 - BIOS Version: 2003 (2014-11-05)
 - GPU:
 - [x] CSM is turned off. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [?] 4G decoding is enabled. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [x] UEFIPatch is applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [x] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**
So my ASUS motherboard doesn't support BIOS Flashback, so I attempted to flash via the BIOS interface (ASUS EZ Flash 2 Utility v02.00). When I try to flash the modified BIOS however, I get a `Security verification failed.` error.

I'm not sure how to bypass this security feature. I've tried another method using the ASUS DOS bupdater.exe, but I get the same issue, and I can't even do a BIOS backup. I have Secure Boot set to `Other OS`, there's no Disabled state from what I can tell. 

Does this mean I'm SOL on getting Resizable BAR enabled on my motherboard?

Specifications: https://dlcdnets.asus.com/pub/ASUS/mb/LGA1150/H87I-PLUS/E8210_H87I-PLUS.pdf

Daishuku, 2024-02-06:
Some other guy told me, ASUS BIOS.rom has their own header that design for EZ Flash, need convert to .cap to remove header, maybe the the main factor cause UEFI tools ruin ASUS motherboard.

qauff, 2024-02-07:
> Some other guy told me, ASUS BIOS.rom has their own header that design for EZ Flash, need convert to .cap to remove header, maybe the the main factor cause UEFI tools ruin ASUS motherboard.

@Daishuku Ah so that means I'm probably SOL then?

xCuri0, 2024-02-08:
There is no reason for you to flash BIOS if you use Linux, all you have to do is the Linux DSDT fix where you add a QWordMemory entry which the wiki explains