xCuri0/ReBarUEFI #122: ASUS Z170M-E-D3 trouble to enable large memory.

ssvext, 2024-02-06:
**System**
 - Motherboard: ASUS Z170M-E-D3
 - BIOS Version: Z170M-E-D3-ASUS-2604
 - GPU: GIGABYTE RTX 3080 VISION OC Rev. 2.0
 - [+] CSM is turned off. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [+] 4G decoding is enabled. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [+] UEFIPatch is applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). 
 - [-] On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [+] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

Hello folks. Greetings from UA.
I have a problem.
1) Inserted ReBarDxe.ffs in my BIOS, using UEFITool.
2) Patched BIOS Using UEFIPatch with standart patches.exe (xCuri0/ReBarUEFI/blob/master/UEFIPatch/patches.txt)
3) Cheking for pad file issue (no pad`s in DXE volume at all)
4) Cooffee`moded BIOS with new microcodes, etc (using mutant qtj1)
5) Flash bios, using ch341.
6) After first boot turned off CSM, from moded grub changed setup_var 0x3 (setup_var 0x3 0x1).
7) Loading windows 11, and don`t have large memory in device manager (screenshot above)

When i choose in ReBarState.exe "1-7" - GPU-Z  says, that rebar is disabled in bios, but everything else is "yes"
When i choose in ReBarState.exe "8-16" - system freeze on boot with graphical artifacts on POST.
When i choose in ReBarState.exe "32" - system boot, but only from iGPU, Gpu-Z says "Rebar Enabled? 16gb". RTX3080 has error 12.

Trying to DSDT patching, but my bios dosnt have AmiBoardInfo, only AmiBoardInfo2, (no DSDT error, when trying to extract with AmiBoardInfoTool)

Reading all issues posts, reading hundered of pages on winraid.level1techs.com, not sleep at all 40hrs.
I realy want to solve this quest. or understand that this is impossible on my rig.
I understand that you are tired of stupid questions, but I hope for your support.
As bad luck - my MoBo not listed in "ReBar-worked mobo list" =(

[Original_n_modded_bios.zip](https://github.com/xCuri0/ReBarUEFI/files/14180994/Original_n_modded_bios.zip)
[image]
[image]
[image]

xCuri0, 2024-02-08:
You're most likely not enabling 4G decoding correctly. You need to set it to 0x0 and then 0x1 again for it to activate sometimes.

ssvext, 2024-02-08:
Thnx for answer.
But already tried.  0x0, reboot, 0x1. reboot.
Also tried with setup_var_1, setup_var_2, setup_var_vs.
No large memory in device manager =(

xCuri0, 2024-02-12:
can you send ifr extractor results file ?

ssvext, 2024-02-12:
I have two "above 4g" search results. Extract two of them. 
[1 IFR.txt](https://github.com/xCuri0/ReBarUEFI/files/14238535/1.IFR.txt)
[2 IFR.txt](https://github.com/xCuri0/ReBarUEFI/files/14238538/2.IFR.txt)

Thnx for help

xCuri0, 2024-02-12:
@ssvext you missed the option above 4gb mmio assignment.

Enable and it will work

closed 2024-02-12 by xCuri0