xCuri0/ReBarUEFI #127: asus formula ix issue

shikulja, 2024-02-07:
**System**
windows 11
 - Motherboard:
 asus formula ix
 - BIOS Version:
 M9F-1501
 - GPU: rx580 8gb
 - [y] CSM is turned off.
 - [y] 4G decoding is enabled. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [n] UEFIPatch is applied
 - [y] I have read

**Description**

everything is as usual, I modified the UEFI tool, did Bios Unlock (uefi) and flashed it via FPT.
I also tried bios unlock + flashrom (internal)

I also tried updating gop for sata, because the disks disappeared from loading, and unlocking regions through coffe time. There was no point in this, since rebaruefi does not work.
There are similar boards in the confirmed list. Therefore, I think that the problem is with the method of injection of the module into the firmware

[biosCLEAN_Unlock_rebar03.zip](https://github.com/xCuri0/ReBarUEFI/files/14200908/biosCLEAN_Unlock_rebar03.zip)

[image]

Rebar support in Radeon-ID also did not turn on, although I have been using these drivers for a long time and updating them, I have not noticed any other problems.

I'm still not sure what should work first. rebar support in bios or drivers

xCuri0, 2024-02-08:
Your ReBarState is set to 8 (256MB) , how are you expecting it to do anything when you have told it to set 256MB ?

closed 2024-02-08 by xCuri0

shikulja, 2024-02-08:
@xCuri0 
If I install more than 9 (512MB) the PC freezes completely. Only flashing helps. And as in the instructions, it is possible to remove the battery

shikulja, 2024-02-08:
@xCuri0  more than 512mb(9) pc hangs. only removing the battery helps as in the instructions

xCuri0, 2024-02-08:
@shikulja you say you haven't used uefipatch, without uefipatch that will happen

shikulja, 2024-02-08:
@xCuri0 
it was not necessary, as I thought. Only for specific conditions. Please tell me which patches to apply?

xCuri0, 2024-02-08:
Follow the guide again you do not need to manually select any patches just use the provided patches.txt