xCuri0/ReBarUEFI #141: PC in Power off - Power on Infinite Loop

IvanTheNewOne, 2024-02-10:
Hi guys, 
I have a Gigabyte GA-Z97P-D3 motherboard (Z97 chipset) with an i7-4790K, 2x8GB RAM and an RX 6700 XT.

I tried enabling Resizable BAR and setting it to 4GB, but my system wouldn't boot. I reset the CMOS and tried again, this time at just 1GB, and that worked: system booted normally, benched fine etc. 

Then I then tried 2GB and disaster struck. On power on, the PC just shuts down, then powers itself on again, then off and so on, infinitely. 

This time, clearing the CMOS (including leaving the PC unplugged and without the battery over night) didn't help. 

I tried disconnecting all unnecessary hardware (storage, discrete GPU, mouse...) and leaving just the CPU, a single stick of RAM, the keyboard and plugging the monitor into the integrated GPU - the same. The PSU passes the paperclip test - powers on and stays on, with correct values across all of the rails. 

The system usually shows only a blank screen and makes no sound, although sometimes it makes the initial beep and once in a while it does show the splash screen with "Press Del to enter Setup" etc, but even then does not respond to the keyboard input.

Help! :(

ZOXZX, 2024-02-10:
Switch to other bios chip, boot and make corrections to original bios.

IvanTheNewOne, 2024-02-10:
Hi ZOXZX and thanks for the reply.

The board does have a dual BIOS, but I don't think that there is a switch to go from one to another. I checked the board, the documentation and online resources I could find and all I found was that the backup BIOS should kick in if there is a problem with the main one. 

Online I read that the second BIOS should take over after a certain number of failed boots, that I could force it by holding power and reset or that, of all else fails, I could "disable" the first BIOS and force the board to use the second one of I short certain pins on the main one, but none of that succeeded (shorting the pins does cause the system to behave slightly differently, but it does not really seem to try to boot).

ZOXZX, 2024-02-11:
Is it possible that second bios is an early one and doesn't support your cpu?

IvanTheNewOne, 2024-02-11:
Hmm, I guess that it would be theoretically possible, but unlikely. Back in the day, I just bought the parts and put them together, so the main BIOS supported the CPU out of the box. I doubt that the motherboard would ship with two different BIOS versions. Although I do know that the main BIOS came with the very earliest BIOS that supported Haswell Refresh, so... maybe?! Not sure where to find a spare LGA1150 CPU.

ZOXZX, 2024-02-11:
4790k is supported from F5, first release was F3. It is unlikely to have two revisions on a same board, but who knows. It behaves like that from your descriptions? Did someone just readied the board for refresh? 
Guess you could find some second hand celerons. I have two of them and never owned a haswell systems :-)

IvanTheNewOne, 2024-02-11:
I can't remember, but I don't think that the box was opened when I bought it. 

By coincidence, I found an identical board second hand for a reasonable cost (around 35€ shipped), so I ordered it. Hopefully it works out.

xCuri0, 2024-02-12:
@IvanTheNewOne it looks like gigabyte dual bios recovery triggered and restored a corrupt second bios.

it looks like you tried 2gb BAR on a board without 4g decoding which made it not boot and dual bios recovery got triggered.

there are multiple ways to fix this if you search gigabyte power cycling issue