xCuri0/ReBarUEFI #191: EFI Error (at Create Timer) (0x01001202000000B4)

mrlmao, 2024-03-21:
**System**
 - Motherboard: H61M-DS2
 - BIOS Version:F7
 - GPU: MSI Gtx 1660 Super Gaming-x
 - [ ] CSM is turned off. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [ ] 4G decoding is enabled. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [ ] UEFIPatch is applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [ ] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**

Describe the issue here.

mrlmao, 2024-03-21:
[image]
I cant turn on rebar

mrlmao, 2024-03-21:
[image]

xCuri0, 2024-03-21:
Wrong repo open in NvStrapsRebar instead

Also 64GB BAR won't work and isn't needed for a 1660 u only need 8GB

closed 2024-03-21 by xCuri0