xCuri0/ReBarUEFI #192: Site needs updat

crashr, 2024-03-22:
Hi.
I think the info about ifrextract is outdated. I just used ifrextract on linux without any issues.
https://github.com/xCuri0/ReBarUEFI/wiki/Enabling-hidden-4G-decoding

MrAnonymous7122, 2024-03-23:
@xCuri0 I've updated the [wiki page](https://github.com/MrAnonymous7122/ReBarUEFI/wiki/Enabling-Hidden-4G-Decoding) but github unfortunately does not allow pull requests for wiki. My updated guide uses [IFRExtractor-RS](https://github.com/LongSoft/IFRExtractor-RS/releases/) instead of the archived [Universal IFR Extractor](https://github.com/LongSoft/Universal-IFR-Extractor/releases).  [IFRExtractor-RS](https://github.com/LongSoft/IFRExtractor-RS/releases/) also works on linux.

xCuri0, 2024-03-23:
@MrAnonymous7122 Thx, I've updated the wiki with it

closed 2024-03-23 by xCuri0