xCuri0/ReBarUEFI #201: Resizable Bar not enabled after patching

MrHaber, 2024-04-07:
**System**
 - Motherboard: Gigabyte GA-Z77-D3H
 - CPU: i5-3470
 - BIOS Version: F23b
 - GPU:
 - [V] CSM is turned off. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [V] 4G decoding is enabled. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [V] UEFIPatch is applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [V] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**
My 4G decoding offset above is "0x3". I set it to 0x0, before using ReBarState I set it to 0x1 and reboot. CSM in my BIOS mode: “Never”. When loading the OS, the Gigabyte logo is displayed in full screen mode. After fixing the UEFI and using Q-Flash, I started the system. But I noticed that after fixing UEFI, the Large Access Memory item disappeared in the device manager. Before the UEFI change, LAM was present in the device manager and showed a long message that did not look like something like: PCI-Bus, it was different. I ran ReBar state, and set the value to 13. After the reboot, absolutely nothing changed. GPU-Z showed that everything was “yes”, except for Resizable Bar is enabled in bios, this item was “no”. My video card: Intel Arc A770 (16 GB) latest drivers, I had a control panel installed which gave an error stating that the Resizable Bar is not enabled. The device manager did not contain the "Large access memory" item. Next, I tried to reboot several times, this did not give any effect, except that while the system was loading, the logo appeared and disappeared in less than a second, before changing the UEFI, the system boot took much longer, and the logo was displayed for 3-4 seconds. After reboots, I changed the ReBar State values, but this did not have any effect. I entered GRUB, and decided to "reboot", 4G above decoding. After the command setup_var 0x3 0x0, the system, as described in the article, did not boot, there was no image, and there was no POST(CMOS fixed). Could it be that I didn't perform the DSDT patch correctly?

xCuri0, 2024-04-07:
when you flash bios all settings get reset

from what you've written it looks like you forgot to disable csm.

and missing dsdt patch on gigabyte lga1155 will cause windows to freeze or bsod on boot logo

MrHaber, 2024-04-07:
> when you flash bios all settings get reset
> 
> from what you've written it looks like you forgot to disable csm.
> 
> and missing dsdt patch on gigabyte lga1155 will cause windows to freeze or bsod on boot logo

hi, it seems that I write in issues when I am one step away from solving a problem. 
CMOS seems to have helped. Serial 
`setup_var 0x3 0x0`
and 
`setup_var 0x3 0x1`
solved my problem. After the reboot, LAM appeared, as well as GPU-Z and the Intel Arc A770 driver write that ReBar is enabled. It seems so far so good.
[image]

MrHaber, 2024-04-07:
[image]

closed 2024-04-07 by MrHaber