xCuri0/ReBarUEFI #205: ReBar not working on Asus B450F.

Artasdmc, 2024-04-14:
[image]
Windows 11
 - Motherboard: Asus B450F
 - BIOS Version: 5404 (AGESA version to ComboV2PI 1.2.0.Ca)
 - GPU: Nvidia RTX 2070 Super
 - CSM is turned off. 
 - 4G decoding is enabled
 - UEFIPatch is applied

This motherboard does not support bios flashback, originally used AFUDOS to force a bios flash. Which in turn just bricked my motherboard. Now I'm back with modified bios which I flashed with CH341A, DXE Driver is included and recognized.

Motherboard has 2 options for ReBar (Off/Auto), no matter the setting ReBar is still not working. Rebar is set to 4GB with your tool.

[image]
[image]

xCuri0, 2024-04-14:
You have to use NvStrapsRebar on Turing, not ReBarUEFI

closed 2024-04-14 by xCuri0

Artasdmc, 2024-04-14:
My mistake, I'll be right back to confirm.

Artasdmc, 2024-04-14:
Just reset BIOS, enabled Rebar which disables CSM and enabled 4G decoding, and loaded in to windows. I'm getting an error that the driver has not been loaded, even though it was flashed with modified BIOS.

I'm attaching bios file I used to flash my motherboard. Does location matter where it is added?
It is there, just by searching nvs you can find it in uefitool.
[bios.zip](https://github.com/xCuri0/ReBarUEFI/files/14971788/bios.zip)

Artasdmc, 2024-04-14:
This board got ReBar support with BIOS 4007 which also added new CPUs with it. Could maybe roll back to that, but I'm giving away all security updates with it.