xCuri0/ReBarUEFI #210: ResizableBar still disabled after doing all the steps

BloodDrag0nS, 2024-05-08:
Windows 11

Motherboard: Asus B450-F Gaming II (BIOS FLASHBACK)
BIOS Version: 5404
GPU: Nvidia RTX 2060 6GB
CSM -> OFF
4G Decoding -> Enabled via BIOS
UEFIPatch not needed

I followed every step but GPU-Z still shows Resizablebar is disabled. Here are the steps i did

1. Downloaded plain BIOS
2. Used BIOS renamer to convert it into a useable BIOS
3. Inserted ReBarDxe.ffs to 2 volumes since i got 2 results
4. Saved the BIOS
5. Ran UEFIPatch to get an message called "No patches can be applied to input file"
6. Went to DSDT page
7. 4G Decoding is enabled (Attached a picture related to it maybe its gonna help or something)
[image]

8. Went Straight to flashing modified UEFI
9. Flashed the BIOS with an USB Stick via BIOS FLASHBACK
10. Enabled Above 4G Decoding in BIOs and disabled CSM, secure Boot was disabled
11. Used ReBarState set number to 10
12. Rebooted
13. ResizableBar state -> DISABLED

I have no idea why this is happening after following all the steps correctly.

BloodDrag0nS, 2024-05-08:
Everything was running smoothly without errors etc. Just the Resizablebar is disabled.

thoulegend, 2024-05-08:
I would think its because you chose the 4g decoding limit rather then 4g decoding enabled disabled below that perhaps

xCuri0, 2024-05-09:
you need nvstrapsrebar for turing gpus, rebaruefi will not work on them

closed 2024-05-09 by xCuri0

BloodDrag0nS, 2024-05-11:
Oh didnt See that there is a Turing Version of it. Thanks!