xCuri0/ReBarUEFI #212: 0x95 PCI Bus Request Resources hang during post with Tesla M40 on Asrock Z77 Extreme4

minelolpride, 2024-06-12:
**System**
 - Motherboard: Asrock Z77 Extreme4
 - BIOS Version: 3.00
 - GPU: NVIDIA Tesla M40 24GB
 - [X] CSM is turned off. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [X] 4G decoding is enabled. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [X] UEFIPatch is applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [X] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**

To preface, the target machine runs Windows, and I have done DSDT patching as described [here](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching#sandyivy-bridge-dsdt-patch), but the same symptoms come up without DSDT patching too.

I modded the BIOS for my motherboard as per the steps on this repo (Starting from inserting FFS module), but after flashing the machine will not finish POST as it sits (indefinitely?) on 0x95, which on my board, according to my manual, should be "PCI Bus Request Resources", though the manual is from 2013 and the BIOS I am modding is from 2018. I can go get an earlier BIOS (2.90 is the latest not tagged "Beta", and is from 2013), but am unsure if this is what had to be done for anyone else with this board that got it working.

When I take out the M40, the system boots totally fine.
I have a GTX 1650 lying around not doing much so I put it in instead of the M40 and that also booted just fine, though the BARs reported are BAR0:16MB BAR1:256MB BAR2:32MB both in GPU-Z and HWiNFO (hey, ReBAR is on), even though I set ReBarState to 15 for 32GB (darn). I took the ReBarState on this card down to 12 (4GB) to no avail (If testing that requires having a monitor plugged into the card in question, I don't have any monitors that take the card's outputs, and all my adapters are the wrong way around :sob:).

I am aware that having too much RAM on Ivy Bridge interferes with things, so I cut down to 8GB of system RAM. I am aware of #77 having a workaround but it looks like a linux specific thing? Haven't tried it yet...

I had utilized both UEFITool and MMTool on separate occasions to make the modifications to the BIOS when inserting ReBarUefi and modifying AmiBoardInfo, all pad-files over all attempts appear intact when compared to the original file.

I will attempt BIOS version 2.90 and see if that gets me any further...

(Also, every screenshot I have seen shows the Large Memory thing "Pci Bus" in Device Manager, but for me it says "PCI Express Root Complex". I don't know if this is complicating anything, but I am willing to re-install Windows on the target machine if maybe it's a driver from my post-install that's messing with it.)
[image]

minelolpride, 2024-06-12:
I tried this with BIOS version 2.90 and nothing changes, still hanging on 0x95 with the card, still booting without it.

minelolpride, 2024-06-13:
Went back to 3.00 and performed the workaround that I read as being only really useful on Linux, and the board POSTs fine. Got into Windows and am met with a Code 12. Also, both GPU-Z and HWiNFO display that ReBAR is still not supported. (I hear GPU-Z is pretty flaky with its readings, it also tells me that 4G decoding is on even though I disabled it)

xCuri0, 2024-06-13:
@minelolpride 

yes, UEFIPatch is required for these boards to POST with any BAR larger than 4GB.

Windows doesn't allocate PCIe resources by itself but if you really need it working on Windows you could write an EFI application to assign the GPU BAR address and run it before Windows loads.

xCuri0, 2024-06-13:
#45

closed 2024-06-13 by xCuri0