xCuri0/ReBarUEFI #214: ASUS Sabertooth X99 pad-file issue

gravufo, 2024-06-16:
**System**
 - Motherboard: ASUS Sabertooth X99
 - BIOS Version: 4101
 - GPU: AMD RX480
 - [x] CSM is turned off. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [x] 4G decoding is enabled. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [x] UEFIPatch is applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [x] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**

First off, thanks a lot for this repo and all the effort put into the wiki. It was really helpful!

I have successfully flashed a BIOS with the ReBarSmiDxe module as well as the NvramSmiDxe patch (required for X99).
However, I end up with no video at POST and I can't see anything if I manage to enter the UEFI settings (similar to issue #4 ).
I realise that the UEFIPatch was also trying to patch my PciBus with the patch `PciBus | Don't downgrade 64-bit BARs to 32-bit`, which is probably what is causing the no video issue.

Unfortunately, if I apply that patch whether it's through UEFIPatch, UEFITool or even MMTool 5.x (can't use 4.x on my BIOS), it seems to add a new pad-file that wasn't there before:
[image]

I have even tried messing around with the hex editor, however I can only do that on the extracted uncompressed version of the module, so it ends up being the same problem at the end.

Any idea how I could proceed to fix that?

ZOXZX, 2024-06-18:
I've used another version of mmtool to replace pcibus and now bios is free of padding issue, but beware:
on x79 that version caused CPU voltage to be fixed at 1.5V. Thankfully, it's under water cooling and I've seen the voltage error in BIOS and shut down the system.
Maybe there was some config error or it's x79 specific, but use it at your own risk and check immediately for cpu voltage ...
[asus.rom.zip](https://github.com/user-attachments/files/15882520/asus.rom.zip)

gravufo, 2024-06-19:
> I've used another version of mmtool to replace pcibus and now bios is free of padding issue, but beware: on x79 that version caused CPU voltage to be fixed at 1.5V. Thankfully, it's under water [...]

Thanks for the response! So the zip you provided is for which board? Highly doubt I can use it if it's a BIOS for another model.
1.5V on CPU is quite high, even with watercooling, it will degrade the CPU pretty fast. I am on air cooling, so not willing to risk running at 1.5V.

ZOXZX, 2024-06-19:
It is for your board, what else?
Voltage problem was with x79 and mmtool that supports both 4 and 5, maybe it's not present here ...

gravufo, 2024-06-19:
Thanks, opening it in UEFITool NE does show that the pad-file is not there, so it looks pretty good (I'm assuming both the PciBus and the NvramSmiDxe patches were applied).
That's really helpful! I'm not sure when I'll have time to test/flash it, but I'll report back as soon as I do.

AiSuperBot, 2024-07-10:
The cause of the added pad file is that after the patch you applied the module has changed his size causing it to be unaligned, hence needing to add a pad file to realign the other module after the patched one.

There is more than one solution to this:
Try modding an older BIOS version.
Try [compressing the module](https://www.bios-mods.com/forum/showthread.php?pid=71747#pid71747) and the insert it to the BIOS (@[Sweet_Kitten](https://winraid.level1techs.com/u/sweet_kitten)’s solution).
If you have more than one module to patch, try patching them (like DSDT or whatever, while ignoring the added pad file) and if you have enough luck it might fix the pad file issue.

I went through the same issue. [Here](https://winraid.level1techs.com/t/release-resizable-bar-bios-efi-module/100235/711?u=blizz) is the post that concludes all the process, in case this would be helpful to you.

Ruffy47, 2024-07-18:
Hey I tried to modify the BIOS file from Asus. I don´t own this board and can´t test it, but you can try if you want. I added the NVStrapsRebar module too.

https://easyupload.io/yq82fu