xCuri0/ReBarUEFI #216: Asus Z10PE-D16 WS C612 - ReBAR requires pci=realloc in linux and not really working in Win10

LegionnaireENR34, 2024-06-30:
**System**
 - Motherboard: Asus Z10PE-D16 WS - this is dual socket board on C612 PCH
 - Two Xeon e5-2600v3 (haswell-e) installed
 - 128GB RAM
 - BIOS Version: 4101
 - GPU: Intel Arc A770 16GB VRAM
 - CSM is turned off in BIOS
 - 4G decoding is enabled in BIOS
 - UEFIPatch is applied 
 - DSDT Patch not applied as I assumed it is not needed
 - [ ] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**

Hey, I'd like to ask for an advice and a feedback. I run Z10PE-D16 WS - it is a dual socket board with 16 memory slots and c612 chipset. BIOS used for modification is v.4101, latest available at asus.com
The intention was to make ReBAR work with ARC A770

Current state:
1) I've installed UEFI .ffs module as per instructions
2) UEFI patch applied. The one applied was about whitelisting on x99/c612 motherboads, as reported by hex string
3) Above 4G decoding enabled
4) CSM disabled

**First step** 
of enabling resizable BAR was performed in **Linux Mint** with Kernel v.6.5
- If rebarstate variable is set to 0, dmesg shows 256M register size, i915 kernel driver loads successfully 
- If rebarstate variable is set at any value ranging from 14 to 32, lspci reports that intended BAR size is reported to be 16G in lspci, but i915 kernel driver does not load, dmesg shows conflicts in addresses, stating that there is not enough space to allocate BAR of required size.
- If I add pci=realloc to kernel boot parameters, then any value from 14 to 32 in rebarstate works, assigned BAR size is 16G, kernel mode i915 driver is loaded successuly, but I see a bunch of address space reallocations in dmesg. 
So to that end adding ReBAR _appears to be working_ provided that pci=realloc is configured during linux boot. However I have an impression that reallocation should not be required.

**Second step** 
Now, to confirm that ReBAR was really working I tried to install a clean **Windows 10**. At this time Rebarstate is set at 32.
- Installation went smoothly, OS booted fine. 
- Then I've installed Intel's GPU driver. Immediately after installation and before reboot it complained about insufficient resources for the driver to start in the device manager. After reboot Win10 hangs during boot process.
- When rebarstate is set to 8, win10 boots fine as it should with the default 256MB BAR
- When rebarstate is set to 9, it's the same
- When rebarstate is set to 10, motherboard boots1 time from 3, diagnostic codes are pretty messed up
- Rebarstate set to 11 would send the board to cyclic reboot until GPU is unplugged
- When rebarstate is set to 12, MB would boot, but win10 would BSOD
- With rebar of 13 and above win10 just hangs during boot, no BSOD, no recovery, just locks up requiring hard reset  

So is there something wrong with either resource allocation, or with the way .ffs module behaves on dual-CPU/c612 platforms? I guess this result cannot be considered a complete failure, and yet I would not regard it as a success either. Can you please share any thoughts or suggestions as to what should be done next? I can provide dmesg or lspci output if needed.

xCuri0, 2024-07-01:
Try changing BIOS settings related to MMIOH, I know X99/C612 have a few options relating to it under IntelRCSetup

LegionnaireENR34, 2024-07-01:
Thanks for the advice, will try that. Found two options under IntelRCSetup -> Common RefCode Configuration. They are hidden by default, will need to expose them in AMIBCP in order to play with: 
- MMIOH Base - seems to default at **56T**, options: 56T, 40T, 24T, 16T, 12T, 4T, 2T, 1T
- MMIO High size - seems to default at **256G**, options: 128G, 256G, 512G, 1024G 
Both default values seem pretty high already, should I reduce them? Or try to set MMIOH size to 512G and above and leave 56T Base value alone? 

There are other options in BIOS that are PCIe-port specific, about single 64-bit bar or two split 32-bit bars, I tried to play with them briefly, but as far as I can tell those don't affect my problem.

Another question that interests me: does it matter which CPU the GPU is connected to? Do they both share common address space? Do QPI settings and limitations play any role in ReBAR at all?

LegionnaireENR34, 2024-07-14:
Hey @xCuri0, so I've done some additional testing. 
I have exposed MMIO-related options that seemed to be relevant in BIOS, there were 3 of them: 
- MMCFG base - set to 2G by default
- MMIOH base- set to 56T by default
- MMIOH size - there are 4 sizes: 128G, 256G, 512G, 1024G. 

I played with all 4 of the "MMIOH size", but none worked without pci=realloc. 
Here's dmesg report of the result. Device 85:00 is Intel Arc A770 16G:
[image]
If i'm reading this correctly, it appears to be trying to allocate memory starting from address 0x0 and towards higher addresses, not the other way around. In trying to do so it displaces some other devices' memory ranges.
Another interesting thing is that with pci=realloc it succeeds in allocating required BAR size of 16G regardless of rebarstate.

Any ideas as to what I should try next? Why is it trying to allocate memory from low addresses? Is there a way to change this behavior? Is it defined by device driver, OS or UEFI itself?
Can this issue be relevant to QPI and some limitations of how memory is accessed via QPI?
Should I try different brand GPU?

LegionnaireENR34, 2024-08-11:
@xCuri0 
Some new development here. To ensure that my issues were not hardware-specific, I've managed to obtain AMD RX6900 for testing.
Can report that is was at least some success. 
-  On **Windows 10**  with RX6900 and this UEFI module ReBAR **works successfully** with any rebarstate variable above 12. Here's the screenshot of it with rebarstate at 32 
[image]
[image]
- On **Linux Mint** kernel v.6.5 with the same rebarstate amdgpu driver **fails** to start without pci=realloc kernel parameter - it gets stuck in permanent "amdgpu trn=2 ack should not assert! wait again!" message in dmesg. With pci=realloc it appears to load successfully.

So As far as **ReBarUEFI** with ***Z10PE-D16 WS*, we can declare partial success, but not full success. 
Whether the new info justifies any updates to this UEFI module or not - not for me to decide. As of now, I think we can close this thread.

LegionnaireENR34, 2025-01-11:
> I have a board that's C612 and although UEFIPatch initially wouldn't patch anything I'm curious which patch you applied.

As you can tell, it was quite some time ago, so my recollection is a bit vague about specifics, but I'm pretty certain that it was this one that was found and updated:
`# AMI APTIO V NvramSmiDxe (Socket 2011-v3 MB: C612, X99) NVRAM whitelist unlock
54B070F3-9EB8-47CC-ADAF-39029C853CBB 10 P:0F84B300000041F6:90E9B300000041F6` 

 
> Edit: I happen to have a dual-CPU/C612 and I'm trying for two different GPUs, an Nvidia Quadro P4000 (8GB) and an Intel Arc 310 (4GB). I haven't been able to get the bar resized but I'm using Linux and checking via nvidia-smi -q -x and dmesg. 

P4000 is pascal generation, isn't it? If it is, I'd say it is not the correct GPU to check for ReBAR, as I'm quite certain that pascal and its drivers are not aware of resizable BAR. Arc A380 should be the GPU to check is ReBAR works - be mindful of kernel memory allocations though, some extra parameters might be required as I referred above.

> Also, isn't our E5-26XXv4 CPUs and this chipset Broadwell-EP (is that different from Broadwell?) I come from AMD-land, so I'm unaware of a lot of Intel's chipsets.

Sorry, I don't understand the question. E5-26xx and C61x are indeed Haswell-EP and Broadwell-EP, in the sense that they use hsw/brw core structure, but otherwise they differ. It goes as follows:
Haswell/Broadwell for desktop:
- socket 1150,
- Z/Q/B/H 80/90-series chipset, or C22x for Xeon E3 v3,
- 2-channel DDR3 memory controller,
- 16 PCIe lanes.
Haswell/Broadwell-E for HEDT:
- socket 2011-3,
- x99 chipset,
- 4-channel DDR4 memory controller,
- 28/40 PCIe lanes.
Haswell/Broadwell-EP for servers:
- socket 2011-3,
- c61x chipset,
- 4-channel DDR4 memory controller,
- 28/40 PCIe lanes,
- can have QPI for multi-CPU config, so 1-, 2- and 4-socket motherboards available. 

>  I wonder if the boards try to put MMIO space above 64GB.
As far as I'm aware, there's nothing wrong with putting MMIO region above 64GB (which is not a lot of space actually) provided that Above 4G decoding is enabled. MMIOH base in my case was way above 64GB, it was at 56T

savvadesogle, 2025-02-21:
Working on Z10PE-D8 WS with 2x Intel Arc A770

https://github.com/xCuri0/ReBarUEFI/issues/275

LegionnaireENR34, 2025-02-22:
> Working on Z10PE-D8 WS with 2x Intel Arc A770
> 
> [#275](https://github.com/xCuri0/ReBarUEFI/issues/275)

Interesting! Are you using just the patch, or something else was required? What OS are you running?

savvadesogle, 2025-02-22:
> > Working on Z10PE-D8 WS with 2x Intel Arc A770
> > [#275](https://github.com/xCuri0/ReBarUEFI/issues/275)
> 
> Interesting! Are you using just the patch, or something else was required? What OS are you running?

1. Сперва, взял оригинальный 4301 ROM (последний для моей платы) и извлек из него тело (body) BIOS. UEFITool -> Extract body. 
2. Удалил mcu (все) через mmtool (CPU Patch)
3. Дальше собрал Turbo unlock dxe-драйвер (для двухсокетной платы) DXETurboHack v.10
4. Добавил turbohack и Rebar dxe драйвера в BIOS (UEFITOOL по инструкции)
5. Применил к полученному файлы патчи  (uefipatch)
6. Дальше начал менять параметры по умолчанию в BIOS Setup (AmiBCP) - часть настроек указал в https://github.com/xCuri0/ReBarUEFI/issues/275
Включил скрытые настройки (переключил с Default на USER доступ). 
Включил above 4G, а CSM отключил. Еще по мелочи настроил разные параметры, чтобы каждый раз при сбросе не переставлять. Которые точно не влияют. Сделал так, чтобы гарантированно плата стартовала с картой и был вывод картинки.
Выключил PCI-E ASPM L1 для PCI в нескольких местах (не уверен, что влияет, но все же). 
В Main есть безымянная папка - в ней так же выключил CSM и включил Above 4G.
Trusted Computing - решил не отключать.

7. Зашил BIOS через программатор ch341A + Neoprogrammer (у меня он черный)

Плата стартанула с A770 (раньше приходилось долбаться с 1660ti сперва запускать).

8. После успешной загрузки в Windows (или Ubuntu) приступил к конфигурации настроек BIOS.
PCIe Max Read Request Size изменил на авто ((не уверен, что влияет, но все же).
MMCFG Base: 3G
MMIOH Base 56T
MMIO High Granularity Size (MMIO High Size): 1024G
Отключил деление 64 на несколько 32 (Enable SPLIT BARs) - по идее должно патчем исправляться, но в BIOS настройку отключил. 
Я не помню названия, но в одном из IIO находил подключеные устройства и жестко выбирал их скорость и версию PCI. Например у меня есть карта A770 в PCI 1 - нашел ее, и выбрал вместо авто, x16 и 8GT скорость. А есть Wi-Fi модуль в PCI 6 - ему так же прописал Gen3 и X8 (слот как раз на X8 рассчитан)

9. После того, как все настройки нормально работают и система грузится (потестировал, нормально поиспользовал) - включил ReBARState в 14 (то есть 16GB).
10. После перезагрузки был успешный запуск (на видео выход не было цветного мусора)

ps видеовыход (монитор) подключен к PCI 7 HDMI.

savvadesogle, 2025-02-22:
> With ASPM enabled GPU sits around 7-8W. In my case ASPM messes with virtualization and device sharing, so I have it disabled

Да, читал про это и применил патч от производителя для моих карт (и у тех карт, которые не подключены к мониторы и не активны реально 7-8w потребление). Но сейчас решил отключить, может позже попробую вновь включить. 

> MMCFG base and MMIOH granularity

У меня они по умолчанию были 56T и 256GB. Но решил поменять их на 1024GB, а 56T оставил. 
Читал, что можно пробовать уменьшать  MMCFG base  (40T, 24T...) но вот пока сработало в таком виде. 

Если важно, я еще батарейку вытаскивал, помимо сброса (через джампер). Ну и естественно отсоединял кабель питания из БП.

Davehillbo, 2025-09-09:
Hi there, Did you ever get around the resizable bar issue? 
I have my bios patched and it works fine with an AMD 7900GRE (though black screen until it boots into windows)
Put in a RTX 3090 and fails to post with any rebarstate above 9.

Ta!

Davehillbo, 2025-09-09:
Yes bios has been reset quite a few times.

Ah well, ill leave it off

LegionnaireENR34, 2025-09-09:
> Hi there, Did you ever get around the resizable bar issue? I have my bios patched and it works fine with an AMD 7900GRE (though black screen until it boots into windows) Put in a RTX 3090 and fails to post with any rebarstate above 9.
> 
> Ta!

I take it you have Z10PE-D16 WS as well?
As far as I remember, Ampere GPUs did not initially support ReBAR, there were VBIOS patches to enable it, did you install those? Considering that 3090 was one of the first released GPUs of that generation

Davehillbo, 2025-09-09:
I’m using the z10pe-d8, though the bios are very similar .

Gpuz states my card is good to go for rebar , just not enabled in bios

Davehillbo, 2025-09-09:
> I see. If you put ReBarState to 9, what is the assigned BAR size? Does it show 512MB as it should?

It show 512mb without bar size 9, and even if I set size 9 it’s showing as rebar disabled .

Davehillbo, 2025-09-09:
> That's odd IMO, default BAS size should be 256MB as far as I know.
> Well, if VBIOS is up to date, rebar module is in bios, CSM is disabled and Above4G is enabled, and 3090 still does not work, then I guess we're out of luck. Those motherboards can be temperamental. Sorry for not being very useful

No it’s fine thanks for the help .
Looks like NVIDIA rebar ain’t possible on these boards

LegionnaireENR34, 2025-09-09:
> > > Working on Z10PE-D8 WS with 2x Intel Arc A770
> > > [#275](https://github.com/xCuri0/ReBarUEFI/issues/275)
> > 
> > 
> > Interesting! Are you using just the patch, or something else was required? What OS are you running?
> 
>     1. Сперва, взял оригинальный 4301 ROM (последний для моей платы) и извлек из него тело (body) BIOS. UEFITool -> Extract body.
> 
>     2. Удалил mcu (все) через mmtool (CPU Patch)
> 
>     3. Дальше собрал Turbo unlock dxe-драйвер (для двухсокетной платы) DXETurboHack v.10
> 
>     4. Добавил turbohack и Rebar dxe драйвера в BIOS (UEFITOOL по инструкции)
> 
>     5. Применил к полученному файлы патчи  (uefipatch)
> 
>     6. Дальше начал менять параметры по умолчанию в BIOS Setup (AmiBCP) - часть настроек указал в [Asus Z10PE-D8 WS with Arc A770 (working) 💪 #275](https://github.com/xCuri0/ReBarUEFI/issues/275)
>        Включил скрытые настройки (переключил с Default на USER доступ).
>        Включил above 4G, а CSM отключил. Еще по мелочи настроил разные параметры, чтобы каждый раз при сбросе не переставлять. Которые точно не влияют. Сделал так, чтобы гарантированно плата стартовала с картой и был вывод картинки.
>        Выключил PCI-E ASPM L1 для PCI в нескольких местах (не уверен, что влияет, но все же).
>        В Main есть безымянная папка - в ней так же выключил CSM и включил Above 4G.
>        Trusted Computing - решил не отключать.
> 
>     7. Зашил BIOS через программатор ch341A + Neoprogrammer (у меня он черный)
> 
> 
> Плата стартанула с A770 (раньше приходилось долбаться с 1660ti сперва запускать).
> 
>     8. После успешной загрузки в Windows (или Ubuntu) приступил к конфигурации настроек BIOS.
>        PCIe Max Read Request Size изменил на авто ((не уверен, что влияет, но все же).
>        MMCFG Base: 3G
>        MMIOH Base 56T
>        MMIO High Granularity Size (MMIO High Size): 1024G
>        Отключил деление 64 на несколько 32 (Enable SPLIT BARs) - по идее должно патчем исправляться, но в BIOS настройку отключил.
>        Я не помню названия, но в одном из IIO находил подключеные устройства и жестко выбирал их скорость и версию PCI. Например у меня есть карта A770 в PCI 1 - нашел ее, и выбрал вместо авто, x16 и 8GT скорость. А есть Wi-Fi модуль в PCI 6 - ему так же прописал Gen3 и X8 (слот как раз на X8 рассчитан)
> 
>     9. После того, как все настройки нормально работают и система грузится (потестировал, нормально поиспользовал) - включил ReBARState в 14 (то есть 16GB).
> 
>     10. После перезагрузки был успешный запуск (на видео выход не было цветного мусора)
> 
> 
> ps видеовыход (монитор) подключен к PCI 7 HDMI.

Sorry, apparently I've missed your reply, very interesting! Not sure if we should continue, this is sort of a reference mainly for English-speaking audience. Nonetheless let us discuss, and if the discussion turns out to be useful, I'll summarize later for the rest of the readers.

 1. Я правильно понял, что шаги 2-3-4 - это из известной инструкции по разрешению всем ядрам бустить до максимальных частот? Я давно не ковырялся с хазвеллами, но MCU - это вроде то ли профили питания, то ли апдейт, который латает дырки для спектра/мелтдауна, то ли и то, и другое сразу. Я этого не делал у себя, но в значительной степени уверен, что не влияет на работу/неработу ребара
 2. У меня MMCFG=2G, база MMIOH=56T, гранулярность окна - 512G, я с этим игрался, разницы нет - с А770 без танцев с бубном не работало.
 3. А вот дальше я не понял, что ты делал, вернее, на чем остановился. В этом треде у тебя написано, что отключил деление баров на 32/32, т.е. слот обслуживается одним х64 регистром. А в 275 теме написано, что включено? 
 Я у себя выключал деление, т.е. у меня один 64-разрядный регистр - я решил, что надо отключить, потому что если гранулярность обращения к области памяти с точностью до байта, то адрес, обращающийся в 16 гигам, банально не поместится в 32-разрядный БАР, ему хотя бы 34 разряда надо. Хотя позже я бегло глянул в спецификацию PCIe, и там вроде как БАР адресует не отдельный байт, а окна размером от 1 мегабайта, тогда 32 разряда должно хватить.
4. Скорости вот я у себя не выставлял на слотах. Не видел нужды, в линуксе lspci сообщал, что процы с видяхой правильно договорились о скорости и кол-ве линий - х16 и 8ГТ/с
5. Подключал тоже в разъем HDMI, но это не помогало, ибо при значениях выше 10 винда либо БСОДила, либо наглухо висла - как я написал в сообщении выше.

В общем, пока напрашивается вывод, что платы сильно отличаются друг от друга, несмотря на сходство в названиях - ну ессно вместе с биосами. Отличаются настолько, что аллокация ресурсов при загрузке сильно разная. Ибо тебе с А770 сопутствовал полный успех, а мне - только частичный, с виндой совсем не удалось, а в линуксе - вон настройки ядра надо крутить.
У тебя есть на плате ВМС-чип?

savvadesogle, 2025-09-09:
@LegionnaireENR34 
1. Полагаю, что удаление микрокода не должно влиять на ReBAR. Но есть вероятность, что добавляли какие-то патчи безопасности, которые и касались работы pci, но это только предположение. Делал для того, чтобы при avx инструкциях частоты не сбрасывал процессор и не работал медленнее из-за защит от spectre meltdown etc. 
2. Сейчас, по сути, уже не трогаю базовые настройки. MMCFG, MMIOH,  оставляю такими, какие они изначально в прошивке идут. 
3. Сейчас уже не помню. Много вариантов тестировал и пытался понять разницу в производительности ML задач от этих параметров.
4.  Отключение L1 и наоборот включение скоростей x16/v3 делал скорее на всякий случай. Сама Intel писала, в некоторых issue, что лучше отключить экономию энергии. 
5. Сейчас я и hdmi и dp на a770 использую. Но экономия энергии у карточек включается только по hdmi и монитор 50hz не выше. 

В общем ключевой момент был для меня поменять в настройках CSM off, 4g above on  до прошивки, потому что иначе без сторонней карты сразу нельзя было запустить a770. И был постоянный риск при сбросе CMOS получить кирпич (без картинки) систему. 
Сейчас у меня dual boot windows 11 и ubuntu. 
И да, мне повезло, все 4 карты работают в обеих системах. Производительность в ml задачах идентичная референсным, которые показывает intel в своих тестах. 

Один лишь нюанс (монитор у меня подключен к карте в 7ом порте), pcie=realloc нужен, иначе на карте в первом слоте ресурсы нормально не инициализировать, на остальных слотах в default параметрах ядра - стартует и bar2 16gb. После pcie=realloc и в pcie1 (x16 самый верхний) тоже bar2 16gb. 
При том в windows 11 все нормально с rebar на всех картах. 

BMC вроде отсутствует 

ps иногда бывало помогало полное обестачивание системы ("из розетки" вытаскивать) и дать чуть постоять и повторно запускать, чтобы нормально стартовала система. Может из-за dual boot разный