xCuri0/ReBarUEFI #219: It works well, but an extra display appeared.

crazyhulk, 2024-07-07:
**System**
 - Motherboard: EPC621D8A
 - BIOS Version: 2.10
 - GPU: RX 6950xt
 - [Y] CSM is turned off. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [Y] 4G decoding is enabled. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [N] UEFIPatch is applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [Y] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**
I have some questions:

1. There is an extra display appeared when enable resizable bar.
[image]
2. What is the difference between bar0 and bar1 ? The BAR0 is 16G.
[image]
3. I can not boot my hackintosh(mac OS) after enable resizeable bar. I set ResizeAppleGpuBars to 0 which the OpenCore suggested.
4. No video output during boot before OS loads, I had read  [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes)), and  my motherboard does not need  Above 4G Decoding Patch. So if I want a bios change, I must do it by **BMC**, the bmc can do it by KVM.

Thx!

xCuri0, 2024-07-07:
@crazyhulk regarding hackintosh read this https://github.com/xCuri0/ReBarUEFI/issues/32

BAR1 is used for something else on AMD, I don't think it being 256MB matters it's like this on supported motherboards too.

crazyhulk, 2024-07-07:
> @crazyhulk regarding hackintosh read this #32
> 
> BAR1 is used for something else on AMD, I don't think it being 256MB matters it's like this on supported motherboards too.

Thx very much, I will test it and give a reply.