xCuri0/ReBarUEFI #220: Cannot correct padding error on Asus Maximus V Gene

kcajjones, 2024-07-17:
**System**
 - Motherboard: Asus Maximus V Gene
 - BIOS Version: 1903
 - GPU: Palit RTX 3080
 - [Y ] CSM is turned off. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [ Y] 4G decoding is enabled. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [ N] UEFIPatch is applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [ Y] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**

I have succesfully injected the RebarDxe FFS module, modified the BIOS menu to show PCI Settings -> Above 4G Decoding option but cannot use UEFIPatch to correct the various issues with limited BAR size on my Ivy Bridge / Z77 platform.
As described in the guide, the patches are applied but padding errors are introduced. I have tried to follow the further "fixes" but these lead to a Hex edit to move the changes from the patched (but glitched) file to the original file. This does not work as the PciBus and PciHostBridge modules are compressed and the file sizes are different. If I extract the "Body only" with UEFITool then I can see that the body data is the same size, with the hex editor confirming the correct changes, however the header/compression (?) has somehow been altered and the total size is different so the whole structure beyond the PciBus module is corrupt, including the padding errors.

I've not been able to work out how to fix this issue as MMTool does the exact same thing.

I see in the list of compatible motherboards that a user (who is no longer an active username on github?!) has confirmed this motherboard works with UEFIPatch. I can confirm Resizable BAR is enabled, but limited to 2GB until I can get the UEFI patches to work.

Any help would be greatly appreciated as I have Rebar enabled with 2GB but the system fails to boot if I set it to >=4GB.

Ruffy47, 2024-07-18:
Hey I tried to modify the BIOS file from Asus. I don´t own this board and can´t test it, but you can try if you want. I added the NVME5 and NVStrapsRebar module too.

[Uploading MAXIMUS-V-GENE-ASUS-1903.Above4gDecode.UEFI.patches.NVME5.ReBar.NvStrapsReBar(corrected.pad.file).zip…]()

Ruffy47, 2024-07-18:
Hey I tried to modify the BIOS file from Asus. I don´t own this board and can´t test it, but you can try if you want. I added the NVME5 and NVStrapsRebar module too.

https://easyupload.io/ievkuc

kcajjones, 2024-07-18:
> Hey I tried to modify the BIOS file from Asus. I don´t own this board and can´t test it, but you can try if you want. I added the NVME5 and NVStrapsRebar module too.
> 
> https://easyupload.io/ievkuc

Thanks for trying, but on inspection with UEFITool, this is not correct. There are padding errors.

Upon inspection, the PciBus and PciHostBridge modules are different sizes from the original BIOS, with the PciBus module not having a compressed section at all.
I (think) this is what causes the padding errors.

kcajjones, 2024-07-18:
I'm using UEFITool 0.28 but it also shows the same in UEFITool NE A68.
What are you using?

Original:
[image]

Yours:
[image]

Ruffy47, 2024-07-18:
I just downloaded the original file from ASUS and there it looks exactly the same like in my modified file.

kcajjones, 2024-07-18:
> I just downloaded the original file from ASUS and there it looks exactly the same like in my modified file.

How does it look exactly the same? Via which app?

Ruffy47, 2024-07-18:
UEFI Tool

Ruffy47, 2024-07-18:
Ok anyway, you can try the mod file if you want or not. That´s ok too of course, I just tried to help. :)

kcajjones, 2024-07-18:
> UEFI Tool

Which version?
Can you show a screenshot comparison to confirm your modified version has the correct padding?

kriosad, 2025-02-03:
@kcajjones Hello, have you found a solution yet? I am experiencing a similar problem.

kcajjones, 2025-02-03:
> [@kcajjones](https://github.com/kcajjones) Hello, have you found a solution yet? I am experiencing a similar problem.

Yes, I had to use MMTool and not UEFITool. UEFITool would corrupt the BIOS image.

Here's a BIOS file I made without the Primary MAC, System UUID and Motherboard S/N. If you take a backup of your current BIOS file you can then use FD44Editor v0.9.2 to input the required parameters from the backup to my updated BIOS.
I can't remember if I had to do something to flash the BIOS file. I think BIOS Flashback didn't work so I used Intel ME Tool to flash the BIOS from in Windows.

Here's the modules that have been updated:
[
[image]
](url)

It also contains ReBAR, NVMe support and UEFIPatch and enables hidden options in the BIOS for Above 4G Decoding.

https://www.sendspace.com/file/cana1r

kriosad, 2025-02-05:
@kcajjones

pntsrgd, 2025-04-23:
> Yes, I had to use MMTool and not UEFITool. UEFITool would corrupt the BIOS image.
>
> Here's a BIOS file I made without the Primary MAC, System UUID and Motherboard S/N. If you take a backup of your [...]

Any chance you could give a little information on how you used MMTool?

I'm seeing an extremely similar problem with the Maximus V Formula.

I attempted to insert the rebardxe module with MMTool and patched image.

After patching the image, UEFITool shows pad file issues.

I extracted the modules "as is" using MMTool and then reinserted them into the original image.

The image then showed the same pad file issues when opened in UEFITool.

Koekieezz, 2026-01-04:
Hello @pelelademadera @misterright1, Could you please test this bios: [M5G_Test.zip](https://github.com/user-attachments/files/24423056/M5G_Test.zip)

To flash it, plug your flash drive to the USB Port that designated for Bios Flashback purpose, then extract the file, put the .CAP to your flash drive, and use bios flashback to flash it. If bios flashback isn't recognizing the flash drive because your flash drive is too big (16GB+) even if they're formatted at FAT32, try to make a small partition on the flash drive (100MB  is fine), and format it as FAT32, then put the .CAP file on the 100MB FAT32 Partition, that way the board should recognize the .CAP in the 100MB FAT32 Partition.

If bios flashback works, it flashed the bios, and it booted fine and resizable bar could actually be used, please tag reply me :)

kcajjones, 2026-01-04:
> > > > [@kcajjones](https://github.com/kcajjones) Hello, have you found a solution yet? I am experiencing a similar problem.
> > > 
> > > 
> > > Yes, I had to use MMTool and not UEFITool. UEFITool would corrupt the BIOS image.
> > > Here's a BIOS file I made without the Primary MAC, System UUID and Motherboard S/N. If you take a backup of your current BIOS file you can then use FD44Editor v0.9.2 to input the required parameters from the backup to my updated BIOS. I can't remember if I had to do something to flash the BIOS file. I think BIOS Flashback didn't work so I used Intel ME Tool to flash the BIOS from in Windows.
> > > Here's the modules that have been updated: [ [image] ](url)
> > > It also contains ReBAR, NVMe support and UEFIPatch and enables hidden options in the BIOS for Above 4G Decoding.
> > > https://www.sendspace.com/file/cana1r
> > 
> > 
> > Any chance you could give a little information on how you used MMTool?
> > I'm seeing an extremely similar problem with the Maximus V Formula.
> > I attempted to insert the rebardxe module with MMTool and patched image.
> > After patching the image, UEFITool shows pad file issues.
> > I extracted the modules "as is" using MMTool and then reinserted them into the original image.
> > The image then showed the same pad file issues when opened in UEFITool.
> 
> I second this, can you upload it again? the link is dead.

I'll upload another copy of the BIOS when I'm back home.

As regards how I used MMTool, it was a while ago and I can't remember the exact procedure. I realised that uefi tool appeared to work but the BIOS wouldn't work correctly when flashed.

Koekieezz, 2026-01-04:
> I'll upload another copy of the BIOS when I'm back home.
> 
> As regards how I used MMTool, it was a while ago and I can't remember the exact procedure. I realised that uefi tool appeared to work but the BIOS wouldn't work correctly when flashed.

Hello @kcajjones , I've been reading your problem and after i tried to edit the bios from stock, i came up with a potential solution. I did PM you in win-raid, but i think you didn't use that forum anymore, and since you're here, i hope you do read this comment 😁.

So basically in my edit from stock, the PciBus module is the cause of missing pad files, which then i tought to "relocate" the PciBus module to the bottom. Before doing so, i actually did try on my Gigabyte H77 DS3H, that board bios doesn't have any issue with the uefipatches, but i feel silly and i re did the bios from scratch, but this time, i Delete Runtime, PciBus, AmiBoardInfo, and PciHostBridge, and then i insert as-is using MMTool the patched modules (which is Runtime, Pcibus, AmiBoardInfo, and PciHostBridge), which came down to the bottom, it looks like this:
[image]

And surprisingly, it worked.. at least in my board. (Screenshot above is captured on the pc, i gave the bios name Cursed.rom bcs i feel silly to do the experiment 😂).

Above your comment, i did give the modded bios for M5G using the method i did, and the pad files are still there! But since i dont have the board, if you don't mind, would you kindly test it? I would like you to downgrade the bios version to any version below 1903, and then use bios flashback the modded bios i did gave above, and if it works and stable, i might try several Asus Z77/68/P67 bioses that does have Above 4G Decoding using the method i mentioned above.

Koekieezz, 2026-01-05:
> > Hello [@pelelademadera](https://github.com/pelelademadera) [@misterright1](https://github.com/misterright1), Could you please test this bios: [M5G_Test.zip](https://github.com/user-attachments/files/24423056/M5G_Test.zip)
> > To flash it, plug your flash drive to the USB Port that designated for Bios Flashback purpose, then extract the file, put the .CAP to your flash drive, and use bios flashback to flash it. If bios flashback isn't recognizing the flash drive because your flash drive is too big (16GB+) even if they're formatted at FAT32, try to make a small partition on the flash drive (100MB is fine), and format it as FAT32, then put the .CAP file on the 100MB FAT32 Partition, that way the board should recognize the .CAP in the 100MB FAT32 Partition.
> > If bios flashback works, it flashed the bios, and it booted fine and resizable bar could actually be used, please tag reply me :)
> 
> I was planning on using a USB flashing tool, CH341a, but I'll have a chance to do so after the holidays(so like a week from now).

If you're planning to use CH341A to flash it directly, then you would need FD44Editor to transfer your current motherboard bios dump, i suggest to use Bios Flashback method if possible

Koekieezz, 2026-01-15:
> ✅ Just to let you know that this BIOS build did successfully flash with BIOS Flashback!
> 
> It did not boot at first because **CSM** was turned on by default and my **Intel Arc A380** prevented the system from posting (stuck at code `b2`), I had to remove it and use the integrated graphics to disable **CSM**
> 
> Then I manage to build **ReBarState** on a live Linux Mint and set the Resizable BAR size to `32`, rebooted and checked that it was applied successfully.
> My **TrueNAS** installation booted successfully from my NVME SSD too! **Ollama** is now working great (the `ollama-intel-gpu` build)

Hello @nsb002 , Thank you very much for testing the bios ✨️🫡, this is actually an experimental "solution" due to the infamous "Padding file issue" i've experienced personally doing Asus bios ever since i've started bios modding. I've contacted @xCuri0 itself and [sweet_kitten](https://winraid.level1techs.com/u/sweet_kitten) in Win-Raid forums regarding to this, and they never saw such a feat being done for now, and i decided to test the method itself to my Gigabyte H77-DS3H. But knowing the gigabyte bios that i've modded from the first place didn't have any issues from the first place, doing the method seems to be "useless" for me since the normal way i did the bios has never caused the board to brick. The workaround i did use for this Asus bios that i've applied to my own Gigabyte H77 DS3H did works anyway, no difference between the already made bios i did vs the workaround test bios mod build.

**How i did achieve this:**
The way that i achieved this was by Deleting the PciBus dxe module in the original location, and "Insert As Is" or you can say "relocate" the PciBus dxe driver, just like when you're inserting any modded dxe driver (for bios insert/replace/delete, i always use MMTool 5.0.2.24 modded version which could do most Ami Uefi Aptio 4 image except most Asrock Bios for some reason, which needs MMTool 5.0.0.7). You can open this modded bios on UEFITool NE and see that the PciBus module is in the very bottom of DXE Volume, while the padding files are still intact as the original bios, you can compare both stock .CAP downloaded from asus web and my modded .CAP and the Pad-file location is still exactly the same.

I never expect anyone to test this, since i did it on a free time as a way to relieve my thought regarding pad file issue and theoritical thoughts 😂

If you don't mind, could you make a Screenshot similar to this: [image]

I will use your screenshot to post this on Win-Raid forums, with also explaining how i did achieve it for documentation purposes, and will credit your git user page for the screenshot.

Once again, im very thankful for the test 🫡

nsb002, 2026-01-15:
> Hello [@nsb002](https://github.com/nsb002) , Thank you very much for testing the bios ✨️🫡, this is actually an experimental "solution" due to the infamous "Padding file issue" i've experienced [...]

Unfortunately I don’t have any Windows available on this machine, so I can’t produce this kind of screenshot easily.
I did check using `lspci`, I could find the command I used and paste the output here maybe.
Thanks for your awesome work, it really made my BIOS mod easy this time around! It’s really astonishing to see such old hardware able to use this feature and run new workloads.

Koekieezz, 2026-01-15:
> Unfortunately I don’t have any Windows available on this machine, so I can’t produce this kind of screenshot easily.
> I did check using `lspci`, I could find the command I used and paste the output [...]

You can post the lspci log here (and post your success result on Working Motherboard list in this git page) or maybe use [GPU-Z](https://github.com/atareao/cpu-g) or similiar app that exposes that the bar is assigned correctly in GUI form, it would be useful for those who seeks on why did it work.

BTW, where did you enable Above 4G Decoding?

nsb002, 2026-01-15:
Here's the result of `lspci` on my **TrueNAS Scale 25.10.1 - Goldeye**:

```shell
$ sudo lspci -v -s 03:00.0
03:00.0 VGA compatible controller: Intel Corporation DG2 [Arc A380] (rev 05) (prog-if 00 [VGA controller])
        Subsystem: Device 172f:3943
        Flags: bus master, fast devsel, latency 0, IRQ 56, IOMMU group 1
        Memory at f6000000 (64-bit, non-prefetchable) [size=16M]
        Memory at c00000000 (64-bit, prefetchable) [size=8G]
        Expansion ROM at f7000000 [disabled] [size=2M]
        Capabilities: [40] Vendor Specific Information: Len=0c <?>
        Capabilities: [70] Express Endpoint, MSI 00
        Capabilities: [ac] MSI: Enable+ Count=1/1 Maskable+ 64bit+
        Capabilities: [d0] Power Management version 3
        Capabilities: [100] Alternative Routing-ID Interpretation (ARI)
        Capabilities: [420] Physical Resizable BAR
        Capabilities: [400] Latency Tolerance Reporting
        Kernel driver in use: i915
        Kernel modules: i915, xe
```

For where I enabled `Above 4G Decoding`, I don't remember exactly I think it was under a menu relative to PCI...

It's a headless box, but I can check tomorrow using one of my USB HDMI capture card, I could also take some screenshots of the BIOS with that.

nsb002, 2026-01-15:
> > Here's the result of `lspci` on my **TrueNAS Scale 25.10.1 - Goldeye**:
> > $ sudo lspci -v -s 03:00.0
> > 03:00.0 VGA compatible controller: Intel Corporation DG2 [Arc A380] (rev 05) (prog-if 00 [VGA controller])
> >         Subsystem: Device 172f:3943
> >         Flags: bus master, fast devsel, latency 0, IRQ 56, IOMMU group 1
> >         Memory at f6000000 (64-bit, non-prefetchable) [size=16M]
> >         Memory at c00000000 (64-bit, prefetchable) [size=8G]
> >         Expansion ROM at f7000000 [disabled] [size=2M]
> >         Capabilities: [40] Vendor Specific Information: Len=0c <?>
> >         Capabilities: [70] Express Endpoint, MSI 00
> >         Capabilities: [ac] MSI: Enable+ Count=1/1 Maskable+ 64bit+
> >         Capabilities: [d0] Power Management version 3
> >         Capabilities: [100] Alternative Routing-ID Interpretation (ARI)
> >         Capabilities: [420] Physical Resizable BAR
> >         Capabilities: [400] Latency Tolerance Reporting
> >         Kernel driver in use: i915
> >         Kernel modules: i915, xe
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > For where I enabled `Above 4G Decoding`, I don't remember exactly I think it was under a menu relative to PCI...
> > It's a headless box, but I can check tomorrow using one of my USB HDMI capture card, I could also take some screenshots of the BIOS with that.
> 
> I'll be looking forward for the bios screenshot, BTW, In Bios Setting -> Advanced, PCI Subsystem Settings is available for you right? The above 4g decoding option is under that menu, if so, you only need to take a screenshot inside the contents under Advanced -> PCI Subsystem Settings -> Above 4G Decoding, i will use that as a pointer in my post [here](https://winraid.level1techs.com/t/offer-asus-rog-maximus-v-gene-bios-mods-with-rebar-support/113684?u=koekieezz)
> 
> AFAIK In Windows (maybe also in linux), installing the latest driver would update the firmware of the gpu itself, which solves the issue of the card acting wierdly on old platforms. After it was updated, it should be booting fine in older platforms even with above 4g decoding disabled.

As promised, here are some screenshots of my BIOS:

# Main BIOS Information screen

[image]

# Advanced - PCI Subsystem Settings

This is where I enabled `Above 4G Decoding`

[image]

# Advanced - PCI Subsystem Settings - PCI Express Settings

I changed nothing there, but I think I never saw this screen before the mod.

[image]

# Advanced - System Agent Configuration

We can see that `Memory Remap Feature` is enabled there, I don't remember enabling it from the BIOS, maybe it was activated with the configuration of [ReBarState](https://github.com/xCuri0/ReBarUEFI/tree/master/ReBarState), I'm not sure.

[image]