xCuri0/ReBarUEFI #222: rebar stopped working after half a year working good

fab98bc, 2024-08-01:
**System**
 - Motherboard: asus maximus VIII gene
 - BIOS Version: 3801
 - GPU: RTX 3070 ROG STRIX
 - [ ] CSM is turned OFF
 - [ ] 4G decoding is enabled. 
 - [ ] UEFIPatch is applied
 - [ ] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**
after half a year running flawlessly, today just stopped working and I dont know why. wake from sleep always work without single issue but today it fails error code E3 o wake. 

then I set rebar size 32 as always and it wont post. tried another bios chip with a fresh install via epromm programer, same issue. after activate 4g decoding and desactivate cms. rebar set to 32 it wont boot. keeps in a0 state with white vga led on.

all of this never happened before. It always worked good .

only new thing in my pc is win11 update (KB5040527). 

Please help :(

xCuri0, 2024-08-01:
Smaller sizes work ? (like 10)

xCuri0, 2024-08-02:
@fab98bc did you verify if GPU-Z shows that it's running at x16 Gen3 (under load otherwise it will show Gen1.1), reseating the GPU might fix if it's some issue like that.

fab98bc, 2024-08-02:
I have a pci nvme adapter so my gpu runs at 8x 3.0 
But I never had a single issue there and since I haven't done any modifications, software or hardware related is very strange that it stopped working... 

I tried removing the adapter but got the same result, gpu is run 16x 3.0. 

Testing different things, If I enable igpu it doesn't give me enter bios screen just boots right into windows and wake from sleep works. But it keeps vga check led on in motherboard.

fab98bc, 2024-08-02:
> @fab98bc did you verify if GPU-Z shows that it's running at x16 Gen3 (under load otherwise it will show Gen1.1), reseating the GPU might fix if it's some issue like that.

 I have a pci nvme adapter so my gpu runs at 8x 3.0 But I never had a single issue there and since I haven't done any modifications, software or hardware related is very strange that it stopped working...
> 
> I tried removing the adapter but got the same result, gpu is run 16x 3.0.
> 
> Testing different things, If I enable igpu it doesn't give me enter bios screen just boots right into windows and wake from sleep works. But it keeps vga check led on in motherboard

fab98bc, 2024-08-08:
Still nothing works. After enbaled rebar pc boots with no image. Using remote desktop i can see that it boot but seems the gpu doesn't start correctly. After disable rebar it works normally but I noticed cms enables itself again on bios. All my drives are gpt partitioned