xCuri0/ReBarUEFI #233: Bios is in .exe formate

Musa-Irfan, 2024-09-10:
I tried to extract it but it dosent work my mother board is the xps 8500 here is the bios https://dl.dell.com/FOLDER05052628M/1/A14.EXE

Musa-Irfan, 2024-09-10:
i tried it again now its saying command line parameter is not correct

localhosts-A, 2024-09-10:
Dell's bios seems to require special software to decompress, on Github there are

localhosts-A, 2024-09-10:
Or try to extract it directly from the machine?
Use a programmer or software

localhosts-A, 2024-09-10:
https://github.com/vuquangtrong/Dell-PFS-BIOS-Assembler

Musa-Irfan, 2024-09-10:
oh wow lemme try that

Musa-Irfan, 2024-09-10:
oh how do i use that assembler

localhosts-A, 2024-09-11:
I don't know, I've only used it on a Dell 5559

Musa-Irfan, 2024-09-11:
I Tried using 7 zip and i get a file called FFFFFFFF[VOL] is this the bios?

Musa-Irfan, 2024-09-11:
nvm i used uefi tool on this and it worked and i found pcie bus thanks everyone

closed 2024-09-11 by Musa-Irfan