xCuri0/ReBarUEFI #239: Uefitool output not allowing post, mmtool missing PciBus filename

AbdullahMPrograms, 2024-09-15:
**System**
 - Motherboard: T7920 Proprietary
 - BIOS Version: 2.42.0
 - GPU: RTX 3080
 - [X] CSM is turned off. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [X] 4G decoding is enabled. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [ ] UEFIPatch is applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [X] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**
I'm currently trying to use uefitool to add the rebardxe driver to my dumped stock T7920 bios (dumped via ch341, have erified this bios works by itself). When using uefitool to add the rebardxe driver, the T7920 will not post, instead it displays a bios recovery "please flash your bios" light code. Removing the rebardxe driver from the modified bios via uefitool (essentially reverting it to stock) and flashing that also gives the same issue, this leads me to believe the issue is caused by uefitool. I've tried using mmtool, but it seems I am missing the "PciBus" filename, thus I am unable to proceed. Uefipatch was not used as there were no patches to apply and dsdt patching is not needed. Any suggestions would be appreciated.

Image of mmtool output:
[image]

ZOXZX, 2024-09-15:
Is that a dual Xeon platinum?
Anyway, just checked bin dump and all names are listed.
Maybe try to use newer version of MMTool?

AbdullahMPrograms, 2024-09-15:
Its dual xeon 4110's, I also tried MMTool 5.007 and it was the same thing.

I see you were able to see the file names, can you confirm which version of mmtool you are using? Its possible Im just missing something on my side.

ZOXZX, 2024-09-15:
MMv5 (opens both Aptio 4 and 5), dump from Haswell system (aptio 5 I believe) ...
Does uefitool show correct entries?
Send the file to have a look ...

AbdullahMPrograms, 2024-09-15:
Heres the file: https://drive.google.com/file/d/1qOOfHJmvMQnGFX1pzJjn-6fJqe9byrGO/view?usp=sharing

Searching via guid in uefitool does bring up a volume so I believe all is good on that side

ZOXZX, 2024-09-15:
Entries can't be seen in any version of uefitool/mmtool, except 7 unrelated drivers.
BTW, BIOS is bootguard protected:
https://edc.intel.com/content/www/us/en/design/ipla/software-development-platforms/client/platforms/alder-lake-desktop/12th-generation-intel-core-processors-datasheet-volume-1-of-2/010/boot-guard-technology/

It is a fairly modern system, maybe you should contact the DELL for support ...

AbdullahMPrograms, 2024-09-15:
That’s unfortunate, I doubt Dell would be of any help here. Are there any other options to getting the rebar driver on the bios?

ZOXZX, 2024-09-15:
Are you sure you don't have BIOS option - by a different name maybe? 
That workstation class supports multiple GPUs and is not cheap so don't hesitate to ask for their support.
Any newer BIOS revisions?

AbdullahMPrograms, 2024-09-15:
No newer bios revisions, what do you mean by BIOS option by a different name?

ZOXZX, 2024-09-15:
I see that you have Above 4G decoding available.
This rebar version is not tested with dual socket boards AFAIK and the author said it probably won't work.
@xCuri0 has some shenanigans to enable rebar (only for Linux I believe), but In your place I would contact Dell first and if they decline your request, ask xCurio to see if he is willing/able to help.

AbdullahMPrograms, 2024-09-15:
I see, I did see that one user was able to enable rebar on a Dell t7810 which is also dual CPU. I'll try reaching out to Dell to see what they say.

xCuri0, 2024-10-05:
Some systems with Bootguard still have a small DXE region which can be manipulated, I think UEFITool NE can show it.

AbdullahMPrograms, 2024-10-05:
oh? would that allow me to add the ReBarDxe on the bootguard bios?

xCuri0, 2024-10-06:
Most likely yes

AbdullahMPrograms, 2024-10-19:
@xCuri0 do you know what this manipulatable dxe region would look like in uefitool or any identifiers I can use to find this region?

xCuri0, 2024-10-19:
The first page of the wiki says how to locate it, your problem might be that the bios image has some sort of compression/encryption which isn't uncommon on prebuilts.

It's best to use a BIOS dump when modding these systems.

AbdullahMPrograms, 2024-10-19:
Yes, I'm working with a dumped bios via a CH341a. I'm currently looking at the bios in uefitool and have noticed that the only regions denoted with a yellow marker are the volume free spaces, would I even be able to insert the dxe region in these locations? The volume containing the pcibus header is entirely cyan which I believe is causing the bootguard issue.

xCuri0, 2024-10-19:
@AbdullahMPrograms Hi read this https://medium.com/firmware-threat-hunting/bypass-intel-boot-guard-cc05edfca3a9

It explains what the colours displayed by UEFITool NE mean.

AbdullahMPrograms, 2024-10-20:
I tried adding the dxe driver to a volume denoted with yellow but that lead to a bootloop, is there any specific address I should be looking to place the dxe driver in? [dumped bin file](https://drive.google.com/file/d/18f1kE-N6D1QYXWuVorXS8gDK1RjfvJnt/view?usp=sharing) in case its helpful

xCuri0, 2024-10-20:
That article says you need to add it to a white area because they don't have bootguard

AbdullahMPrograms, 2024-10-20:
I see, in that case would I be able to insert the dxe driver in say this region?
[image]

xCuri0, 2024-10-20:
Probably, but you will have to try and see to know if it will actually work

AbdullahMPrograms, 2024-10-20:
@xCuri0 unfortunately this resulted in another bootloop, at this point I'm quite desperate to try anything. Do you possibly know of any other way to find which volume I can put the dxe driver into to avoid bootloops? it appears that placing the dxe driver in any volume other then the volume containing the pcibus header causes bootloops

xCuri0, 2024-10-21:
Try other uncolored volumes maybe (they should have no bootguard) ? I wouldn't be suprised if it still didn't work though because the whole point of bootguard is so that no one can add or modify UEFI modules

AbdullahMPrograms, 2024-10-23:
in case it doesnt work, I see above it was mentioned there is a way to enable rebar exclusively in linux. Luckily I only plan to use linux on this PC, how would I go about this linux method?

xCuri0, 2024-10-23:
@AbdullahMPrograms For AMD all you need to do is enable 4G decode

I think NVIDIA drivers on Linux need some option enabled so it can do the resizing but it does work.

AbdullahMPrograms, 2024-10-23:
Sorry just to confirm, do you mean that it is possible to run ReBarState on linux whilst only requiring above 4G decode? I intend to set a max rebar size of 32.

AbdullahMPrograms, 2024-10-23:
Answering my question above, yes its as easy as building and running ReBarState and setting a rebar size of 32. I now have all of my P40's running. Thank you for the support!

closed 2024-10-23 by AbdullahMPrograms

Donalaren, 2025-11-15:
Hello, I am in urgent need of your assistance. I am attempting to use a Tesla V100 16G PCIe graphics card on a Dell Precision 7920T. 

- [x] Memory Map IO above 4GB is enabled.
- [ ] There is no CSM in BIOS of Dell 7920T.
- [x] BIOS version is updated (2.50.0).

I first attempted this on Ubuntu 22.04 using @xCuri0 's ReBarState. Following your method, after building and running RebarState, set the value to 32. But this approach proved ineffective. After rebooting and reopening the RebarState, I can see `the Current ReBarState 32/Unlimited`. `nvidia-smi` still can't see my Tesla V100 (though `lspci | grep -i nvidia` correctly identifies the GPU model `0000:73:00.0 3D controller: NVIDIA Corporation GV100GL [Tesla V100 PCIe 16GB] (rev a1)`). 
Through `dmesg` I can see:
```
[   45.605811] nvidia 0000:73:00.0: enabling device (0140 -> 0142)
[   45.606250] NVRM: This PCI I/O region assigned to your NVIDIA device is invalid:
               NVRM: BAR1 is 0M @ 0x0 (PCI:0000:73:00.0)
[   45.606542] nvidia: probe of 0000:73:00.0 failed with error -1
```
This is consistent with the situation without ReBarState. I have only followed this [method](https://github.com/xCuri0/ReBarUEFI/wiki/Building#rebarstate) to build and run ReBarState on the Ubuntu system without making any modifications to the BIOS. My actions were as follows:
```
git clone https://github.com/xCuri0/ReBarUEFI.git
cd ReBarUEFI/ReBarState
mkdir build
cd build
cmake ..
make
sudo ./ReBarState
```
So I suspect ReBarState may not be functioning. How to verify if the RebarState is working in this scenario? Note that I'm not dismissing the software itself; I believe my execution may be incorrect.