xCuri0/ReBarUEFI #240: Dell BIOS in EXE format

vlerror, 2024-09-18:
Hello, I’ve tried everything to get the EXE format to a bin file and I have made it into a “Bin” file but using uefi tool nothing shows up. I need a way to get the correct image file from my bios (dell optiplex 5050)

ZOXZX, 2024-09-18:
When the file is executed, does it create any temporary files/folders locally or in standard temp places?

vlerror, 2024-09-18:
> When the file is executed, does it create any temporary files/folders locally or in standard temp places?

Well I'm not sure, what method of extraction would lead to the files being the the temp folder? I've used like 3 separate methods in doing such.

ZOXZX, 2024-09-18:
Which CPU generation? 
Please, provide a link to .exe file.

vlerror, 2024-09-18:
> Which CPU generation? 
> 
> Please, provide a link to .exe file.

Dell optiplex 5050
My cpu is an i5-7500
Gpu intel arc a380
16gb 2400MHz

https://www.dell.com/support/home/en-us/product-support/product/optiplex-5050-desktop/drivers
This is the link to the DELL optiplex 5050 drivers and download. The BIOS should be in the categories of driver installs.

DealsBeam, 2024-09-19:
I have the same OptiPlex 5050, here is the latest BIOS

BIOS: https://dl.dell.com/FOLDER11831678M/1/BIOS_IMG.rcv
Version: 1.31.0
Release date: August, 16, 2024

vlerror, 2024-09-19:
> I have the same OptiPlex 5050, here is the latest BIOS
> 
> BIOS: https://dl.dell.com/FOLDER11831678M/1/BIOS_IMG.rcv Version: 1.31.0 Release date: August, 16, 2024

hey so after i download the rcv file how would i modify it in the uefi tool? do i need to make it into a BIN file.

DealsBeam, 2024-09-20:
> > I have the same OptiPlex 5050, here is the latest BIOS
> > BIOS: [dl.dell.com/FOLDER11831678M/1/BIOS_IMG.rcv](https://dl.dell.com/FOLDER11831678M/1/BIOS_IMG.rcv) Version: 1.31.0 Release date: August, 16, 2024
> 
> hey so after i download the rcv file how would i modify it in the uefi tool? do i need to make it into a BIN file.

No idea since I have not looked into this for myself yet

vlerror, 2024-09-20:
> No idea since I have not looked into this for myself yet

Oh, okay thanks anyway. If you figure out anything please let me know if you can.