xCuri0/ReBarUEFI #241: Intel Server Board S2600CW

nalilord, 2024-09-23:
Does anybody tried to run the ReBar patch on an Intel server board? I had a spare Arc 380 and thought it would make a good addition to my ARM setup for video encoding/transcoding, but what an adventure! 
Obliviously the Intel board does not support ReBar out of the box. Flashing a modified capsule BIOS was also a no go because iflash32 deemed it invalid! I had to extract the BIOS with my XGecu T48, modify it, and reflash. But so far this was pretty (surprisingly) straight forward.

Putting the Board from Legacy into UEFI mode -> It still posts but it takes a while to get into the BIOS now when the Arc card is installed.

Now the problem, using ReBarState and settings any value above 512MB causes the system to no longer POST successfully, 1024MB causes a boot loop with a logged system error of "Out of PCI Resources" and everything above 1024MB caused the system to POST hang at code 0x84 which is "DXE PCI Bus Enumeration".

At least I could double the ReBar value and to be honest I don't even know if it is really necessary to increase it more when only using the video encoder part of the GPU, maybe someone has experience?

Also, if someone is interested to look at the patched BIOS, I attached it.
[S2600CW2_REBAR.zip](https://github.com/user-attachments/files/17098256/S2600CW2_REBAR.zip)

nalilord, 2024-10-03:
Yeah, I skipped this step because this is not an AMIBIOS (I believe Intel uses Phoenix?) so no AmiBoardInfo to find and extract. And I have no idea how to find and extract the ACPI/DSDT table otherwise.

xCuri0, 2024-10-05:
POST issues are never caused by DSDT issues, they usually result in the OS crashing/BSOD/broken GPU driver.

I don't think any X99/LGA2011-3 systems need DSDT patching anyways