xCuri0/ReBarUEFI #242: [Z97I-PLUS] Pad problem in v2704 / Pad OK on v2703 but XMP loss

nds2204, 2024-09-23:
**System**
 - Motherboard: ASUS Z97I-PLUS w/ i7-4790K @ 4.5GHz
 - BIOS Version: Tried both 2703 & 2704
 - GPU: Gigabyte RTX 4070 Super Windforce OC
 - [OK] CSM is turned off.
 - [OK] 4G decoding is enabled.
 - [OK] UEFIPatch is applied
 - [OK] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**

I can't solve the Pad problem on BIOS v2704, even with MMTOOL v4.50.0.23 (or higher)

Then I decided to tried with BIOS v2703 >> No pad problem, ReBAR set to 32 (unlimited) with RTX4070(S) (12GB) fully working but... I can't run my RAM above 1333MHz (2x8GB XMP 2400MHz)

XMP working on stock 2703 & 2704.

Please see files attached for comparison.

Thanks !  

[Z97IPLUS BIOS COMPARISON.zip](https://github.com/user-attachments/files/17105303/Z97IPLUS.BIOS.COMPARISON.zip)

nds2204, 2024-09-23:
Sorry, I just saw "No POST, overclocking not working, etc"

nds2204, 2024-09-23:
Actually I didn't work either so I reopen :

> No POST, overclocking not working, etc
> This is caused by corruption from using UEFITool which happens on some firmwares. Redo the guide but use MMTool instead for all steps involving adding/replacing modules so you would use the pad file workaround for all patched modules.

0x6A7232, 2025-03-20:
Apparently you're supposed to extract and save only the modified section of your patched BIOS file using UEFITool and UEFIPatch, and save the modified bit as ffs, then use MMTool to replace that section of your stock BIOS file using the ffs file you created with UEFITool.