xCuri0/ReBarUEFI #245: error when i apply the patches in cmd

ricardohvt, 2024-10-03:
**System**
 - Motherboard: PRIME-H510M-E-ASUS
 - BIOS Version: 05/07/2022 (1601)
 - GPU:
 - [ On ] CSM is turned off. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [ On ] 4G decoding is enabled. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [ No ] UEFIPatch is applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [ Yes ] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

Desc: 
In CMD/PowerShell, i got this error when i apply the patches:
C:\Users\pc\Desktop\UEFIPatch_0.28.0_win32>UEFIPatch PRIME-H510M-E-ASUS-2402rebar.rom
parseImageFile: Aptio capsule signature may become invalid after image modifications
parseBios: one of volumes inside overlaps the end of data
parseBios: one of volumes inside overlaps the end of data
parseBios: one of volumes inside overlaps the end of data
parseFile: non-empty pad-file contents will be destroyed after volume modifications
No patches can be applied to input file
[image]

Also, i got a question too, my original BIOS File is .CAP, when i use UEFITool i got .rom, have any issue about it???

xCuri0, 2024-10-05:
Nothing to apply just ignore the errors

closed 2024-10-05 by xCuri0

ricardohvt, 2024-10-05:
@xCuri0 also, my bios doesnt support .rom files (doesnt show in his file explorer), only .CAP, how i make the bios mod file to .CAP?
Also, my bios got Asus Bios Rollback Support, if i have any issue in my bios, how i do it??