xCuri0/ReBarUEFI #253: Gigabyte X299 UD4 Pro

c4801725870, 2024-12-01:
**System**
 - Motherboard: Gigabyte X299 UD4 Pro
 - BIOS Version: F7c
 - GPU: RTX 3090 TI 24GB
 - [x ] CSM is turned off. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [x ] 4G decoding is enabled. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [ ] UEFIPatch is applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [ x] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**

Is rebar supported on 64GB RAM? I have 8x8GB DDR4 installed and also have a 24GB RTX 3090 TI. I enabled 4G decoding in the BIOS but the EVGA program still shows nonsense about checking with the motherboard manufacturer. 

Any advice about compatibility? I read someone with F8a bios got it working. I got a copy of the .bin file. But I thought I would ask before flashing a random bin file from the internet. Thanks

[image]

ZOXZX, 2024-12-03:
There is Above 4G entry in BIOS, but your board doesn't appear to have ReBar module.
If I've somehow skipped it, check that your onboard VGA (didn't know Intel put those on HEDT) doesn't have 2GB assigned or it will default to A4G off as per warning message ...

c4801725870, 2024-12-03:
> There is Above 4G entry in BIOS, but your board doesn't appear to have ReBar module. If I've somehow skipped it, check that your onboard VGA (didn't know Intel put those on HEDT) doesn't have 2GB assigned or it will default to A4G off as per warning message ...

Thanks for the reply. It does not have iGPU. I would not be surprised if it does not support. The story with this motherboard is BIOS version F6 to F8 had the Above 4G Decoding option. They released F9 and took it away. They also made it impossible to downgrade the BIOS version. Called it "capsule mode" or something on Gigabyte docs. I read some users had it working and then it stopped working when they got F9 bios. 

I have F7c and I can update to F8a if it makes a difference (also downgrade again to F7c) but wanted to check here first about my situation with the 64GB RAM and 24GB VRAM. Everyone I read got it working with less than 32GB RAM and had a 12GB VRAM card. I couldn't find any examples of 64/24.

ZOXZX, 2024-12-03:
I thought you were sticking to F7 because of the security patch ...

F9 has the same options as F9 as far as rebar goes (none that I can see). 
You can enable A4G but you'll need to add a rebar module to your BIOS.
Adding works with UEFItool, but haven't checked for errors!

Anyway, you have dual BIOS board (double check) and If you want to experiment: 
backup your primary BIOS, mod and flash it.

Capsule is not important if you have bios programmer - only when officially downgrading.

c4801725870, 2024-12-03:
> I thought you were sticking to F7 because of the security patch ...
> 
> F9 has the same options as F9 considering rebar (none that I can see). You can enable A4G but you'll need to add a rebar module to your BIOS. Adding works with UEFItool, but haven't checked for errors!
> 
> Anyway, you have dual BIOS board (double check) and If you want to experiment: backup your primary BIOS, mod and flash it.
> 
> Capsule is not important if you have bios programmer - only when officially downgrading.

Ok thanks I will try the mod when I get a chance. I would like to avoid the microcode update due to performance limitations of the branch prediction behavior modification... I learned in 24H2 Windows 11 you can rename GenuineIntel.dll and GenuineAMD.dll (swap the names) to prevent using a microcode downloaded from Windows Update. Before you could just delete the files but now you have to rename/swap them. 

I don't like the idea of using F9 bios so I will stick with modding F7c.