xCuri0/ReBarUEFI #254: Asus Z170-A Bios Help

tbroo94, 2024-12-06:
**System**
 - Motherboard: Asus Z170 - A
 - BIOS Version: 3802
 - GPU: Intel A750
 - [ *] CSM is turned off. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [ *] 4G decoding is enabled. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [* ] UEFIPatch is applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [* ] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**

I'm having a slight issue and I believe it's my own fault for not reading properly/ being an idiot. I've patched the bios after adding ReBarDxe.ffs. I'm unsure how to get the bios back into the .CAP format the bios requires as well I need some guidance flashing the bios after. System is: i5 6500, Asus Z170-A mobo, 32gb ram, intel A750 gpu. I appreciate any help.

ZOXZX, 2024-12-06:
Just rename the extension to cap...

tbroo94, 2024-12-06:
I'm wrong. Manual calls CrashFree  Bios 3. I do apologise about that.

tbroo94, 2024-12-06:
Manual says it's meant to. But there is no such switch on the back io panel

ZOXZX, 2024-12-06:
No need to apologize, I was just trying to point you to Flashback feature on your MB.
But... 
After taking another look, it does have flashback feature, but only through internal connector on MB.
"Flash card" for that is sold separately it seems...

tbroo94, 2024-12-06:
I can't seem to find said card either. Which is very odd.

exalented, 2025-04-06:
@tbroo94 Did you ever get this setup? I got stuck here a while back as well.

tbroo94, 2025-04-06:
@exalented I have not. Didn't have enough time to pick up CH341 tool. As that particular mobo was for a pc as a Christmas present. Maybe try what @ZOXZX has mentioned to see if that helps.