xCuri0/ReBarUEFI #255: Thinkstation P510 BIOS extraction issues

MrMupfin, 2024-12-12:
Hi, I am unable to extract a usable BIOS file from the Lenovo BIOS updater for the P510 Thinkstation. Anyone out here who can take a look at this together with me? The closest I could find was a File simply labelled CEF5B9A3 located within IMAGES00.ROM. But UEFITool doesn´t know what to do with it. HxD reports the first 4 bytes as being 4E 56 41 52 which would mean the file is likely an NVAR type file.

ZOXZX, 2024-12-14:
Tried opening images00.rom and both uefitool and NE version read the bios file ...

LordLeland, 2024-12-27:
hello, I'm a P510 user too, u need a programmer like CHA341A to continue the procedure - load the origin BIOS, hack it then flash it back.
u cannot do it through official BIOS updater cause a secure lock prevent modified bios check in.

tylerebowers, 2025-05-26:
> hello, I'm a P510 user too, u need a programmer like CHA341A to continue the procedure - load the origin BIOS, hack it then flash it back. u cannot do it through official BIOS updater cause a secure lock prevent modified bios check in.

@LordLeland Do you still have that p510? Looking for a bin of the bios. Trying to restore my system and I cant find a read anywhere.

LordLeland, 2025-06-11:
@tylerebowers 
https://winraid.level1techs.com/t/request-modded-lenovo-thinkstation-s30-type-0606-bios/103361/25
bottom of this topic has what u need