xCuri0/ReBarUEFI #258: Resizable BAR disabled (ASUS P8Z77-V LX, i7 3770K, Ivy Bridge)

simrem, 2024-12-18:
**System**
 - Motherboard: Asus P8Z77-V LX
 - CPU: Intel i7 3770K (Ivy Bridge)
 - Memory: 32GB (also tried with 16GB)
 - Storage: Intel SATA Raid 0 (2xSATA6)
 - BIOS Version: 2501
 - GPU: Intel Arc B580
 - [V] CSM is turned off.
 - [V] 4G decoding is enabled.
 - [V] UEFIPatch is applied 
 - [V] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**
I have tried to follow the guides to enable Resizable BAR, but it seems that I need some assistance :)

Here are the steps that I've done:
- Used the UEFI Tool method to add the ReBarDxe.ffs
- UEFI Patch
```
PS C:\Temp\BIOS\02_uefipatch> .\UEFIPatch.exe .\P8Z77-V-LX-ASUS-2501_rebardxe.CAP
parseImageFile: Aptio capsule signature may become invalid after image modifications
patch: replaced 29 bytes at offset EDBh 77B6488B0F493BCF73AE48FFC1E81BFFFFFF488B1748FFC8483BD0759B -> 669066909066909066906690909090669090488B176690906690906690
patch: replaced 29 bytes at offset 13E3h 48BA000000000100000049B80000000004000000483BDA8BCE480F47D3 -> 49B800000000100000004929D8909090909090906690908BCE4889DA90
patch: replaced 11 bytes at offset 8A88h B8FFFFFFFF030000004C3B -> B8FFFFFFFFFFFFFF004C3B
patch: replaced 17 bytes at offset 7EBAh B9FFFFFFFF490FAFC14903C0483BC1776C -> 6690669090490FAFC14903C06690906690
patch: replaced 17 bytes at offset 7F96h B9FFFFFFFF490FAFC14903C0483BC1776C -> 6690669090490FAFC14903C06690906690
patch: replaced 7 bytes at offset 1795h 493B4C24C0771B -> 66906690669090
patch: replaced 7 bytes at offset 183Dh 493B4C24C0771C -> 66906690669090
Image patched
PS C:\Temp\BIOS\02_uefipatch> .\UEFIPatch.exe .\P8Z77-V-LX-ASUS-2501_rebardxe.CAP.patched .\IvyUSB3.txt -o P8Z77-V-LX-ASUS-2501_rebardxe_usb3.CAP.patched
parseImageFile: Aptio capsule signature may become invalid after image modifications
patch: replaced 13 bytes at offset DFD4h DE10D301FFFF00000B00000014 -> 8680311EFFFF00000B00000010
Image patched
```
- DSDT patch using this guide: https://github.com/xCuri0/ReBarUEFI/issues/11#issuecomment-2544176811
```                
Else
{
   CreateQWordField (BUF0, \_SB.PCI0._Y0F._LEN, M2LN)  // _LEN: Length
   CreateQWordField (BUF0, \_SB.PCI0._Y0F._MIN, M2MN)  // _MIN: Minimum Base Address
   CreateQWordField (BUF0, \_SB.PCI0._Y0F._MAX, M2MX)  // _MAX: Maximum Base Address
   M2MX = 0xFFFFFFFFF
   If ((TUUD >= 0x1000))
   {
      M2MN = (TUUD << 0x14)
   }
   Else
   {
      M2MN = 0x0000000100000000
   }
   M2LN = (M2MX - M2MN) + One
}
``` 
- Unlock hidden "Above 4G decoding" using AMIBCP v4
- Reset BIOS to default settings
- Flashed the BIOS by using AFUDOS after removing the Aptio Capsule header. The Asus EZ Flash Utility gave me a "Security verification failed".
- Disabled CSM
- Used grub-mod-setup_var to enable Above 4G encoding. After this the Large Memory section is visible in device manager in Windows.

So this is where I am now:

GPU-Z shows that Resizable BAR is disabled
[image]
I have tried RebarState with different values (2GB, 4GB and unlimited), but it has no effect. I have also tried to reduce the RAM from 32GB to 16GB.

Device manager looks like this:
[image]

Intel Graphics Software crashes when I click on the GPU
[image]
Error event from Event Viewer
[image]

I don't know where to go from here, and some help is appreciated :)
Maybe someone already has a modded BIOS for my board?

03vmate, 2024-12-18:
Unfortunately I cannot help, my use case was strictly compute and only on Linux, so I did not actually need ReBAR, just enough space for Linux to allocate a sufficiently large BAR(explained in https://github.com/xCuri0/ReBarUEFI/issues/77#issuecomment-1684007504)

simrem, 2024-12-18:
It appears that the modified BIOS has reverted to the original version, 2501.

I used the following command: `afudos.exe bios.rom /GAN`
Without rebooting, I verified that the flashed image contained the ReBarDxe section by running: `afudos.exe backup.rom /O`
However, after rebooting, I performed the same verification and found that the ReBarDxe section was missing.

I also attempted to run:
`afudos.exe bios.rom /P /B /N /GAN`
but encountered the following error: 
`18 - Error: Unable to start a Secure Flash session` error message. 

The file was checked for padding issues. Does anyone have suggestions on how to proceed?

03vmate, 2024-12-18:
I had a lot of trouble with flashing too, so ended up buying a cheap EEPROM flasher and used that. It saved a bunch of time too, and cost about 5 bucks.

xCuri0, 2024-12-19:
@simrem You need to use BIOS Flashback.

I'm interested in how B580 performs with Ivy Bridge so do update if possible.

simrem, 2024-12-19:
Thank you for the feedback! I followed this guide and successfully flashed the BIOS: https://tomvanveen.eu/flashing-bios-chip-raspberry-pi/

With 16GB of RAM and ReBarState set to "unlimited," GPU-Z now reports a 16GB BAR size:
[image]

The computer would not boot with 32GB of RAM installed. However, reducing the RAM to 24GB worked. Additionally, setting ReBarState to 8GB and installing 32GB of RAM also worked.
[image]

What would be the recommended configuration in my case. Prefer BAR size or RAM size?

@xCuri0 I have not tested the performance yet, but I will give you an update later.

03vmate, 2024-12-19:
The B580 will almost certainly utilize a BAR larger than 8 gigs if it can, so I would personally go for 24GB RAM + 16GB BAR, but you should benchmark some workloads to see how much of a difference it makes.

ArtemioUrbina, 2024-12-24:
@simrem Such a coincidence, I was looking into enable Resizable BAR for this exact same motherboard. Would you be so kind as to share the BIOS you flashed? Thanks for sharing this information.

simrem, 2024-12-31:
> @simrem Such a coincidence, I was looking into enable Resizable BAR for this exact same motherboard. Would you be so kind as to share the BIOS you flashed? Thanks for sharing this information.

Here you go:
[P8Z77-V-LX-ASUS-2501_rebar.zip](https://github.com/user-attachments/files/18281083/P8Z77-V-LX-ASUS-2501_rebar.zip)

Be aware that afudos did not work for me.

ArtemioUrbina, 2024-12-31:
Thank you @simrem. I had patched mine following teh steps, but I probably made some mistakes. Will try yours soon. I use an EPROM programmer so I am covered there. Thank you again, and happy new year

GianniDPC, 2025-01-01:
> Here you go: [P8Z77-V-LX-ASUS-2501_rebar.zip](https://github.com/user-attachments/files/18281083/P8Z77-V-LX-ASUS-2501_rebar.zip)
>
> Be aware that afudos did not work for me.

What BAR size does this one have? Also does anyone know any other way to successfully flash it without using a programmer? So using EZ Flash from within a working bios wouldn't work too? (Sorry if I am asking stupid questions I'm new to this)

simrem, 2025-01-02:
> @simrem You need to use BIOS Flashback.
> 
> I'm interested in how B580 performs with Ivy Bridge so do update if possible.

@xCuri0 We've tried games like Rocket League, Fall Guys, Roblox, and Fortnite. Rocket League runs smoothly on high settings at 60 FPS. However, in Fortnite, the CPU hits 100% usage when video settings are set to low at 1080p. The CPU is overclocked to 4.2 GHz and installed RAM is 16GB. But it's still playable. At the moment I'm using an old 4k 60hz monitor, I'll try to change it out with a 1440p 144hz monitor.

closed 2025-01-06 by xCuri0