xCuri0/ReBarUEFI #260: FPS drop on rx 470 4gb

ExLimeX, 2024-12-27:
**System**
 - Motherboard: China 2011 PlexHd x79
 - CPU: Xeon 2689 (Sandy Bridge)
 - BIOS Version: -
 - GPU: RX 470 4GB
 - CSM is turned off. 
 - 4G decoding is **disable**. Black screen when 4G decoding is on, tried to make a patch, but didn't figure out how to edit DSDT correctly.
 - [Y] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**

BAR Size is 1gb, when I turn on rebar, the frame rate becomes lower.

Cyberpunk off: 32 fps

Cyberpunk on: 31 fps

RDR2 on: 50 fps (2.6 gb vram use)

RDR2 off: 59 fps (3.2 gb vram use)

David112x, 2024-12-27:
As far as I know, RDR2 has issues managing high BAR sizes, probably why the Nvidia drivers limit the usable BAR size for the game.

As for cyberpunk, well that's not enough information to truly tell if this is a ReBar issue, since a one frame difference seems to be within margin of error and not all games benefit from ReBAR unless they're being bottlenecked by the GPU's memory and BAR size limitations.

ExLimeX, 2024-12-27:
New test:

Far Cry 5 off: 41 fps

Far Cry 5 on: 41 fps

Cyberpunk off: 51.07 fps

Cyberpunk on: 50.80 fps

Half Life 2 off: 262 fps

Half Life 2 on: 264 fps

Maybe it's because of the 1GB buffer, I can't put more, but I couldn't find a game where there would be at least some increase. Fps is either the same or worse.

I would like to see the results of other users with 4 GB cards and a buffer of 1-2 GB.

xCuri0, 2025-01-06:
This is an issue with the AMD driver nothing I can do about it, for some reason 4GB Polaris doesn't work with 4GB BAR requiring use of a smaller size which causes all kinds of issues