xCuri0/ReBarUEFI #261: ReBar on Datto S4P2143/ASRock D2143D8UM not working

Tualua, 2024-12-29:
**System**
 - Motherboard: Datto S4P2143/ASRock D2143D8UM
 - BIOS Version: 1.11
 - GPU: Intel A380 (Sparkle Pioneer)
 - [ x] CSM is turned off. 
 - [x] 4G decoding is enabled. 
 - [x] UEFIPatch is applied. **I says nothing to patch**
 - [x] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**
Hello! Thank you for your amazing efforts to create ReBar support for older boards.
I'm trying to patch BIOS for my Datto/ASRock board. I can patch it with both UEFITool/MMTool, although I had to use newer MMTool version. The board starts to boot, throws some POST codes on the screen and then just shuts itself down. 

Because I will be using this setup only for Linux I tried to enable ReBar without BIOS mod and it does not work.
RebarState says it is enabled

```
ReBarState (c) 2023 xCuri0

Current ReBarState 12 / 4096 MB
```

But i915 driver says
```
[   72.186179] i915 0000:19:00.0: BAR 0 [mem 0xc4000000-0xc4ffffff 64bit]: releasing
[   72.194159] i915 0000:19:00.0: BAR 2 [mem 0x27fe0000000-0x27fefffffff 64bit pref]: releasing
[   72.315451] i915 0000:19:00.0: BAR 2 [mem size 0x200000000 64bit pref]: can't assign; no space
[   72.334537] i915 0000:19:00.0: BAR 2 [mem size 0x200000000 64bit pref]: failed to assign
[   72.344010] i915 0000:19:00.0: BAR 0 [mem 0xc4000000-0xc4ffffff 64bit]: assigned
[   72.426964] i915 0000:19:00.0: [drm] Failed to resize BAR2 to 8192M (-ENOSPC)
[   72.435493] i915 0000:19:00.0: BAR 2 [mem 0x27fe0000000-0x27fefffffff 64bit pref]: assigned
[   72.445266] i915 0000:19:00.0: [drm] Using a reduced BAR size of 256MiB. Consider enabling 'Resizable BAR' or similar, if available in the BIOS.
```

I tried to patch DSDT, checked root bus resources

```
# sudo dmesg | grep "root bus resource"
[    2.936635] pci_bus 0000:00: root bus resource [io  0x0000-0x03af window]
[    2.943633] pci_bus 0000:00: root bus resource [io  0x03e0-0x0cf7 window]
[    2.950635] pci_bus 0000:00: root bus resource [io  0x03b0-0x03bb window]
[    2.957639] pci_bus 0000:00: root bus resource [io  0x03c0-0x03df window]
[    2.963635] pci_bus 0000:00: root bus resource [io  0x1000-0x3fff window]
[    2.970633] pci_bus 0000:00: root bus resource [mem 0x000a0000-0x000bffff window]
[    2.978634] pci_bus 0000:00: root bus resource [mem 0x000c4000-0x000c7fff window]
[    2.985633] pci_bus 0000:00: root bus resource [mem 0xfe010000-0xfe010fff window]
[    2.993635] pci_bus 0000:00: root bus resource [mem 0x90000000-0xaaffffff window]
[    3.000637] pci_bus 0000:00: root bus resource [mem 0x20000000000-0x23fffffffff window]
[    3.008636] pci_bus 0000:00: root bus resource [bus 00-15]
[    3.657635] pci_bus 0000:16: root bus resource [io  0x4000-0x7fff window]
[    3.664633] pci_bus 0000:16: root bus resource [mem 0xab000000-0xc5ffffff window]
[    3.672634] pci_bus 0000:16: root bus resource [mem 0x24000000000-0x27fffffffff window]
[    3.680634] pci_bus 0000:16: root bus resource [bus 16-63]
[    4.471634] pci_bus 0000:64: root bus resource [io  0x8000-0xbfff window]
[    4.478633] pci_bus 0000:64: root bus resource [mem 0xc6000000-0xe0ffffff window]
[    4.485643] pci_bus 0000:64: root bus resource [mem 0x28000000000-0x2bfffffffff window]
[    4.493634] pci_bus 0000:64: root bus resource [bus 64-b1]
[    4.943634] pci_bus 0000:b2: root bus resource [io  0xc000-0xffff window]
[    4.950633] pci_bus 0000:b2: root bus resource [mem 0xe1000000-0xfbffffff window]
[    4.957635] pci_bus 0000:b2: root bus resource [mem 0x2c000000000-0x2ffffffffff window]
[    4.965634] pci_bus 0000:b2: root bus resource [bus b2-ff]
```

Can't get it - does it qualifies as _fffffffff or above_ or not.
Extracted and decompiled DSDT.
Searched last DWordMemory. Found that I already have QWordMemory 

```
QWordMemory (ResourceProducer, PosDecode, MinFixed, MaxFixed, NonCacheable, ReadWrite,
                    0x0000000000000000, // Granularity
                    0x0000000000000000, // Range Minimum
                    0x0000000000000000, // Range Maximum
                    0x0000000000000000, // Translation Offset
                    0x0000000000000000, // Length
                    ,, , AddressRangeMemory, TypeStatic)
```
And I have a lot of QWordMemory  entries

Tried to search for M1LN - no entries.

Am I out of luck with this board or there is a chance to fix it?

Tualua, 2024-12-29:
Tried xe driver:

```
[   10.789723] xe 0000:19:00.0: [drm] Attempting to resize bar from 256MiB -> 8192MiB
[   10.800199] xe 0000:19:00.0: BAR 2 [mem 0x27fe0000000-0x27fefffffff 64bit pref]: releasing
[   10.809842] xe 0000:19:00.0: [drm] Finished loading DMC firmware i915/dg2_dmc_ver2_08.bin (v2.8)
[   10.923219] xe 0000:19:00.0: BAR 2 [mem size 0x200000000 64bit pref]: can't assign; no space
[   10.923222] xe 0000:19:00.0: BAR 2 [mem size 0x200000000 64bit pref]: failed to assign
[   11.021461] xe 0000:19:00.0: [drm] Failed to resize BAR2 to 8192M (-ENOSPC). Consider enabling 'Resizable BAR' support in your BIOS
[   11.034674] xe 0000:19:00.0: BAR 2 [mem 0x27fe0000000-0x27fefffffff 64bit pref]: assigned
[   11.042914] xe 0000:19:00.0: [drm] VISIBLE VRAM: 0x0000027fe0000000, 0x0000000010000000
[   11.052678] xe 0000:19:00.0: [drm] Small BAR device
[   11.058568] xe 0000:19:00.0: [drm] VRAM[0, 0]: Actual physical size 0x0000000180000000, usable size exclude stolen 0x000000017c800000, CPU accessible size 0x0000000010000000
[   11.074021] xe 0000:19:00.0: [drm] VRAM[0, 0]: DPA range: [0x0000000000000000-180000000], io range: [0x0000027fe0000000-27ff0000000]
[   11.085939] xe 0000:19:00.0: [drm] VRAM: 0x0000000180000000 is larger than resource 0x0000000010000000
[   11.095259] xe 0000:19:00.0: [drm] Total VRAM: 0x0000027fe0000000, 0x0000000180000000
[   11.103105] xe 0000:19:00.0: [drm] Available VRAM: 0x0000027fe0000000, 0x000000017c800000
[   11.242173] xe 0000:19:00.0: [drm] vcs1 fused off
[   11.247773] xe 0000:19:00.0: [drm] vcs3 fused off
[   11.252474] xe 0000:19:00.0: [drm] vcs4 fused off
[   11.257180] xe 0000:19:00.0: [drm] vcs5 fused off
[   11.261888] xe 0000:19:00.0: [drm] vcs6 fused off
[   11.266594] xe 0000:19:00.0: [drm] vcs7 fused off
[   11.271299] xe 0000:19:00.0: [drm] vecs2 fused off
[   11.276092] xe 0000:19:00.0: [drm] vecs3 fused off
[   11.280882] xe 0000:19:00.0: [drm] ccs0 fused off
[   11.285589] xe 0000:19:00.0: [drm] ccs2 fused off
[   11.290295] xe 0000:19:00.0: [drm] ccs3 fused off
[   11.322830] [drm] Initialized xe 1.1.0 for 0000:19:00.0 on minor 0
[   11.379999] xe 0000:19:00.0: [drm] Cannot find any crtc or sizes
[   11.413974] xe 0000:19:00.0: [drm] Cannot find any crtc or sizes
```

OS: Fedora Server 41, kernel 6.12

Tualua, 2024-12-30:
I recorded a couple of videos of boot process

https://github.com/user-attachments/assets/fc7ce820-142e-49ff-9c31-f2ddb31b106f

https://github.com/user-attachments/assets/2eae8b02-8eed-45ed-9fd2-87cc7371a97f

0dragosh, 2025-09-27:
has anyone been able to turn on rebar for this board? it's basically unusable for me in this state

Tualua, 2025-09-27:
Mine works fine