xCuri0/ReBarUEFI #275: Asus Z10PE-D8 WS with Arc A770 (working) 💪

savvadesogle, 2025-02-21:
**System**
 - Motherboard: Z10PE-D8 WS
 - BIOS Version: 4301
 - GPU: Intel Arc A770 (and AMD RX 7900 XTX)

**Description**

Hi. THANK YOU FOR ReBar dxe driver! ❤️❤️❤️
I confirm that on Windows 11 with 2x A770 on this configuration the rebar works.

8x 16gb (128 GB )
2x Xeon 2699V3
2x A770 pci 1 (x16) , pci 7 (x16) 
wifi pci module pci 6 (x8)
nvme 512gb m.2 (x4)

_You should change the access to USER via AmiBCP_
IntelRCSetup

Common RefCode Configuration
MMCFG Base: 3G
MMIOH Base 56T
MMIO High Granularity Size (MMIO High Size): 1024G

Miscellaneous Configuration
PCIe Max Read Request Size: AUTO

IIO - > near pci x16 slot (that was trained)
Maybe it matters "Enable SPLIT BARs"

PCI-E ASPM disabled (does it make sense?) 

**ReBarState -> 14**

Do no flash "4301-body-nomcu-amibios-rebar_turbo303030.dmp". This bin is just for preview settings in AmiBCP (For inspiration/example).

[image]
[image]
[image]
[image]
[image]

[4301-body-nomcu-amibios-rebar_turbo303030.dmp](https://github.com/user-attachments/files/18915119/4301-body-nomcu-amibios-rebar_turbo303030.dmp)

savvadesogle, 2025-02-26:
Update 1
ReBAR on 3x Arc A770 (x16) working on this board Z10PE-D8 WS. 

ps I wanted to install a fourth card, but unfortunately the riser was buggy.

[image]

savvadesogle, 2025-03-12:
Update 2
ReBAR on 4x Arc A770 (x16) working on this board Z10PE-D8 WS.

[image]

romanescugxg, 2025-10-07:
Hi, I have also Z10PE-D8 WS with dual Xenon 2687W and I woul like to put an A770 gpu. can you help be with the moded bios , please ? 
Thanks 🙏

savvadesogle, 2025-10-07:
> Hi, I have also Z10PE-D8 WS with dual Xenon 2687W and I woul like to put an A770 gpu. can you help be with the moded bios , please ? Thanks 🙏

Hello, you can try this instruction (just translate from russian). 
https://github.com/xCuri0/ReBarUEFI/issues/216#issuecomment-2676334518
ps I also recommend buying a separate BIOS chip so that you can leave the original working bios and not touch it.