xCuri0/ReBarUEFI #286: Tried to edit Aorus Maximus Hero VIII bios.

Xethy, 2025-04-04:
I have a .cap file. 
I got a i7 6700k
I tried Uefipatch: pad-file issue.
I tried MMtool: pad-file issue.
I tried the method to edit hex code.
In the hex editor HxD and LmHex: it failed because 
these cannot find either the changed hex code or the original hex code in the patched or unpatched file. I tried little debian and large debian versions of the hex code. None of both worked.
UEFI-Tool can find the Hex code with no issue.

Can some pc magican tell me what i am doing wrong?

Xethy, 2025-04-05:
I found a bit more about the issue; it seems that the driver is LZMA compressed. Therefore with the compression in place its not possible to edit the code in plain binary text. 
I found where the LZMA compressed area is exactly but as long as i dont know how to change the compressed part accuratly with the right compression and more i am stuck.

Xethy, 2025-04-07:
Let me summarize what i've learned so far

It seems that the LZMA compressed version with rebar enabled is exactly (3 x 8) 24 bytes longer  than the LMZA compressed not rebar version. AKA i am stuck with this 24 byte longer version of my bios. - No hex hex edit can help.
Doesnt aound too bad though? Does it?

Well somehow it seems to be related to the pad file though. In the bios i am editing the padfile and a following microcode build a unit and the microcode seems to need to be aligned to a 16 bit basis (at least thats what i've figured out from UEFItoolNE and editing hex a lot and Using the do not rebuild option on all pad files in my bios with uefitool2.8)

As the pad file seems unimportant the system alligns the microcode to a 16 basis and erases the padfile that way. 

Possible solutions:

Add a driver i dont need which gives the system another uneven 16xY 8x1 bytes more

Just ignore this and edit the microcode ignoring its 16 bit allignment. (I got that bios - i get allignment warnings though)

Also on a sidenote:

You can tell UEFI tool to not rebuild certain areas. pad files included. - if it detects a microcode that is not in a 16 bit allignment it adds another pad file then to get it to a 16 bit allignment. - aka with an edited bios youll then have 2 pad files?

No clue if that bios would boot.

Well at this point i am stuck. I exited around but i am not brave enough to test around with my only bios.

Xethy, 2025-04-12:
I think i got another option to get a running bios;

One of the patches applied to my bios is:

# PciBus | Remove <64GB BAR size limit (Skylake/Kaby Lake/Coffee Lake)
3C1DE39F-D207-408A-AACC-731CFB7F1DD7 10 P:B800000000100000004C3B:B8FFFFFFFFFFFFFF004C3B

I looked at this with ghidra, and i think even though i dont totally understand this... B8FFFFFFFFFFFF00004C3B could work out as well.
This might reduce the compressed LMZA archive by 2 byte, which then in turn cause the archive has a 2 byte overlap makes the whole thing 8 byte smaller.

Aka, i'd only have a 16 byte change and the padfile would not disappear.

xCuri0, 2025-04-12:
@Xethy you can submit a PR for this if you can confirm doing this fixes the LZMA compression issue. I just used the max uint64 value didn't think about compression when making it. Try and find the largest value that compresses properly

Xethy, 2025-04-12:
@xCuri0 Thank you for responding!
I am still thinking about how to go at this, cause i am very scared to brick my motherboard.
I kind of need my computer the next two weeks, so i wont do anything crazy rn. Ill order some Programmer and a clamp from amazon to be safe even if it fails. Id rather not have it fail at all though. No dual bios here, so i guess a brick is a brick without a suitable programmer.

Seems the programmer will take at least until end of may to get here (from china most likely, so i guess. i will wait until then, before telling you more.)

Since i don't know if i have the newest version of bios already installed. Ill first will install the newest bios (which i patched) and the install the patched bios over that... Should give me the highest chance of success.

So for the values, i can use both:  B8FFFFFFFFFFFF00004C3B and B800000100000000004C3B without generating a pad file issue.
I think the most can be learned if i actually go with the 1 TB maximum bar size. If that works, others who also have boards with compressions would know, that anything between:
0000010000000000 FFFFFFFFFFFFFF00 can give them the desired result and adjust the value to whatever they like. In my case the compression extends by a byte with every F but with none with every 0 they can try different bars between these and see if a pad file issues occurs.

Do you also think this would be the best option?

Xethy, 2025-07-19:
Let me tell you what exactly happened for me.
I remembered that i had a old MB lying around for the i7-6700k so i went ahead for the bios edit without the chip. First i installed the newest bios version. - As i tried to install the edit the Software didn't allow for the edit to be installed. I couldn't verify why, but i guess it might be because i updated the bios to the newest version, maybe it has a protection for installing the same bios over an existing version. (As this is what modders would do)
Afterwards a huge work related issue came in and i couldn't make do without working at home too, so i kind of haven't made much more progress with it yet. Sorry.

SleepInfinity, 2025-08-07:
I guess i am in the same situation here, my board is the Asus P8Z68 Deluxe Gen 3 and my issue is #311