xCuri0/ReBarUEFI #289: MSI Z97 Gaming 5 failure

NHogweed, 2025-04-10:
- BIOS Version: [original](http://www.station-drivers.com/download/msi/E7917_1E2.zip), [modified](https://github.com/user-attachments/files/19680543/E7917IMS_1E2_p.zip)
 - GPU: ARC B580, Radeon RX 580X
 - [v] CSM is turned off. I guess setting boot mode to UEFI only corresponds to that.
 - [v] 4G decoding is enabled like described [here](https://github.com/xCuri0/ReBarUEFI/wiki/Enabling-hidden-4G-decoding). It caused video output to dissapear until loading of an OS driver.
 - [v] UEFIPatch is applied. Both standard and BdwUSB3.
 - [v] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

Result: system refuses to boot with BAR size above 256 MB.

dripsnek, 2025-05-05:
I've recently got this working on a MSI Z97M-G43 using basically the same process you described. Maybe check the following?

- When disabling CSM mode I did not touch the boot menu. Instead I enabled "Windows 8/8.1/10 feature" under Advanced->Windows 8/8.1/10 Configuration section. Doesn't matter if you are using Windows or not. This automatically makes the appropriate changes, doing anything else did not work for me. Ensure this is set before enabling above 4G decoding to get video during boot.
- In my case,  Radeon RX 480 (8GB) under Linux does not need a BIOS flash, simply enabling the "hidden" above 4G decoding is enough.
- In my case,  Intel Arc A770 could only get 16GB allocated by flashing a modified BIOS using the guide exactly. Haswell directions, no DSDT issues, no padding issues. Must use ReBarState (once) to set the BAR value, in my case 14 (16384 MB).
- Like all other Haswells, 32GB RAM + 16GB BAR is no POST. Must have < 32GB RAM installed for this or a reduced ReBAR. 32GB + 8GB BAR (say, RX 480) is OK.

NHogweed, 2025-05-05:
> * I enabled "Windows 8/8.1/10 feature" under Advanced->Windows 8/8.1/10 Configuration section.

Yes, this seems to make ReBAR working with ARC B580, but breaks video output in BIOS. Not completely, but enough to make its GUI unusable. Thank you anyway.

closed 2025-05-05 by NHogweed

xCuri0, 2025-11-06:
@dripsnek Could you try replacing `PciHostBridge` in your BIOS with this by doing "Replace Body" on PE32 Image in UEFITool ? The other ReBarUEFI patches and DSDT modification (if applicable) need to be applied too which I assume you have already done.

[Z97PciHostBridgeMod.zip](https://github.com/user-attachments/files/23399592/Z97PciHostBridgeMod.zip)

Modified it to use almost the entire 512GB physical address space instead of just 64GB. This should solve many of the issues on LGA1150 systems with 16GB BAR and/or 32GB RAM.

[image]

Send the output from `MMIOPrint` and also either Windows Device Manager showing "Resources By Type" -> "Large Memory" or Linux `/proc/iomem` output as root after enabling 4G decoding (and CSM off) on the modded BIOS. Don't think you need a rebar capable GPU installed to test it.

Based on BIOS version `E7924IMS.1C0` for `MSI Z97M-G43`.

If it works I can create a UEFIPatch that should work on all LGA1150 boards.

xCuri0, 2025-11-07:
Thx for testing it, this confirms the issue I thought was there.

For some reason on Haswell/Broadwell the GCD memory space map only goes upto 64GB (36-bit) even though the CPU supports upto 512GB (39-bit) addressing, likely because code was re-used from Sandy/Ivy Bridge which only support 36-bit. Skylake and above don't have this issue.

https://uefi.org/specs/PI/1.8/V2_Services_DXE_Services.html seems like it is passed from the PEI stage ?
https://uefi.org/specs/PI/1.8/V2_DXE_Foundation.html

dripsnek, 2025-11-08:
> [@dripsnek](https://github.com/dripsnek) Try this patch
> 
> ```
> # NBPEI | Correct EFI_HOB_TYPE_CPU SizeOfMemorySpace from 36-bit to 39-bit (Haswell)
> 79AA6086-035A-4AD9-A89A-A6D5AA27F0E2 10 P:8B45FCC64008248B45FCC6400910:8B45FCC64008278B45FCC6400910
> ```
 I think we're on to a winner!

I regenerated and flashed the firmware as before, inserting the above into patches.txt and using the modified PciHostBridge PE32.

Now with 32GB RAM installed -

- Above 4G decoding hidden setting works again
- ReBarState works as expected, 32 (unlimited) seems to be fine and 14 also works.
- A770 gets its full 16GB BAR & RX480 its 8
- Sleep works (though it always did for this board)

Now if only software can fix the missing CLKREQ# pin on the Z97 PCH that prevents PEG L1 substates from working...

All logs are 32GB + Arc A770.

[NOV-B-E7924IMS.1C1.zip](https://github.com/user-attachments/files/23429846/NOV-B-E7924IMS.1C1.zip)
[nov-b-mmio.txt](https://github.com/user-attachments/files/23429830/nov-b-mmio.txt)
[nov-b-lspci.txt](https://github.com/user-attachments/files/23429831/nov-b-lspci.txt)
[nov-b-iomem.txt](https://github.com/user-attachments/files/23429833/nov-b-iomem.txt)

[image]
[image]