xCuri0/ReBarUEFI #307: Can't turn on Above 4g Decoding on P8Z77-V LX

GrabOid89, 2025-07-24:
Motherboard: ASUS p8Z77-V LX
BIOS: P8Z77-V-LX-ASUS-2501

What I've done:
1. Unhide PCI subsystem sttings "Above 4g decoding" menu with AMIBCP tool;
2. Added FFS module with MMTool method;
3. Used UEFIPatch (patch.txt + IvyUSB3.txt) with UEFIPatch;
4. DSDT Patching - skiped;
5. Flshed BIOS with AI Suite;

And windows won't start in result if "Above 4g decoding" turned on, if it's not windows starts...
What I'm doing wrong? 

[image]

[image]

[P8Z77-V-LX-ASUS-2501.zip](https://github.com/user-attachments/files/21420081/P8Z77-V-LX-ASUS-2501.zip)

xCuri0, 2025-07-26:
You probably need DSDT patch, not sure why you skipped it because this is the exact issue you get if you skip it.

closed 2025-07-26 by xCuri0

GrabOid89, 2025-07-26:
> You probably need DSDT patch, not sure why you skipped it because this is the exact issue you get if you skip it.

Thanks fo reply

I skipped it because I've checked large memory resource ends on FFFFFFFFF

[image]

Looks like I missed that 81EFFFFFF is less tnan FFFFFFFFF

I'll try DSDT patch tomorrow and give you feedback