xCuri0/ReBarUEFI #308: The system does not detect the resizebar enabled in the BIOS.

Areqyz1337, 2025-07-27:
**System**
 - Motherboard: Msi B450  Gaming plus max
 - BIOS Version: 7B86vHN (latest)
 - GPU: Evga 1660 super
 - [ Yes ] CSM is turned off. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [ Yes ] 4G decoding is enabled. **Make sure to confirm this in the BIOS and not with GPU-Z or similar since it can be inaccurate**
 - [ not because errors keep popping up ] UEFIPatch is applied (see [Using UEFIPatch](https://github.com/xCuri0/ReBarUEFI/wiki/Using-UEFIPatch) for more information). On some motherboards [DSDT Patching](https://github.com/xCuri0/ReBarUEFI/wiki/DSDT-Patching) is also needed
 - [ yes ] I have read [Common issues (and fixes)](https://github.com/xCuri0/ReBarUEFI/wiki/Common-issues-(and-fixes))

**Description**
I had to submit ReBar Dxe.ffs twice because there were two results in the UEFITool program. I did everything the same as in the guide except UEFIPatch because error kept popping up

C:\Users\Rysiek\Desktop\xcvf>UEFIPatch.exe 7B86vHNReBar.rom
parseFile: non-empty pad-file contents will be destroyed after volume modifications
parseFile: non-empty pad-file contents will be destroyed after volume modifications
No patches can be applied to input file.

the bios was installed correctly, the system boots normally.
I installed the nvidia driver but the error still persists. Do you have any ideas on how to fix it?

[image]

Areqyz1337, 2025-08-02:
doesnt work too

xCuri0, 2025-08-02:
Ask there then, this issue doesn't relate to RebarUEFI