xCuri0/ReBarUEFI #316: TUF GAMING X570-PLUS (WI-FI) Rebar 5600 XT not working

chris1892006, 2025-09-15:
Hi I set the rebarstate to 512MB then to 2048MB, no changes took place.
Advice?
[image]

Koekieezz, 2025-09-20:
the 5600xt itself does not need rebarstare on modern system like yours, it was 1gb due to bar size not alligned to 8GiB in the vbios, it is a known issue.

chris1892006, 2025-09-24:
Oh, so need to disable REBAR in the BIOS?  Otherwise how can we tweak the ReBar using YAABE (Yet Another Atom BIOS Editor)?

xCuri0, 2025-09-24:
@chris1892006 do you even have ReBarDxe installed in your BIOS ?

It's not going to work without it.

There is someone on win-raid forums who claims to know how to modify 5600 XT VBIOS to fix this issue properly you could ask there if you want.

chris1892006, 2025-09-24:
@xCuri0 In the system bios or vga bios?  I'll ask on Win-raid forums.

xCuri0, 2025-09-24:
@chris1892006 system obviously