xCuri0/ReBarUEFI #318: UEFITool cannot find PciBus module

ntn888, 2025-09-25:
Hi I'm in the first step in patching bios for my HP EliteDesk 800 G2. I downloaded the bios application from HP site and extracted the `.bin`. However UEFITool doesn't find the PciBus module. I've tried all three search methods..

[image]

ntn888, 2025-09-25:
> Likely that the .bin from HP website is encrypted or compressed, iirc prebuilds require that you read the BIOS from chip and flash with it too.

Thank you for the kind attention! So this means reading with a hardware flasher, like CH341A, right?

Koekieezz, 2025-09-26:
> > Likely that the .bin from HP website is encrypted or compressed, iirc prebuilds require that you read the BIOS from chip and flash with it too.
> 
> Thank you for the kind attention! So this means reading with a hardware flasher, like CH341A, right?

Yes, you could update the bios to the latest if it isnt the latest, and dump the bios chip afterwards.